wolfSSL/wolfssl #3570: Relaxed ASN.1 parsing seems incorrect

taviso, 2020-12-16:
I think wolfSSL is too relaxed parsing ASN.1, it's possible to add additional objects to certificates and signatures still verify.

Here is an example, [here](https://crt.sh/?d=7446095) is an original intermediate certificate that all the major SSL vendors agree is valid.

https://crt.sh/?id=7446095

If I modify the DER to append another object, openssl and others all reject it:

```diff
--- /dev/fd/63  2020-12-16 10:40:23.435299796 -0800
+++ /dev/fd/62  2020-12-16 10:40:23.435299796 -0800
@@ -1,4 +1,4 @@
-    0:d=0  hl=4 l=1571 cons: SEQUENCE          
+    0:d=0  hl=4 l=1590 cons: SEQUENCE          
     4:d=1  hl=4 l=1291 cons: SEQUENCE          
     8:d=2  hl=2 l=   3 cons: cont [ 0 ]        
    10:d=3  hl=2 l=   1 prim: INTEGER           :02
@@ -72,3 +72,9 @@
  1301:d=2  hl=2 l=   9 prim: OBJECT            :sha256WithRSAEncryption
  1312:d=2  hl=2 l=   0 prim: NULL              
  1314:d=1  hl=4 l= 257 prim: BIT STRING        
+ 1575:d=1  hl=2 l=   0 prim: EOC               
+ 1577:d=1  hl=2 l=  15 cons: SEQUENCE          
+ 1579:d=2  hl=2 l=   3 prim: OBJECT            :X509v3 Basic Constraints
+ 1584:d=2  hl=2 l=   1 prim: BOOLEAN           :255
+ 1587:d=2  hl=2 l=   5 prim: OCTET STRING      [HEX DUMP]:30030101FF
+ 1594:d=0  hl=2 l=   0 prim: EOC               

```

Result:

```
$ openssl verify 00c1bacffa6c259d18cd0d67be30720b.pem 
unable to load certificate
139856317945152:error:0D078094:asn1 encoding routines:asn1_item_embed_d2i:sequence length mismatch:../crypto/asn1/tasn_dec.c:386:Type=X509
139856317945152:error:0906700D:PEM routines:PEM_ASN1_read_bio:ASN1 lib:../crypto/pem/pem_oth.c:33:
```

But WolfSSL still accepts it:

```
$ ./wolfdecode 00c1bacffa6c259d18cd0d67be30720b.der 
** Message: 10:43:04.748: Certificate 00c1bacffa6c259d18cd0d67be30720b.der DOES parse AND VERIFIES with wolfssl
```

(I'm just using `wolfSSL_CertManagerVerifyBuffer`)

Here is the modified PEM, 

```
-----BEGIN CERTIFICATE-----
MIIGNjCCBQugAwIBAgIIGN2Hrh9LtnkwDQYJKoZIhvcNAQELBQAwgZUxCzAJBgNVBAYTAkdSMUQw
QgYDVQQKEztIZWxsZW5pYyBBY2FkZW1pYyBhbmQgUmVzZWFyY2ggSW5zdGl0dXRpb25zIENlcnQu
IEF1dGhvcml0eTFAMD4GA1UEAxM3SGVsbGVuaWMgQWNhZGVtaWMgYW5kIFJlc2VhcmNoIEluc3Rp
dHV0aW9ucyBSb290Q0EgMjAxMTAeFw0xNTA1MDUxMzM1NTRaFw0yMzA1MDMxMzM1NTRaMHkxCzAJ
BgNVBAYTAkdSMS0wKwYDVQQKDCRBcmlzdG90bGUgVW5pdmVyc2l0eSBvZiBUaGVzc2Fsb25pa2kx
OzA5BgNVBAMMMkFyaXN0b3RsZSBVbml2ZXJzaXR5IG9mIFRoZXNzYWxvbmlraSBDZW50cmFsIENB
IFI1MIIBIjANBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEA47zolifYL6IOiRyscXwvwn2opSkq
cqSgZ+tjPDN0o8ylifFdPIDx6rBgvI6ZzZAucPiKRQiQ3M03jk1QA5iWWsqz93s03CUL9oe2jyWe
pa3qYfhs10nOZabxpG/PrXNAy6IMmDUcrUKrnEe2fAwIWQft60tvMxSzpOGMP/EqHeNURVObHL+H
dG085LY77AHBlJFpV2Nq6eED+WYEjWh+kR6yy20o0qLdqh1SGEKMw1kzyRtDIiski8tg+54y7Fqr
gBm7l6TZ64iceyOyMer6zb8cnu41ejO6nM0NbCfEEz+tLuxZA4IxbcnQN1k3LRmpRqdFlTCG2vJv
xM/ymNdyKQIDAQABo4ICkDCCAowwDwYDVR0TAQH/BAUwAwEB/zAOBgNVHQ8BAf8EBAMCAQYwHQYD
VR0OBBYEFHSRhM2Pm8XI+z6jdPJOdLv5mu5GMEYGA1UdHwQ/MD0wO6A5oDeGNWh0dHA6Ly9jcmx2
MS5oYXJpY2EuZ3IvSGFyaWNhUm9vdENBMjAxMS9jcmx2MS5kZXIuY3JsMB8GA1UdIwQYMBaAFKaR
Qv0TYUojngikKeXYEwQj7kElMG4GCCsGAQUFBwEBBGIwYDAhBggrBgEFBQcwAYYVaHR0cDovL29j
c3AuaGFyaWNhLmdyMDsGCCsGAQUFBzAChi9odHRwOi8vd3d3LmhhcmljYS5nci9jZXJ0cy9IYXJp
Y2FSb290Q0EyMDExLmNydDCCAUAGA1UdIASCATcwggEzMIIBLwYLKwYBBAG8HQIAAwYwggEeMDQG
CCsGAQUFBwIBFihodHRwOi8vd3d3LnBraS5hdXRoLmdyL2RvY3VtZW50cy9DUFMucGhwMIHlBggr
BgEFBQcCAjCB2DBKFkNIZWxsZW5pYyBBY2FkZW1pYyBhbmQgUmVzZWFyY2ggSW5zdGl0dXRpb25z
IENlcnRpZmljYXRpb24gQXV0aG9yaXR5MAMCAQEagYlUaGlzIGNlcnRpZmljYXRlIGlzIHN1Ympl
Y3QgdG8gR3JlZWsgbGF3cyBhbmQgb3VyIENQUy4gVGhpcyBDZXJ0aWZpY2F0ZSBtdXN0IG9ubHkg
YmUgdXNlZCBmb3IgYWNhZGVtaWMsIHJlc2VhcmNoIG9yIGVkdWNhdGlvbmFsIHB1cnBvc2VzLjAt
BgNVHR4EJjAkoCIwCYIHYXV0aC5ncjAJgQdhdXRoLmdyMAqBCC5hdXRoLmdyMA0GCSqGSIb3DQEB
CwUAA4IBAQCmS8Poeu8bzFF6Ze6UPnB+cge2pJSHp9xInG/K8BmVRl+uyUaMGHYnN0kh4Y90H1VK
6mPj0OA8FCoj4ZvDwq9K5QUv/rmN8jWE/v/oqUo1ZLuX/fwGvuBZf5Mj+VStEQ3DEyUXGkDqGv8U
bveDLLwg9bzM5bTn7G62grmMrp6/GcNQX63IIxN3Lj6dfV1ntaiJHqO+o7bk/m+QHAcivqOQ7e5E
y67PDNQQjH3dNRKgDdloDltnBURactJ3DeFuwcOFb259BKYbLDMivVj7OMUkPnVOQANqpwEWvPjD
OURDWNt3uq34+OnB49JYQUztsf34smogqh6urQP5OzgSYN3xAAAwDwYDVR0TAQH/BAUwAwEB/wAA
-----END CERTIFICATE-----
```

I feel like this is a bug, does this seem correct? I feel like there might be a vulnerability here, but I'm not sure.

tmael, 2020-12-17:
Hi @taviso, 

Thanks for reaching out to wolfSSL with a potential issue. 
Will review it and get back to you tomorrow. 

Thanks,
Tesfa

SparkiDev, 2020-12-18:
Hi @taviso 

I've prepared a pull request that will fix this issue: https://github.com/wolfSSL/wolfssl/pull/3578
Please check this works as you require.

Thanks,
Sean

taviso, 2020-12-19:
Thank you, perfect!

tmael, 2020-12-19:
Fixed with #3578

closed 2020-12-19 by tmael