wolfSSL/wolfssl #3576: Error while runnin ./configure command

ines96, 2020-12-17:
When running ./configure I got this error:

"...
checking whether make supports nested variables... (cached) yes
./configure: line 5159: syntax error near unexpected token `2.2'
./configure: line 5159: `LT_PREREQ(2.2)'
"

Does someone have any idea what is causing this?

elms, 2020-12-17:
Ola @ines96

I don't see anything like in my Ubuntu 20.04 generated configure script. The `LT_PREREQ(2.2)` is something from `configure.ac`

Can you tell us a little about what system you are building on to better be able to reproduce?
 * Are you building from the head of tree or a release?
 * What OS (eg `uname -a`?
 * can you share the output of `autoconf -V`? and the configure script?
 * What commands did you run to get to this error?

elms, 2020-12-17:
@ines96 one suggestion is to make sure you ran `autogen.sh` to generate the configure script for wolfSSL.

ines96, 2020-12-18:
Hello @elms 

- Are you building from the head of tree or a release?
I cloned this github page.

- What OS (eg uname -a)?
Linux raspberry 4.19.0-8-amd64 #1 SMP Debian 4.19.98-1 (2020-01-26) x86_64 GNU/Linux

- can you share the output of autoconf -V? and the configure script?
autoconf (GNU Autoconf) 2.69
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+/Autoconf: GNU GPL version 3 or later
<http://gnu.org/licenses/gpl.html>, <http://gnu.org/licenses/exceptions.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by David J. MacKenzie and Akim Demaille.

- What commands did you run to get to this error?
This happens when I run ./configure after autogen.sh

elms, 2020-12-18:
@tmael pointed me to https://github.com/ngtcp2/ngtcp2/issues/78 and I was able to reproduce your issue by removing libtool on a fresh checkout.

You should just need to install `libtool` via `sudo apt install libtool`. Let me know if that resolves your issue.

ines96, 2020-12-18:
@elms I have already installed `libtool`.

elms, 2020-12-18:
@ines96 Does running `autogen.sh` report any errors? Can you share the output from running it?

When I was able to generate the same error, here is the output I saw from `autogen.sh`

<details>
<summary>autogen libtool error</summary>

```
 ./autogen.sh 
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
autoreconf: running: /usr/bin/autoheader --force
autoreconf: running: automake --add-missing --copy --force-missing
Makefile.am:8: error: Libtool library used but 'LIBTOOL' is undefined
Makefile.am:8:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
Makefile.am:8:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
Makefile.am:8:   If 'LT_INIT' is in 'configure.ac', make sure
Makefile.am:8:   its definition is in aclocal's search path.
autoreconf: automake failed with exit status: 1
INSTALL libtool and rerun autogen.sh
```

</details>

<details>
<summary>and resulting configure error</summary>

```
$ ./configure 
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking build system type... armv7l-unknown-linux-gnueabihf
checking host system type... armv7l-unknown-linux-gnueabihf
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking for style of include used by make... GNU
checking whether make supports nested variables... yes
checking whether UID '1000' is supported by ustar format... yes
checking whether GID '1000' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking dependency style of gcc... gcc3
checking whether make supports nested variables... (cached) yes
./configure: line 5192: syntax error near unexpected token `2.2'
./configure: line 5192: `LT_PREREQ(2.2)'
```

</details>

ines96, 2020-12-19:
@elms I managed to run configure smoothly. It's weird but I just ran again autogen.sh and then ./configure and it worked.

Thank you for your help.

elms, 2020-12-19:
👍 Glad your issue is resolved. I'll close this issue. Always happy to help.

closed 2020-12-19 by elms