wolfSSL/wolfssl #3639: wolfssl cannot be used via CMake add_subdirectory()

tlewowski, 2021-01-09:
I wanted to add `wolfssl` to my build (to compile it together with my project). Since `wolfssl` uses CMake, I thought of simply adding it via `add_subdirectory(wolfssl)`. However, I encountered problems with file `cmake/config.in` not existing in my project sources.

This is caused by lines 1297-1298 of `CMakeLists.txt`:

```
configure_file("${CMAKE_SOURCE_DIR}/cmake/config.in"
               "${CMAKE_BINARY_DIR}/config.h" )
```

which means that files `cmake/config.in` and `config.h` are taken from the build root, not from `wolfssl` project. 

Was that the intention? All the other places in the file use `CMAKE_CURRENT_SOURCE_DIR` and `CMAKE_CURRENT_BINARY_DIR`, and after modifying this line I'm able to compile without any issues.

haydenroche5, 2021-01-10:
This was indeed unintentional. Coincidentally, I've fixed this same issue in this PR, too: https://github.com/wolfSSL/wolfssl/pull/3617

tlewowski, 2021-01-11:
awesome, thank you!

closed 2021-01-11 by tlewowski