wolfSSL/wolfssl #3651: SSL_connect return WOLFSSL_FATAL_ERROR with WANT_READ in GetInputData

DevriesL, 2021-01-14:
Hello,
    When I using Wolfssl 4.6.0 to do a SSL_connect, it always return `WOLFSSL_FATAL_ERROR`, then I find that the interal error is because of when `GetInputData` is calling `wolfSSLReceive`, the return code `in` is `WANT_READ`, the logic is in internal.c line 14991 like that:
```
        in = wolfSSLReceive(ssl,
                     ssl->buffers.inputBuffer.buffer +
                     ssl->buffers.inputBuffer.length,
                     inSz);
        if (in == WANT_READ)
            return WANT_READ;
```

But when I change it to 

```
        in = wolfSSLReceive(ssl,
                     ssl->buffers.inputBuffer.buffer +
                     ssl->buffers.inputBuffer.length,
                     inSz);
        if (in == WANT_READ)
            continue;
```

Everything works fine, I don't know it that design of `return WANT_READ` is correct or we need more logic to deal with this part?

tmael, 2021-01-15:
Hi @DevriesL,

Thanks for reaching out to wolfSSL. 

How are you setting up your application? Is wolfSSL your client or server or both?  a return of WANT_READ is not an error case.  It sounds like you are application is set up for non-blocking. 

You should consider using something like these when you setup your send/receive application:

```
        do {
            ret = wolfSSL_accept(srv_ssl);
            err = wolfSSL_get_error(srv_ssl, ret);
        }
        while (err == WOLFSSL_ERROR_WANT_READ || err == WOLFSSL_ERROR_WANT_WRITE);

        do
        {
            ret = wolfSSL_connect(cli_ssl);
            err = wolfSSL_get_error(cli_ssl, ret);
        }
        while (err == WOLFSSL_ERROR_WANT_READ || err == WOLFSSL_ERROR_WANT_WRITE);
```

Let me know if this helps.

DevriesL, 2021-01-15:
Hello @tmael , thanks for your reply, I am using the AWS IoT C SDK with WolfSSL, the demo code for SSL_connect is non-blocking, I will try your method and give feedback later.

tmael, 2021-02-12:
Hi @DevriesL,

 Do you have any additional questions or should we consider this issue resolved?

DevriesL, 2021-02-18:
Hello, this issue has been solved, thanks for your help :D

closed 2021-02-18 by DevriesL