wolfSSL/wolfssl #3659: Segfault in the server example program

pictyeye, 2021-01-18:
With current version of wolfssl, `./examples/server/server -v 1 -x` segfaults

-x is supposed to continue execution and not close the connection. However in the case we force the use of TLS 1.0 with a wolfssl installation which does not support this version, we end up with a Segmentation Fault.

Here is a simple reproducer using a debian:buster Docker

    $ apt update && apt install -y git make gcc dh-autoreconf
    $ git clone https://github.com/wolfSSL/wolfssl.git
    $ cd wolfssl
    $ ./autogen.sh
    $ ./configure
    $ make -j
    $ ./examples/server/server -v 1 -x
    wolfSSL error: Bad SSL version
    Continuing server execution...
    
    wolfSSL error: unable to get method
    Continuing server execution...
    
    Segmentation fault (core dumped)

The segfault was found on `v4.5.0-stable` and `v4.6.0-stable`, but also works on current `master` branch (`0ac43bb09557578be61967e802234e05e087540a`)

This might be the expected behaviour, in which case I appologize for the report, but it broke some of our tests in a surprising way. We just added the `--enable-tls-v10` switch to `./configure` to fix our workflow.

embhorn, 2021-01-18:
Hello @pictyeye 

Thank you for bringing this to our attention. I was able to reproduce this, so thanks also for the great reproducer instructions.

This is a bug in the server example, and I also found it in the client example. We are always striving to have the best demonstrations. I've created a fix and posted the PR here:
https://github.com/wolfSSL/wolfssl/pull/3662

Let us know if this also fixes the issue for you.

Kind regards,
Eric @ wolfSSL Support

pictyeye, 2021-01-20:
Thank you for your prompt reply and for the patch.

I just checked out your patch and I confirm that the example program now gracefully stops.

closed 2021-01-27 by embhorn