wolfSSL/wolfssl #3661: Support for WOLFSSL_SSLKEYLOGFILE with TLS 1.3

pictyeye, 2021-01-18:
At configuration time, it is possible to enable `SHOW_SECRETS` and `WOLFSSL_SSLKEYLOGFILE` to compile the `SSLKEYLOGFILE` feature in `WOLFSSL`.

However, as far as I can tell, the feature only works up to TLS 1.2.

Since wolfssl supports TLS 1.3 and since the `SSLKEYLOGFILE` format has been [updated to handle the new protocol version](https://developer.mozilla.org/en-US/docs/Mozilla/Projects/NSS/Key_Log_Format), would it be possible to add support for `SSLKEYLOGFILE` in TLS 1.3?

If I understand correctly, the current implementation lives in `src/tls.c` (in the `MakeTlsMasterSecret()` function, starting from line 583 in the current release). I believe a similar work would have to be done in `src/tls13.c` in the series of `Derive*()` functions (or maybe just in the underlying `DeriveKeyMsg()` and `DeriveKey()` functions?).

Would there be interest in such a feature?

embhorn, 2021-01-19:
Hello @pictyeye 

Thanks for the note. I've added this to our internal feature request list. We try to work on requested features whenever we get free engineering cycles unless a customer decides to back the effort for a finite implementation timeline. If at any time a feature becomes a progress blocker for a project it can be accelerated. That process can be started by contacting the  account business manager.

Thanks,
Eric @ wolfSSL Support

pictyeye, 2021-01-20:
Thank you for the information.

We might try and implement a first version of the proposal on our own. When we do, we will propose a Pull Request.

embhorn, 2021-06-14:
Hi @pictyeye 

Did you have any other questions? Can I close this issue?

dgarske, 2022-06-24:
FYI: We did add support for this using a build option to enable secret callback `HAVE_SECRET_CALLBACK`.

Enables a callback function for the derived secrets:

Example: 
https://github.com/wolfSSL/wolfssl-examples/tree/master/tls#tls-v13-wireshark-logging
https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/client-tls13.c#L52
Make sure you call `wolfSSL_KeepArrays(ssl);` to keep the random secret around for the callback to access.
https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/client-tls13.c#L217

closed 2023-01-31 by dgarske