wolfSSL/wolfssl #3682: Compilation failure on Windows 10 for mingw 32bit

Fulguritude, 2021-01-25:
Pulled most recent commit to master. My shell is cygwin. Ran autogen.sh.

Configured with:
```
 ./configure --enable-static CC=i686-w64-mingw32-gcc
 ```
 
 Ran make:
 
 ```
 $ make
make -j9  all-recursive
make[1]: Entering directory '/home/fulguritude/LEDR/wolfssl'
make[2]: Entering directory '/home/fulguritude/LEDR/wolfssl'
make[2]: warning: -j9 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/benchmark/benchmark.o
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
  CC       wolfcrypt/src/src_libwolfssl_la-random.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha256.lo
  CC       wolfcrypt/src/src_libwolfssl_la-rsa.lo
  CC       wolfcrypt/src/src_libwolfssl_la-aes.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha512.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha3.lo
  CC       wolfcrypt/src/src_libwolfssl_la-logging.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wc_port.lo
  CC       wolfcrypt/src/src_libwolfssl_la-error.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wc_encrypt.lo
  CC       wolfcrypt/src/src_libwolfssl_la-signature.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wolfmath.lo
In file included from wolfcrypt/src/wc_port.c:28:
wolfcrypt/src/wc_port.c: In function ‘wc_ReadDirFirst’:
./wolfssl/wolfcrypt/types.h:508:35: error: ‘strncpy’ output may be truncated copying between 2 and 255 bytes from a string of length 259 [-Werror=stringop-truncation]
  508 |         #define XSTRNCPY(s1,s2,n) strncpy((s1),(s2),(n))
      |                                   ^~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/wc_port.c:459:13: note: in expansion of macro ‘XSTRNCPY’
  459 |             XSTRNCPY(ctx->name + pathLen + 1,
      |             ^~~~~~~~
wolfcrypt/src/wc_port.c: In function ‘wc_ReadDirNext’:
./wolfssl/wolfcrypt/types.h:508:35: error: ‘strncpy’ output may be truncated copying between 0 and 255 bytes from a string of length 259 [-Werror=stringop-truncation]
  508 |         #define XSTRNCPY(s1,s2,n) strncpy((s1),(s2),(n))
      |                                   ^~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/wc_port.c:595:13: note: in expansion of macro ‘XSTRNCPY’
  595 |             XSTRNCPY(ctx->name + pathLen + 1,
      |             ^~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:4922: wolfcrypt/src/src_libwolfssl_la-wc_port.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/fulguritude/LEDR/wolfssl'
make[1]: *** [Makefile:5718: all-recursive] Error 1
make[1]: Leaving directory '/home/fulguritude/LEDR/wolfssl'
make: *** [Makefile:3457: all] Error 2
```

Result: does not compile.

embhorn, 2021-01-25:
Hello @Fulguritude 

Thanks for contacting wolfSSL Support. Was this working previously and the update broke your build? It looks like the standard lib headers are missing.

Thanks,
Eric @ wolfSSL

Fulguritude, 2021-01-25:
Hello @embhorn 

A colleague had made a build of libwolfssl for windows around the time I built libwolfssl.24.1.0 for Linux; we had some compatibility issues on Linux recently (probably due to some things being linked statically and others dynamically) which were fixed by updating everything on every Linux to 24.3.0.

For the sake of update and coherence, I'm trying to update on Windows as well, and make everything static (things still work on windows with the dll version from the time of 24.1.0). Do note that the build does seem to still be working on Windows when I don't specify the CC I want; but we'd like to build with mingw because our Windows devs use it rather than msvc in their pipeline. We'd also like to build both 32bit and 64bit version for compatibility issues.

I'm afraid I can't answer your question better than "we recall 32bit and 64bit mingw builds on Windows to be working before, at the time of 24.1.0", but I do not know what configuration (nor compiler) was used at the time. We think my colleague managed to build with mingw successfully then; he's pretty confident he had, though does not recall how he did it, and has no history or logs to verify.

NB: I tried to build from a raw `git clone` yesterday with the same issue

elms, 2021-01-26:
I was able to work around this with a small [change](https://github.com/wolfSSL/wolfssl/commit/f6d5523135a1dc109532efb827952681e6d67fa7), but need to clean it up to coexist with other build configurations.

@Fulguritude can you confirm this addresses your build issue?

embhorn, 2021-01-26:
Hi @Fulguritude 

We recently had an engineer set up for the Cygwin environment. Could you please review the instructions here:

Fulguritude, 2021-01-27:
Hello @elms. Your fix solves some of the compilation issues for 64bit, and does not affect 32bit. However, the build still fails, on linking this time.

```
$ make
make -j9  all-recursive
make[1]: Entering directory '/home/fulguritude/LEDR/wolfssl'
make[2]: Entering directory '/home/fulguritude/LEDR/wolfssl'
make[2]: warning: -j9 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/benchmark/benchmark.o
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
  CC       wolfcrypt/src/src_libwolfssl_la-random.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha256.lo
  CC       wolfcrypt/src/src_libwolfssl_la-rsa.lo
  CC       wolfcrypt/src/src_libwolfssl_la-aes.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha512.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha3.lo
  CC       wolfcrypt/src/src_libwolfssl_la-logging.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wc_port.lo
  CC       wolfcrypt/src/src_libwolfssl_la-error.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wc_encrypt.lo
  CC       wolfcrypt/src/src_libwolfssl_la-signature.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wolfmath.lo
  CC       wolfcrypt/src/src_libwolfssl_la-memory.lo
  CC       wolfcrypt/src/src_libwolfssl_la-dh.lo
  CC       wolfcrypt/src/src_libwolfssl_la-asn.lo
  CC       wolfcrypt/src/src_libwolfssl_la-coding.lo
  CC       wolfcrypt/src/src_libwolfssl_la-poly1305.lo
  CC       wolfcrypt/src/src_libwolfssl_la-md5.lo
  CC       wolfcrypt/src/src_libwolfssl_la-chacha.lo
  CC       wolfcrypt/src/src_libwolfssl_la-chacha20_poly1305.lo
  CC       wolfcrypt/src/src_libwolfssl_la-tfm.lo
  CC       wolfcrypt/src/src_libwolfssl_la-ecc.lo
  CC       src/libwolfssl_la-internal.lo
  CC       src/libwolfssl_la-keys.lo
  CC       src/libwolfssl_la-wolfio.lo
  CC       src/libwolfssl_la-ssl.lo
  CC       src/libwolfssl_la-tls.lo
  CC       src/libwolfssl_la-tls13.lo
  CC       wolfcrypt/test/test.o
  CC       examples/benchmark/tls_bench.o
  CC       examples/client/client.o
  CC       examples/echoclient/echoclient.o
  CC       examples/echoserver/echoserver.o
  CC       examples/server/server.o
  CC       wolfcrypt/test/testsuite_testsuite_test-test.o
  CC       examples/client/testsuite_testsuite_test-client.o
  CC       examples/echoclient/testsuite_testsuite_test-echoclient.o
  CC       examples/echoserver/testsuite_testsuite_test-echoserver.o
  CC       examples/server/testsuite_testsuite_test-server.o
  CC       testsuite/testsuite_test-testsuite.o
  CC       tests/unit_test-unit.o
  CC       tests/unit_test-api.o
  CC       tests/unit_test-suites.o
  CC       tests/unit_test-hash.o
  CC       tests/unit_test-srp.o
  CC       examples/client/tests_unit_test-client.o
  CC       examples/server/tests_unit_test-server.o
  CCLD     src/libwolfssl.la
/usr/lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld: src/.libs/libwolfssl_la-wolfio.o:wolfio.c:(.text+0x13): undefined reference to `__imp_recv'
/usr/lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld: src/.libs/libwolfssl_la-wolfio.o:wolfio.c:(.text+0x2a): undefined reference to `__imp_WSAGetLastError'
/usr/lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld: src/.libs/libwolfssl_la-wolfio.o:wolfio.c:(.text+0x93): undefined reference to `__imp_send'
/usr/lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld: src/.libs/libwolfssl_la-wolfio.o:wolfio.c:(.text+0xa2): undefined reference to `__imp_WSAGetLastError'
/usr/lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld: src/.libs/libwolfssl_la-wolfio.o:wolfio.c:(.text+0xf5): undefined reference to `__imp_recv'
/usr/lib/gcc/x86_64-w64-mingw32/9.2.0/../../../../x86_64-w64-mingw32/bin/ld: src/.libs/libwolfssl_la-wolfio.o:wolfio.c:(.text+0x105): undefined reference to `__imp_send'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:4047: src/libwolfssl.la] Error 1
make[2]: Leaving directory '/home/fulguritude/LEDR/wolfssl'
make[1]: *** [Makefile:5718: all-recursive] Error 1
make[1]: Leaving directory '/home/fulguritude/LEDR/wolfssl'
make: *** [Makefile:3457: all] Error 2
``` 

@embhorn I have all these packages already installed, here is confirmation:
```
fulguritude@MSI ~/Workspace/wolfssl
$ which unzip
/usr/bin/unzip

fulguritude@MSI ~/Workspace/wolfssl
$which autoconf
/usr/bin/autoconf

fulguritude@MSI ~/Workspace/wolfssl
$ which automake
/usr/bin/automake

fulguritude@MSI ~/Workspace/wolfssl
$ which gcc-core
which: no gcc-core in (/usr/local/bin:/usr/bin:/cygdrive/c/Python39/Scripts:/cygdrive/c/Python39:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/c/Windows/System32/OpenSSH:/cygdrive/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/cygdrive/c/Program Files/dotnet:/cygdrive/c/Program Files/nodejs:/cygdrive/c/ProgramData/chocolatey/bin:/cygdrive/c/Program Files/Git/cmd:/cygdrive/c/Users/fulguritude/AppData/Local/Microsoft/WindowsApps:/cygdrive/c/Users/fulguritude/.dotnet/tools:/cygdrive/c/Users/fulguritude/AppData/Roaming/npm:/usr/lib/lapack)

fulguritude@MSI ~/Workspace/wolfssl
$ apt-cyg install gcc-core
Package gcc-core is already installed, skipping

fulguritude@MSI ~/Workspace/wolfssl
$ which libtool
/usr/bin/libtool

fulguritude@MSI ~/Workspace/wolfssl
$ which make
/usr/bin/make
```

Also, a simple
```
$> ./configure --enable-static
$> make
```
does build fine; it's just not the compilers and architectures (32bit AND 64bit) we need to make sure we avoid any issue.

embhorn, 2021-01-27:
Hi @Fulguritude 

Those linker errors are for system library routines. This sounds similar:
https://stackoverflow.com/questions/2033608/mingw-linker-error-winsock

Could you try adding a linker flag to the configure:
`./configure --enable-static CC=i686-w64-mingw32-gcc LDFLAGS="-lws2_32"`

Fulguritude, 2021-02-06:
I tried it with `./configure --enable-static CC=i686-w64-mingw32-gcc LDFLAGS="-lws2_32"`, and I got a failing build for 32 bit:

```
$ make
make -j9  all-recursive
make[1]: Entering directory '/home/fulguritude/Workspace/wolfssl'
make[2]: Entering directory '/home/fulguritude/Workspace/wolfssl'
make[2]: warning: -j9 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/benchmark/benchmark.o
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
  CC       wolfcrypt/src/src_libwolfssl_la-random.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha256.lo
  CC       wolfcrypt/src/src_libwolfssl_la-rsa.lo
  CC       wolfcrypt/src/src_libwolfssl_la-aes.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha512.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha3.lo
  CC       wolfcrypt/src/src_libwolfssl_la-logging.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wc_port.lo
  CC       wolfcrypt/src/src_libwolfssl_la-error.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wc_encrypt.lo
  CC       wolfcrypt/src/src_libwolfssl_la-signature.lo
  CC       wolfcrypt/src/src_libwolfssl_la-wolfmath.lo
In file included from wolfcrypt/src/wc_port.c:28:
wolfcrypt/src/wc_port.c: In function ‘wc_ReadDirFirst’:
./wolfssl/wolfcrypt/types.h:508:35: error: ‘strncpy’ output may be truncated copying between 2 and 255 bytes from a string of length 259 [-Werror=stringop-truncation]
  508 |         #define XSTRNCPY(s1,s2,n) strncpy((s1),(s2),(n))
      |                                   ^~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/wc_port.c:459:13: note: in expansion of macro ‘XSTRNCPY’
  459 |             XSTRNCPY(ctx->name + pathLen + 1,
      |             ^~~~~~~~
wolfcrypt/src/wc_port.c: In function ‘wc_ReadDirNext’:
./wolfssl/wolfcrypt/types.h:508:35: error: ‘strncpy’ output may be truncated copying between 0 and 255 bytes from a string of length 259 [-Werror=stringop-truncation]
  508 |         #define XSTRNCPY(s1,s2,n) strncpy((s1),(s2),(n))
      |                                   ^~~~~~~~~~~~~~~~~~~~~~
wolfcrypt/src/wc_port.c:595:13: note: in expansion of macro ‘XSTRNCPY’
  595 |             XSTRNCPY(ctx->name + pathLen + 1,
      |             ^~~~~~~~
cc1: all warnings being treated as errors
  CC       wolfcrypt/src/src_libwolfssl_la-memory.lo
make[2]: *** [Makefile:4922: wolfcrypt/src/src_libwolfssl_la-wc_port.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/fulguritude/Workspace/wolfssl'
make[1]: *** [Makefile:5718: all-recursive] Error 1
make[1]: Leaving directory '/home/fulguritude/Workspace/wolfssl'
make: *** [Makefile:3457: all] Error 2
```

I then tried  `./configure --enable-static CC=x86_64-w64-mingw32-gcc LDFLAGS="-lws2_32"`, and got a successful build for 64 bit:

```
$ make
make -j9  all-recursive
make[1]: Entering directory '/home/fulguritude/Workspace/wolfssl'
make[2]: Entering directory '/home/fulguritude/Workspace/wolfssl'
make[2]: warning: -j9 forced in submake: resetting jobserver mode.
  CC       wolfcrypt/benchmark/benchmark.o
  CC       wolfcrypt/src/src_libwolfssl_la-hmac.lo
  CC       wolfcrypt/src/src_libwolfssl_la-hash.lo
  CC       wolfcrypt/src/src_libwolfssl_la-cpuid.lo
  CC       wolfcrypt/src/src_libwolfssl_la-random.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha256.lo
  CC       wolfcrypt/src/src_libwolfssl_la-rsa.lo
  CC       wolfcrypt/src/src_libwolfssl_la-aes.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha512.lo
  CC       wolfcrypt/src/src_libwolfssl_la-sha3.lo
  CC       wolfcrypt/src/src_libwolfssl_la-logging.lo
[... 28 lines cut ...]
  CC       examples/client/testsuite_testsuite_test-client.o
  CC       examples/echoclient/testsuite_testsuite_test-echoclient.o
  CC       examples/echoserver/testsuite_testsuite_test-echoserver.o
  CC       examples/server/testsuite_testsuite_test-server.o
  CC       testsuite/testsuite_test-testsuite.o
  CC       tests/unit_test-unit.o
  CC       tests/unit_test-api.o
  CC       tests/unit_test-suites.o
  CC       tests/unit_test-hash.o
  CC       tests/unit_test-srp.o
  CC       examples/server/tests_unit_test-server.o
  CC       examples/client/tests_unit_test-client.o
  CCLD     src/libwolfssl.la
  CCLD     wolfcrypt/benchmark/benchmark.exe
  CCLD     wolfcrypt/test/testwolfcrypt.exe
  CCLD     examples/benchmark/tls_bench.exe
  CCLD     examples/client/client.exe
  CCLD     examples/echoclient/echoclient.exe
  CCLD     examples/echoserver/echoserver.exe
  CCLD     examples/server/server.exe
  CCLD     tests/unit.test.exe
  CCLD     testsuite/testsuite.test.exe
make[2]: Leaving directory '/home/fulguritude/Workspace/wolfssl'
make[1]: Leaving directory '/home/fulguritude/Workspace/wolfssl'
```

elms, 2021-04-15:
@Fulguritude I have a clean PR #3960 to address one of your issues. If possible can you try the builds using `./configure --enable-static --host=i686-w64-mingw32` and `./configure --enable-static --host=x86_64-w64-mingw32` and report any issues?

Fulguritude, 2021-04-16:
Thanks for giving this issue your attention again. TLDR: 1) 32bit success, 2) 64bit failure, 3) my gcc version. No strncpy error anymore though. Also, writing `CC=` used to work, but now only `--host=` works.

Success for 32-bit (`./configure --enable-static --host=i686-w64-mingw32`):
```
fulguritude@MSI ~/wolfssl_pr
$ ./configure --enable-static --host=i686-w64-mingw32
checking for i686-w64-mingw32-gcc... i686-w64-mingw32-gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether i686-w64-mingw32-gcc accepts -g... yes
checking for i686-w64-mingw32-gcc option to accept ISO C89... none needed
checking whether i686-w64-mingw32-gcc understands -c and -o together... yes
checking build system type... x86_64-pc-cygwin
checking host system type... i686-w64-mingw32
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports the include directive... yes (GNU style)
checking whether make supports nested variables... yes
checking whether UID '197609' is supported by ustar format... yes
checking whether GID '197121' is supported by ustar format... yes
checking how to create a ustar tar archive... gnutar
checking dependency style of i686-w64-mingw32-gcc... gcc3
checking whether make supports nested variables... (cached) yes
checking how to print strings... printf
checking for a sed that does not truncate output... /usr/bin/sed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for fgrep... /usr/bin/grep -F
checking for ld used by i686-w64-mingw32-gcc... /usr/i686-w64-mingw32/bin/ld.exe
checking if the linker (/usr/i686-w64-mingw32/bin/ld.exe) is GNU ld... yes
checking for BSD- or MS-compatible name lister (nm)... /usr/bin/i686-w64-mingw32-nm -B
checking the name lister (/usr/bin/i686-w64-mingw32-nm -B) interface... BSD nm
checking whether ln -s works... yes
checking the maximum length of command line arguments... 8192
checking how to convert x86_64-pc-cygwin file names to i686-w64-mingw32 format... func_convert_file_cygwin_to_w32
checking how to convert x86_64-pc-cygwin file names to toolchain format... func_convert_file_noop
checking for /usr/i686-w64-mingw32/bin/ld.exe option to reload object files... -r
checking for i686-w64-mingw32-objdump... i686-w64-mingw32-objdump
checking how to recognize dependent libraries... file_magic ^x86 archive import|^x86 DLL
checking for i686-w64-mingw32-dlltool... i686-w64-mingw32-dlltool
checking how to associate runtime and link libraries... func_cygming_dll_for_implib
checking for i686-w64-mingw32-ar... i686-w64-mingw32-ar
checking for archiver @FILE support... @
checking for i686-w64-mingw32-strip... i686-w64-mingw32-strip
checking for i686-w64-mingw32-ranlib... i686-w64-mingw32-ranlib
checking command to parse /usr/bin/i686-w64-mingw32-nm -B output from i686-w64-mingw32-gcc object... ok
checking for sysroot... no
checking for a working dd... /usr/bin/dd
checking how to truncate binary pipes... /usr/bin/dd bs=4096 count=1
checking for i686-w64-mingw32-mt... no
checking for mt... no
checking if : is a manifest tool... no
checking how to run the C preprocessor... i686-w64-mingw32-gcc -E
checking for ANSI C header files... yes
checking for sys/types.h... yes
checking for sys/stat.h... yes
checking for stdlib.h... yes
checking for string.h... yes
checking for memory.h... yes
checking for strings.h... yes
checking for inttypes.h... yes
checking for stdint.h... yes
checking for unistd.h... yes
[... 887 lines cut ...]
  CC       src/libwolfssl_la-keys.lo
  CC       src/libwolfssl_la-ssl.lo
  CC       src/libwolfssl_la-tls.lo
  CC       src/libwolfssl_la-tls13.lo
cc1: all warnings being treated as errors
make[1]: *** [Makefile:4936: wolfcrypt/src/src_libwolfssl_la-tfm.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/fulguritude/wolfssl_pr'
make: *** [Makefile:3292: all] Error 2
```

Also:
```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-cygwin/10/lto-wrapper.exe
Target: x86_64-pc-cygwin
Configured with: /mnt/share/cygpkgs/gcc/gcc.x86_64/src/gcc-10.2.0/configure --srcdir=/mnt/share/cygpkgs/gcc/gcc.x86_64/src/gcc-10.2.0 --prefix=/usr --exec-prefix=/usr --localstatedir=/var --sysconfdir=/etc --docdir=/usr/share/doc/gcc --htmldir=/usr/share/doc/gcc/html -C --build=x86_64-pc-cygwin --host=x86_64-pc-cygwin --target=x86_64-pc-cygwin --without-libiconv-prefix --without-libintl-prefix --libexecdir=/usr/lib --with-gcc-major-version-only --enable-shared --enable-shared-libgcc --enable-static --enable-version-specific-runtime-libs --enable-bootstrap --enable-__cxa_atexit --with-dwarf2 --with-tune=generic --enable-languages=c,c++,fortran,lto,objc,obj-c++ --enable-graphite --enable-threads=posix --enable-libatomic --enable-libgomp --enable-libquadmath --enable-libquadmath-support --disable-libssp --enable-libada --disable-symvers --with-gnu-ld --with-gnu-as --with-cloog-include=/usr/include/cloog-isl --without-libiconv-prefix --without-libintl-prefix --with-system-zlib --enable-linker-build-id --with-default-libstdcxx-abi=gcc4-compatible --enable-libstdcxx-filesystem-ts
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 10.2.0 (GCC)
```

Finally, cygwin mintty v 3.4.1.0

elms, 2021-04-16:
@Fulguritude Can you confirm you checked out the PR? What SHA is the tree at(`git describe`)?

As a side note, using `--host` is recommended as it can set other related flags (eg link ws2_32) where as setting `CC` just replaces the compiler, but it should still work with appropriate additional flags.

Fulguritude, 2021-04-16:
You're right, my bad; I think I must have written dashes instead of underscores when checking out `msys_build_fix` and not read the error. I can confirm that both builds work for `cc4116de24248376f2cd932a103099a9a83a1043`. Thanks a lot !

embhorn, 2021-04-16:
I was able to reproduce the original issue, but using #3960 from @elms , `./configure --enable-static --host=x86_64-w64-mingw32`  builds clean for me also.

elms, 2021-04-24:
PR #3960 has merged. Closing this issue. @Fulguritude thanks for the report and let us know if you have any future issues.

closed 2021-04-24 by elms