wolfSSL/wolfssl #3685: Porting Wolfssl for Visual Studio from source code

lijintv07, 2021-01-25:
How to port Wolfssl for windows Visual Studio from source, without using any .lib and .dll : Is there any porting guide available for this?

embhorn, 2021-01-25:
Hello @lijintv07 

Thanks for contacting wolfSSL Support. Yes, wolfSSL is highly portable! We have a porting guide available here:
https://www.wolfssl.com/docs/porting-guide/

Could you tell us a bit more about what you are trying to accomplish?

Thanks,
Eric @ wolfSSL

kaleb-himes, 2021-01-25:
Greetings @lijintv07,

As @embhorn mentioned we would love to hear more about what is driving your efforts and the end goals of your project.

To get you started we provide wolfssl64.sln in the root directory of wolfSSL, it supports build combinations in MSVS (any version 2012 or newer) of:

WIN32 | Debug
WIN32 | Release
WIN32 | DLL Debug
WIN32 | DLL Release

X64 | Debug
X64 | Release
X64 | DLL Debug
X64 | DLL Release

Please let us know if you have any issues getting the solution (which includes the library and some examples) to build.

Warm Regards,

K

embhorn, 2021-06-15:
Closing for now, but please feel free to reopen with follow-up questions.

closed 2021-06-15 by embhorn