wolfSSL/wolfssl #3690: For a cert with critical policyConstraints, wolfSSL rejects it while OpenSSL accepts it

GOODPWDCETCSZ, 2021-01-26:
For a cert with critical policyConstraints, wolfSSL rejects it while OpenSSL v1.1.1f accepts it.

**Env**: Ubuntu x64

**wolfSSL version**: 4.6.0

**Reproduce**: `./wolfssl_verify_pem ca.pem seed-16s31-255s21-363s29.pem`

**Expected result**: 

Consistent verification result between wolfSSL and OpenSSL.

**Actual result**:
```
wolfSSL_CertManagerVerify filed and with return code -198 and error message Name Constraint error
```

**Attachment**
[ca.zip](https://github.com/wolfSSL/wolfssl/files/5870811/ca.zip)
[seed-16s31-255s21-363s29.zip](https://github.com/wolfSSL/wolfssl/files/5870812/seed-16s31-255s21-363s29.zip)

``` # wolfssl_verify_pem.c
// gcc -g wolfssl_verify_pem.c -o wolfssl_verify_pem -lwolfsso
#include <stdlib.h>
#include <wolfssl/ssl.h>
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/error-crypt.h>

int wolfSSL_Verify_PEM(char * cac, char * ec){
	int ret = 0;

	// to create a new wolfSSL cert manager
	WOLFSSL_CERT_MANAGER* cm;
	cm = wolfSSL_CertManagerNew();
	if (cm == NULL){
		printf("Creating a new wolfSSL_CertManager failed!\n");
		exit(1);
	}
	
	// to load cac to the created wolfSSL_CertManger
	ret = wolfSSL_CertManagerLoadCA(cm, cac, NULL);
	if (ret != SSL_SUCCESS){
		printf("Loading cac to the created wolfSSL_CertManger failed!\n");
		exit(2);
	}

	// to verify the ec in the created wolfSSL_CertManger
	ret = wolfSSL_CertManagerVerify(cm, ec, SSL_FILETYPE_PEM);
	if (ret != SSL_SUCCESS){
		printf("wolfSSL_CertManagerVerify filed and with return code %d and error message %s\n",
			 ret,
			 wolfSSL_ERR_reason_error_string(ret));
	}
	else{
		printf("The target cert has passed through verification.\n");
	}

	// to free cm
	wolfSSL_CertManagerFree(cm);

	return ret;
}

int main(int argc, char ** argv){
	char * cac = argv[1];
	char * ec = argv[2];
	wolfSSL_Verify_PEM(cac, ec);
	return 0;
}
```

embhorn, 2021-01-26:
Hello @GOODPWDCETCSZ 

wolfSSL is more stringent ion adhering to the specification than some other libraries. Could you try building wolfSSL with these settings:
`WOLFSSL_ALLOW_CRIT_SKID`
`WOLFSSL_NO_ASN_STRICT`
This will disable many of the critical requirement specified in RFC 5280. As a precautionary note we do not recommend using these settings however we do realize there are CA's that issue non-conforming certificates that violate RFC 5280 so they are available for when absolutely necessary. Let us know if that resolves the issues you are seeing.

GOODPWDCETCSZ, 2021-01-27:
Hello @embhorn  Could you tell us where and how to use `WOLFSSL_ALLOW_CRIT_SKID` and `WOLFSSL_NO_ASN_STRICT`? Thanks!

embhorn, 2021-01-27:
Hi @GOODPWDCETCSZ 

If building in *nix, you can use:
`./configure CFLAGS="-DWOLFSSL_ALLOW_CRIT_SKID -DWOLFSSL_NO_ASN_STRICT"`

GOODPWDCETCSZ, 2021-01-28:
@embhorn Thanks. We will try it.

GOODPWDCETCSZ, 2021-01-28:
@embhorn `WOLFSSL_ALLOW_CRIT_SKID` and `WOLFSSL_NO_ASN_STRICT` take effect. Thanks.

closed 2021-01-28 by embhorn