wolfSSL/wolfssl #3691: Tools (curl, wget) that are built with wolfssl can not disable cert checking?

PolynomialDivision, 2021-01-26:
If you try

    curl --insecure https://127.0.0.1

with a self signed certificate, the check should be skipped and via an tsl connection the website should be downloaded.

Same for

    wget --no-check-certificate -O- https://localhost/

embhorn, 2021-01-26:
Hi @PolynomialDivision - Thanks for contacting wolfSSL Support.

@bagder - Would you consider this an integration issue?

bagder, 2021-01-26:
I don't know. I haven't been able to reproduce this yet so I don't know what wolfSSL returns to curl or why!

bagder, 2021-01-26:
Apart from the -152 error then, which seems to indicate that wolfSSL doesn't like the cert.

dgarske, 2021-01-26:
`ASN_SIG_OID_E      = -152,  /* ASN signature error, mismatched oid */`

bagder, 2021-01-26:
Right, but what does it mean? What oid is mismatched and is that a reason for the entire handshake to fail or is should it rather be allowed when we skip the certificate verification?

hnyman, 2021-01-26:
Looking at the source, the explanation gives some light:

> case ASN_SIG_OID_E :
> return "ASN signature error, mismatched oid";
> Returned if the signature encryption type is not the same as the encryption type of the certificate in the provided file
 
Further debugging on OpenWrt forum:  

@PolynomialDivision tested my self-signed key generated with OpenSSL tools, and the key works for him also with curl/wolfssl, possibly because it has only ecdsa/256 style algorithms in use. 
```
root@router1:/etc# openssl x509 -text -noout -inform der -in /etc/uhttpd.crt
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
1a:8a:7b:92:a3:f8...
        Signature Algorithm: ecdsa-with-SHA256
        Issuer: C = ZZ, ST = Somewhere, L = Unknown, O = OpenWrt3c77dc87, CN = OpenWrt
        Validity
            Not Before: Jan  6 07:11:51 2021 GMT
            Not After : Jan  6 07:11:51 2023 GMT
        Subject: C = ZZ, ST = Somewhere, L = Unknown, O = OpenWrt3c77dc87, CN = OpenWrt
        Subject Public Key Info:
            Public Key Algorithm: id-ecPublicKey
                Public-Key: (256 bit)
                pub:
                    04:e3:c2:74:a...
                    d4:ec:95:94:af
                ASN1 OID: prime256v1
                NIST CURVE: P-256
        X509v3 extensions:
            X509v3 Subject Key Identifier:
3A:34:99:4A:6...
            X509v3 Authority Key Identifier:
keyid:3A:34:99:4...

            X509v3 Basic Constraints: critical
                CA:TRUE
    Signature Algorithm: ecdsa-with-SHA256
         30:44:02...
```

His own key (shown in the curl issue tracker linked above) fails with curl/wolfssl, as it has RSA/ecdsa/P256 elliptic mixed???

```
openssl x509 -text -noout -in uhttpd.crt 
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            1e:78:d7:a6:3b:...
        Signature Algorithm: sha1WithRSAEncryption
        Issuer: C = ZZ, ST = Somewhere, L = Unknown, O = OpenWrt3ca99c2a, CN = OpenWrt
        Validity
            Not Before: Jan 25 10:34:35 2021 GMT
            Not After : Jan 26 10:34:35 2023 GMT
        Subject: C = ZZ, ST = Somewhere, L = Unknown, O = OpenWrt3ca99c2a, CN = OpenWrt
        Subject Public Key Info:
            Public Key Algorithm: id-ecPublicKey
                Public-Key: (256 bit)
                pub:
                    04:a6:7b:7c:9....
                ASN1 OID: prime256v1
                NIST CURVE: P-256
    Signature Algorithm: ecdsa-with-SHA256
         30:45:02:20:79:2...
```

The reason might be in how the OpenWrt tool to create a self-signed cert with wolfssl operates, so any input for that would be nice.  (btw, the current default is EC with P-256)

C code:  https://github.com/openwrt/openwrt/blob/master/package/utils/px5g-wolfssl/px5g-wolfssl.c
script: https://github.com/openwrt/openwrt/blob/master/package/network/services/uhttpd/files/uhttpd.init#L34-L64

PolynomialDivision, 2021-01-26:
@hnyman 
Using only RSA is not working. :/

```
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            3d:e1:a8:a9:31:e0:5e:a1:5f:00:9b:e8:49:f0:88:56
        Signature Algorithm: sha1WithRSAEncryption
        Issuer: C = ZZ, ST = Somewhere, L = Unknown, O = OpenWrtc973989d, CN = OpenWrt
        Validity
            Not Before: Jan 25 21:15:51 2021 GMT
            Not After : Jan 26 21:15:51 2023 GMT
        Subject: C = ZZ, ST = Somewhere, L = Unknown, O = OpenWrtc973989d, CN = OpenWrt
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                RSA Public-Key: (2048 bit)
                Modulus:
                    00:b5:98:65:b9:c5:5f:98:0e:6e:16:98:b0:09:b8:
                    4e:39:a2:d2:91:c9:8a:40:90:43:fa:3e:9f:17:2e:
                    a7:b4:30:ba:41:fc:80:4c:cc:24:d4:86:22:8d:ed:
                    af:5a:69:aa:fa:0e:cc:5b:ac:09:d9:c6:51:62:6f:
                    db:cf:41:82:19:81:d9:e3:ed:d1:ae:0c:6c:c5:a0:
                    54:9c:c4:0f:b7:2f:13:f4:29:1d:87:55:6d:4a:f4:
                    44:a3:81:26:93:df:bb:12:ef:07:5e:b3:67:85:68:
                    5d:11:07:b7:b1:f1:24:71:0b:dd:83:c4:3c:40:5f:
                    8e:65:ca:86:af:b7:8b:73:96:43:64:72:3b:df:bc:
                    3b:3c:30:db:b7:43:16:08:f4:68:87:20:e5:56:49:
                    16:6e:77:1b:b5:5f:57:97:bf:0d:85:db:7a:14:a6:
                    b2:eb:ff:16:b9:3f:8c:59:20:c6:4b:00:80:ca:f2:
                    77:c8:26:d0:bb:8a:04:72:59:7b:cc:9b:32:31:fc:
                    7c:c7:ff:fd:73:d2:11:99:c9:0c:8e:40:76:cd:b2:
                    66:af:2a:ea:b3:68:b3:cd:f4:cc:53:9b:db:5a:ba:
                    2d:93:96:db:3c:c8:8c:30:9d:3f:5d:c7:24:3c:c0:
                    9f:d4:9c:e8:82:28:48:99:ce:48:f9:fa:d9:09:3c:
                    38:0b
                Exponent: 65537 (0x10001)
    Signature Algorithm: sha256WithRSAEncryption
         48:16:5c:28:ae:67:7b:54:09:a1:45:3d:dd:e6:70:53:a6:9f:
         b0:25:a9:fe:25:d8:8b:b4:6c:17:22:ba:8e:d2:a0:82:f7:80:
         70:7c:a2:f8:de:1a:fc:91:af:c1:eb:de:98:6c:d1:92:ed:a8:
         c0:88:08:2d:b2:17:9c:ce:04:0e:24:7f:71:c0:29:c5:60:a3:
         db:e4:81:6b:28:1f:0a:51:5a:31:10:51:aa:1f:d8:80:db:af:
         4d:fb:26:32:90:37:67:7d:56:32:2e:de:87:b2:4c:56:ff:46:
         b0:57:10:d4:25:26:2b:f9:2d:f4:7f:79:40:87:6b:cc:6c:ac:
         41:0e:9d:27:17:bd:62:bc:ba:f9:24:b2:52:a0:4d:ce:f1:63:
         ab:f3:3b:ff:4b:25:6f:cc:fb:74:79:7c:cb:ff:94:15:8a:9f:
         4b:18:06:33:fe:63:28:f4:32:03:3c:e3:27:64:b3:b6:0d:77:
         b5:a4:7a:58:b7:04:c8:69:24:e4:e2:c9:d2:a0:91:45:66:7d:
         31:c4:70:56:d4:cb:69:22:09:75:78:99:49:69:d4:bf:3c:7b:
         d3:67:e7:59:a7:de:3f:80:60:ed:13:25:1f:ad:27:7c:98:ee:
         1e:53:a3:50:f5:62:b9:f1:d8:0c:e3:12:2b:0e:3f:dd:5f:36:
         1d:1a:43:59
```

dgarske, 2021-01-26:
Certificate looks straightforward, but I noticed it is using SHA-1 for the hash. Might be worth making sure its enabled in the wolfSSL ./configure (--enable-sha). It is on by default.

dgarske, 2021-01-26:
Hi @hnyman ,

Looks like the signature algorithm is mis-matched.

Notice: `Signature Algorithm: sha1WithRSAEncryption` says SHA-1
The actual signature is SHA-256: `Signature Algorithm: sha256WithRSAEncryption`.

FYI: Here are the only two places in the code where the `ASN_SIG_OID_E` error occurs:
https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/asn.c#L9535
https://github.com/wolfSSL/wolfssl/blob/master/wolfcrypt/src/asn.c#L9775

Thanks,
David Garske, wolfSSL

elms, 2021-01-26:
@hnyman @PolynomialDivision 
I think this fixes the issue: https://github.com/elms/openwrt/commit/7c93544a5f7b0352984a7b196f5d5e971a9cb2fb

bagder, 2021-01-26:
Based on all the info provided here, it seems to me that wolfSSL correctly identify a problem in the certificate and that the proper fix is to make sure that the cert does not have that inconsistency. Anyone objects or can we then close this issue?

PolynomialDivision, 2021-01-26:
> @hnyman @PolynomialDivision
> I think this fixes the issue: [elms/openwrt@7c93544](https://github.com/elms/openwrt/commit/7c93544a5f7b0352984a7b196f5d5e971a9cb2fb)

Thanks a lot! Can confirm that it fixes the issue.

elms, 2021-01-27:
@PolynomialDivision closing this issue then. Will follow up on https://github.com/openwrt/openwrt/pull/3813

closed 2021-01-27 by elms