wolfSSL/wolfssl #3692: wolfSSL accepts a cert whose basicConstraints.cA==False and keyUsage.keyCertSign is set

GOODPWDCETCSZ, 2021-01-26:
wolfSSL accepts a cert whose basicConstraints.cA==False and keyUsage.keyCertSign is set. According to RFC 5280, basicConstraints.cA indicates whether the public key can be used to verify cert. Here, cA is set False, the keyUsage must not be set. GnuTLS v3.6.13 and mbedTLS v2.25.0 reject the cert. 

**Env**: Ubuntu x64

**wolfSSL version**: 4.6.0

**Reproduce**: `./wolfssl_verify_pem ca.pem seed-5s26-78s27-136s35-369s39-372s19-955c21.pem`

**Expected result**: 

Consistent verification result between wolfSSL and RFC 5280+other TLS implementations.

**Actual result**:
```
./wolfssl_verify_pem ../certs_related/ca.pem ../certs/seed-5s26-78s27-136s35-369s39-372s19-955c21.pem
The target cert has passed through verification.
```

**Attachment**

[ca.zip](https://github.com/wolfSSL/wolfssl/files/5873593/ca.zip)
[seed-5s26-78s27-136s35-369s39-372s19-955c21.zip](https://github.com/wolfSSL/wolfssl/files/5873595/seed-5s26-78s27-136s35-369s39-372s19-955c21.zip)

``` # wolfssl_verify_pem.c
// gcc -g wolfssl_verify_pem.c -o wolfssl_verify_pem -lwolfssl
#include <stdlib.h>
#include <wolfssl/ssl.h>
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/error-crypt.h>

int wolfSSL_Verify_PEM(char * cac, char * ec){
	int ret = 0;

	// to create a new wolfSSL cert manager
	WOLFSSL_CERT_MANAGER* cm;
	cm = wolfSSL_CertManagerNew();
	if (cm == NULL){
		printf("Creating a new wolfSSL_CertManager failed!\n");
		exit(1);
	}
	
	// to load cac to the created wolfSSL_CertManager
	ret = wolfSSL_CertManagerLoadCA(cm, cac, NULL);
	if (ret != SSL_SUCCESS){
		printf("Loading cac to the created wolfSSL_CertManager failed!\n");
		exit(2);
	}

	// to verify the ec in the created wolfSSL_CertManager
	ret = wolfSSL_CertManagerVerify(cm, ec, SSL_FILETYPE_PEM);
	if (ret != SSL_SUCCESS){
		printf("wolfSSL_CertManagerVerify filed and with return code %d and error message %s\n",
			 ret,
			 wolfSSL_ERR_reason_error_string(ret));
	}
	else{
		printf("The target cert has passed through verification.\n");
	}

	// to free cm
	wolfSSL_CertManagerFree(cm);

	return ret;
}

int main(int argc, char ** argv){
	char * cac = argv[1];
	char * ec = argv[2];
	wolfSSL_Verify_PEM(cac, ec);
	return 0;
}
```

embhorn, 2021-01-26:
Hi @GOODPWDCETCSZ 

Could you please the configuration you used with wolfSSL?

GOODPWDCETCSZ, 2021-01-27:
Hi @embhorn  wolfSSL is configured and built by default.
```
 ./configure
  make
  sudo make install
  ./testsuite/testsuite.test
```

embhorn, 2021-01-27:
Hello @GOODPWDCETCSZ 

I was able to reproduce. Thanks for the excellent instructions.

GOODPWDCETCSZ, 2021-01-28:
@embhorn You are vey welcome.