wolfSSL/wolfssl #3694: wolfSSL accepts a cert with non-rfc822name email

GOODPWDCETCSZ, 2021-01-27:
wolfSSL accepts a cert with non-rfc822name email.

**Env**: Ubuntu x64

**wolfSSL version**: 4.6.0

**Reproduce**: `./wolfssl_verify_pem ca.pem seed-4s18-12s35-569s26-398c21.pem`

**Expected result**: 

The cert with non-rfc822name email is rejected.

**Actual result**:
```
./wolfssl_verify_pem ../certs_related/ca.pem ../certs/seed-4s18-12s35-569s26-398c21.pem
The target cert has passed through verification.

```

**Attachment**

[ca.zip](https://github.com/wolfSSL/wolfssl/files/5879108/ca.zip)
[seed-4s18-12s35-569s26-398c21.zip](https://github.com/wolfSSL/wolfssl/files/5879109/seed-4s18-12s35-569s26-398c21.zip)

[wolfssl_verify_pem.zip](https://github.com/wolfSSL/wolfssl/files/5879107/wolfssl_verify_pem.zip)

kaleb-himes, 2021-01-27:
@GOODPWDCETCSZ,

1) I can confirm that yes wolfSSL approves the cert in question (seed-4s18-12s35-569s26-398c21.pem). I noticed you were not including <wolfssl/options.h> before all other wolfSSL headers, I would caution against that as it can lead to undefined behavior. Please be sure to always include <wolfssl/options.h> in every application file to avoid undefined behavior.

2) I ran the cert through OpenSSL as a sanity check, OpenSSL also accepts it without any issue.

But just because the sanity check accepted it doesn't mean it is correct. I am not aware of any certificate spec other than RFC 5280 that we abide by so I did go look up RFC 822 and found it is obsoleted by RFC 2822 (https://tools.ietf.org/html/rfc2822). Can you point to the section which this certificate violates so our team can evaluate it?

Warm Regards,

K

GOODPWDCETCSZ, 2021-01-28:
@kaleb-himes 

In RFC 5280, the format of rfc822name is local-part@domain.
> The format of an rfc822Name is a "Mailbox" as defined in Section 4.1.2 of [RFC2821]. A Mailbox has the form "Local-part@Domain". (https://tools.ietf.org/html/rfc5280, Page 36) 

In RFC 2821, the mailbox format is specified as follow.
> Mailbox = Local-part "@" Domain
      Local-part = Dot-string / Quoted-string
           ; MAY be case-sensitive

(https://tools.ietf.org/html/rfc2821#page-36)

In RFC 5280, freshestCRL has the same syntax as cRLDistributionPoints.
> The same syntax is used for this extension and the cRLDistributionPoints extension, and is described in Section 4.2.1.13. 
 (https://tools.ietf.org/html/rfc5280, Page 48) 

In RFC 5280, cRLIssuer in freshestCRL has the type GeneralNames.
> DistributionPoint ::= SEQUENCE {
        distributionPoint       [0]     DistributionPointName OPTIONAL,
        reasons                 [1]     ReasonFlags OPTIONAL,
        cRLIssuer               [2]     **GeneralNames** OPTIONAL }

 (https://tools.ietf.org/html/rfc5280, Page 47) 

In RFC 5280, GeneralNames has rfc822Name.
> GeneralNames ::= SEQUENCE SIZE (1..MAX) OF GeneralName
GeneralName ::= CHOICE {
     otherName                 [0]  AnotherName,
     rfc822Name                [1]  IA5String,
     dNSName                   [2]  IA5String,
     x400Address               [3]  ORAddress,
     directoryName             [4]  Name,
     ediPartyName              [5]  EDIPartyName,
     uniformResourceIdentifier [6]  IA5String,
     iPAddress                 [7]  OCTET STRING,
     registeredID              [8]  OBJECT IDENTIFIER }

 (https://tools.ietf.org/html/rfc5280, Pages 127-128)

mcr, 2021-04-24:
I don't understand the complaint.  
1. My reading of the wolfSSL_CertManagerVerify() does not say that it validates any of the content of the certificate. It validates that the signature in the certificate is correct.  
2. I looked at certificate (seed...) and I don't see a SubjectDN. (it is allowed to be missing).  The only place I see a DN with an rfc822Name is in the CRL: email:g@h.com, and  the second CRL Issuer,  email:i.  

I don't expect CertManagerVerify() to validate this.  The ASN1 says it is an IA5String. While we expect to see an @ there, if something was going to validate it was correct, it should be the CA that signed it, not the end point.

kaleb-himes, 2021-04-27:
After reviewing this with the team the only RFC section that explicitly states an Implementation must reject a cert after applying RFC822Name rules is in reference to the email address that appears in the "subject distinguished name".

https://tools.ietf.org/html/rfc5280#section-4.2.1.10

```
... the rfc822Name constraint MUST be applied to the attribute of type emailAddress in the subject distinguished name.
```

We are happy to dive further into this discussion as it pertains to the `Freshest CRL:` if warranted but for the time being we are considering this issue as not pertaining to the implemention but rather to the CAs' issuing the certificates. If a deeper discussion is needed please email support [at] wolfssl [dot] com to speak directly with our team further regarding this matter.

- KH

closed 2021-04-27 by kaleb-himes