wolfSSL/wolfssl #3695: Compilation failure with --enable-keygen --enable-sp --enable-sp-math

guidovranken, 2021-01-27:
```sh
./configure --enable-static --enable-keygen --enable-sp --enable-sp-math && make
```

Output:

```
wolfcrypt/src/sp_int.c: In function ‘sp_mul_d’:
wolfcrypt/src/sp_int.c:3081:9: error: implicit declaration of function ‘_sp_mul_d’; did you mean ‘sp_mul_d’? [-Werror=implicit-function-declaration]
         _sp_mul_d(a, d, r, 0);
```

guidovranken, 2021-01-27:
Suggested fix:

```diff
diff --git a/wolfcrypt/src/sp_int.c b/wolfcrypt/src/sp_int.c
index 30d3d01..48e29c2 100644
--- a/wolfcrypt/src/sp_int.c
+++ b/wolfcrypt/src/sp_int.c
@@ -3019,7 +3019,8 @@ int sp_sub_d(sp_int* a, sp_int_digit d, sp_int* r)
 
 #if defined(WOLFSSL_SP_MATH_ALL) || defined(WOLFSSL_SP_SMALL) && \
     (!defined(NO_DH) || defined(HAVE_ECC) || \
-    (!defined(NO_RSA) && !defined(WOLFSSL_RSA_VERIFY_ONLY)))
+    (!defined(NO_RSA) && !defined(WOLFSSL_RSA_VERIFY_ONLY))) || \
+    (defined(WOLFSSL_KEY_GEN) && !defined(NO_RSA))
 /* Multiply a by digit n and put result into r shifting up o digits.
  *   r = (a * n) << (o * SP_WORD_SIZE)
  *
```

SparkiDev, 2021-01-27:
This has been fixed in https://github.com/wolfSSL/wolfssl/pull/3684

closed 2021-01-27 by SparkiDev