wolfSSL/wolfssl #3709: Handling of security issues and stable releases

ynezz, 2021-02-02:
Would it be pretty please possible to start referencing CVEs in either commits/pull requests or release changelogs? For example there is now CVE-2020-36177 which has CVSS score of 9.8, but I'm not able to find it in either. Your list of [security vulnerabilities](https://www.wolfssl.com/docs/security-vulnerabilities/) is missing it as well.

[image]

OpenWrt for example still uses vulnerable 4.5.0 release in the latest stable release, because we weren't aware, that there is such issue in that version. We just got notified about this by @lechner yestreday on the #openwrt-devel IRC channel. We take security seriously at OpenWrt so if we're aware about the issue, then it's usually fixed in matter of hours (depends on severity of the issue).

The amount of changes between 4.5.0 and 4.6.0 is quite large and possibly might be quite risky for OpenWrt as it's running on about 70 different HW targets which in the end boils down to 31 different CPU architectures. I'm wondering if it would be somehow possible to have some kind of security backports for the previous stable releases, which would be easier to review, less risky etc.

Thanks!

kaleb-himes, 2021-02-02:
@ynezz,

The lack of reference to CVE numbers is intentional as it pertains to commits/pull requests as it is often the case that a fix is up BEFORE the CVE is announced or the report is made public. We can not include that sort of information in the pull request or commit less we expose customers to zero-day exploits before we have an opportunity to get them a fix ahead of the announcement. This practice will not change.

I will certainly bring up the request for better clarification in the changelog and on the website but in this particular case there was an unfortunate early release of public information on CVE-2020-24613 ahead of the agreed upon schedule. 

wolfSSL Inc. was coordinating with the researchers but either due to in-experience with proper reporting procedure on the researchers behalf or possibly a mistake on behalf of the CVE personnel, information was made public ahead of the agreed upon date between wolfSSL Inc. and the researcher (also the CVE submitter). The release containing the fix is in progress and will be available to our customers very shortly and the release notes will contain information about the issue and our website will be updated once the release goes out.

- K

kaleb-himes, 2021-02-02:
@ynezz,

One item: "I'm wondering if it would be somehow possible to have some kind of security backports for the previous stable releases, which would be easier to review, less risky etc."

I will ask but this is not something we typically due as all current versions of wolfSSL strive to remain backwards compatible with all previous versions. I will bring it up as well in our next meeting.

- K

closed 2021-02-02 by kaleb-himes

ynezz, 2021-02-08:
> wolfSSL strive to remain backwards compatible with all previous versions. I will bring it up as well in our next meeting.

Just some material for that meeting:

* https://abi-laboratory.pro/index.php?view=compat_report&l=wolfssl&v1=4.5.0&v2=4.6.0&obj=0a67c&kind=abi
* https://lists.infradead.org/pipermail/openwrt-devel/2021-February/033678.html

ynezz, 2022-02-22:
Note to myself about API breakage between 4.8.1 and 5.1.1 update https://github.com/wolfSSL/wolfssl/issues/4879

ynezz, 2022-09-29:
Note to myself about [missing CVE-2022-39173](https://github.com/wolfSSL/wolfssl/issues/5629#issuecomment-1261798226) number for server side RCE fixed in 5.5.1:

[image]

ynezz, 2022-09-29:
> wolfSSL strive to remain backwards compatible with all previous versions. I will bring it up as well in our next meeting.

Note to myself, but might be of interest to @kaleb-himes  as well.

### ABI compatibility v5.4.0 to v5.5.0 ([full results](http://ynezz.true.cz/abi/libwolfssl/5.4.0_to_5.5.0/compat_report.html))
[image]

### ABI compatibility v5.5.0 to v5.5.1 ([full results](http://ynezz.true.cz/abi/libwolfssl/5.5.0_to_5.5.1/compat_report.html))
[image]

kaleb-himes, 2022-09-29:
Thank you so much @ynezz. I have sent a report up the wolfSSL chain, that compatibility rating is less than ideal. I have asked what we can do to help make your team more successful and if management has any suggestions to make to that end. I was also briefed on the inquiry in ZD ticket 14920 related to your previous comment(s). Thank you!

dgarske, 2022-09-29:
> Note to myself about [missing CVE-2022-39173](https://github.com/wolfSSL/wolfssl/issues/5629#issuecomment-1261798226) number for server side RCE fixed in 5.5.1:
> 
> [image]
> 
> BTW it was strange enough, as I've received that CVE number in private email.

Hi @ynezz , we do not publish the CVE in the change log due to timing of the release and CVE's being made public. Instead we publish the details here once they are public: https://www.wolfssl.com/docs/security-vulnerabilities/
We have discussed internally about going back through the change log and adding CVE's in after being made public and we may start doing that.

dgarske, 2022-09-29:
> Note to myself, but might be of interest to @kaleb-himes as well.
>
> ### ABI compatibility v5.4.0 to v5.5.0 ([full results](http://ynezz.true.cz/abi/libwolfssl/5.4.0_to_5.5.0/compat_report.html)) [...]

Hi @ynezz ,

Thanks for bringing this to our attention and hopefully I can clarify things. First we cannot guarantee a binary level compatibility for all API's. We do have API's marked with `WOLFSSL_ABI` and those are binary compatible. Meaning you can update wolfSSL and not have to rebuild your application if those API's are used. 

Also any API definitions marked with `WOLFSSL_API` won't change, but internal structures can and will change. If your application makes any dereferences of internal structures then things like sizeof() on those can change.

We would be happy to discuss how we can improve this for openWRT, just let me know and we can setup a call.

Thanks,
David Garske, wolfSSL

ynezz, 2022-09-30:
> Meaning you can update wolfSSL and not have to rebuild your application if those API's are used.

No, you can't do that, it was already proven several times, there are two real world examples of either ABI or API breakages, during previous wolfSSL version updates:

 * [ABI breakage 4.5.0 to 4.6.0 - hostapd/wpa_supplicant crashing due
to a stack overflow](https://github.com/wolfSSL/wolfssl/issues/3709#issuecomment-775090227)
 * [API breakage 4.8.1 to 5.1.1 - wolfSSL_get_verify_result()](https://github.com/wolfSSL/wolfssl/issues/3709#issuecomment-1047684253)

> If your application makes any dereferences of internal structures then things like sizeof() on those can change.

No, IMO this is not the case.

Upstream `hostapd` project is basic building block for wireless connectivity deployed on billions of devices, thus its very exposed and IMO very well tested and maintained, so I would be surprised if something nasty like that would be allowed there. Anyway, I might be wrong, you can [double check yourself](https://w1.fi/cgit/hostap/tree/src/crypto).

OpenWrt's own thin `ustream-ssl` wrapper around OpenSSL, wolfSSL and mbedTLS is IMO very simple as well, you can check [wolfSSL related bits quite easily](https://gitlab.com/search?search=wolfssl&nav_source=navbar&project_id=14926021&group_id=6335852&search_code=true&repository_ref=master).

> We would be happy to discuss how we can improve this for openWRT, just let me know and we can setup a call.

That's really amazing attitude and offer, thanks! IMO it would be better to discuss it openly, so anyone interested in wolfSSL could find this. In other words, it's not only about OpenWrt, any other project providing LTS releases and support could benefit from those changes.

> how we can improve this

The best improvement would be to keep ABI/API compatible between releases, but as we can see, it's very hard to achieve.

IMO the proper solution would be some form of LTS release/branch, with predictable lifetime support and maintenance commitment. Something like [mbedTLS](https://github.com/Mbed-TLS/mbedtls/blob/development/BRANCHES.md#long-time-support-branches) or [OpenSSL](https://www.openssl.org/policies/releasestrat.html) are for example doing.

Another option, but least feasible is to start properly making and documenting changes related to security fixes. Ideally do such changes in a single Git commit and mark that Git commit as such (ideally with CVE# attached), so anyone interested could spot those fixes and backport them. This might be hard to achieve due to information embargoes etc. and posses other risks, as those backports would be usually done by folks like me, so having no or very basic crypto background and thus such backports might possibly introduce more security issues then it's fixing, so in the essence it fails into "Don’t roll your own Crypto" category.

kaleb-himes, 2022-10-04:
@ynezz I was just talking to our team and management. It was suggested again that we propose a round-table call with you and any other developers that would like to join (https://openwrt.org/about#people). If you're concerned about the discussion being made public we would propose that both wolfSSL and OpenWRT team(s) take good notes on the call and publish the "minutes" here for public benefit. We feel we can make a lot of progress in a short amount of time if we can hear all the OpenWRT concerns and discuss them in a live forum as opposed to the back-and-forth of the github or the wolfSSL email support channel. We can start with either a 30 minute or 1-hour forum and if more time is needed schedule a followup.

Would you and your team be open to this proposal and if so when would be a good date/time? (We are headquartered in Seattle U.S. which is in the Pacific timezone (GMT - 7).

- Kaleb and the wolfSSL team

ynezz, 2022-10-10:
> Would you and your team be open to this proposal and if so when would be a good date/time

Ideally during a week around 18:00 UTC, which is 20:00 for most of us and if I count correctly 11:00 for you. Thanks!

dgarske, 2022-10-10:
Hi @ynezz , that sounds great. I asked Martin Engstrom (your regional business manager) to reach out and schedule a time. 
Looking forward to it! Thanks, David Garske, wolfSSL