wolfSSL/wolfssl #3737: Cannot enable EC compressed keys with CRYPTOCELL

yorickbrunet, 2021-02-08:
Hello,

I would like to use compressed Elliptic Curves keys with the CRYPTOCELL environment (nRF52840). For that I defined `HAVE_COMP_KEY` in *user_settings.h*. I'm using wolfSSL `v4.6.0-stable`.

However, when compiling the example using Segger Embedded Studion, I get the following errors:

```sh
Rebuilding ‘wolfcrypt_pca10056’ from solution ‘wolfcrypt_pca10056’ in configuration ‘Release’
1> Assembling ‘thumb_crt0.s’
2> Compiling ‘nrf_log_backend_serial.c’
...
4> Compiling ‘integration_test_plat_utils.c’
3> Assembling ‘ses_startup_nrf52840.s’
1> Assembling ‘ses_startup_nrf_common.s’
2> Compiling ‘system_nrf52840.c’
1> Generating linker script ‘wolfcrypt_pca10056.ld’
1> Linking wolfcrypt_pca10056.elf
1> Output/Release/Obj/wolfcrypt_pca10056/ecc.o: In function `wc_ecc_import_point_der_ex':
1> C:\work\test_wolfssl\nRF5_SDK_nos_ybr\external\wolfssl\wolfcrypt\src/ecc.c:6897: undefined reference to `mp_sqrtmod_prime'
1> Output/Release/Obj/wolfcrypt_pca10056/ecc.o: In function `wc_ecc_import_x963_ex':
1> C:\work\test_wolfssl\nRF5_SDK_nos_ybr\external\wolfssl\wolfcrypt\src/ecc.c:7791: undefined reference to `mp_sqrtmod_prime'
Build failed
```

In fact, after looking at file *ecc.c*, I could see that function `mp_sqrtmod_prime` is defined when

```c
#ifdef HAVE_COMP_KEY
#if !defined(WOLFSSL_ATECC508A) && !defined(WOLFSSL_ATECC608A) && \
    !defined(WOLFSSL_CRYPTOCELL)
```

which means that it is not defined when using the CRYPTOCELL.

Is it done on purpose because it is not possible to exploit compressed EC keys with the CRYPTOCELL? Or is it a mistake?

Thanks.

Best regards,
Yorick

tmael, 2021-02-08:
Hi Yorick,

Unfortunately, compressed keys with Cryptocell is unsupported at the moment. If you would like this enhancement request implemented, please let me know so I can add it to our backlog.  

Thanks,
Tesfa

yorickbrunet, 2021-02-09:
Hi Tesfa,

Thanks for your quick answer.
It would be great if you could add this feature to your backlog. I think it makes sense to have shorter keys on embedded devices.

Best regards,
Yorick

tmael, 2021-02-12:
Closing since we've started tracking this feature request internally.

closed 2021-02-12 by tmael