wolfSSL/wolfssl #3773: wolfSSL failed to load private key of self-generated cert

sapi01, 2021-02-18:
Hi, I'm very new to wolfSSL, it's the most smooth experience I had with an SSL lib ever and thanks to `wolfssl-examples` :)
I want to use my own self-generated cert for my server, so I generated cert and private key with OpenSSL CLI util with
```
$ openssl req -x509 -newkey rsa:4096 -keyout key.pem -out cert.pem -days 365
```

and when I try to use these generated files on my implementation, I'm getting the following error
```
ERROR: Failed to load ./key.pem, please check the file.
```
from `wolfSSL_CTX_use_PrivateKey_file` function. I'm not sure what goes wrong, Can someone help me with this?

embhorn, 2021-02-18:
Thanks for your kind words. We also think wolfSSL is the greatest.
Hi @sapi01 

I found a solution for this, but not a reason for why it works. If I use openSSL to open the key and simply write it out again using this command:
`openssl rsa -inform pem -in server.key -outform pem -out server.key2`

The new key (server.key2) will work in wolfSSL.

The certs and keys used in the examples are generated with this script:
https://github.com/wolfSSL/wolfssl/blob … ts.sh#L176

Maybe breaking up the openssl commands makes a difference?

Thanks,
@embhorn  wolfSSL Support

elms, 2021-02-18:
Hi @sapi01,

By default openSSL encrypts the key with a passphrase (which is a reasonable default) and wolfSSL doesn't have the passphrase. To not protect the key file with a passphrase, the `-nodes` (that's no DES) can be used.

```
openssl req -x509 -nodes -newkey rsa:4096 -keyout key.pem -out cert.pem -days 365
```

Interestingly telling openssl to convert the RSA format as @embhorn showed, doesn't protect the output key.

Hope that helps.

elms, 2021-02-18:
To clarify wolfSSL does support passpharse protected keys.

In the main wolfSSL repo the server example supports this with a call to [SSL_CTX_set_default_passwd_cb](https://github.com/wolfSSL/wolfssl/blob/ad58478d299d9ef3e030a53352c3b78ebd34f113/examples/server/server.c#L1857) when configured with `--enable-enckeys`.

See the callback implementation [here](https://github.com/wolfSSL/wolfssl/blob/ad58478d299d9ef3e030a53352c3b78ebd34f113/wolfssl/test.h#L646-L659). `wolfSSL_CTX_set_default_passwd_cb_userdata` is also useful for supporting these keys.

sapi01, 2021-02-19:
Thank you so much!
Fixed: https://www.wolfssl.com/forums/post5788.html#p5788

closed 2021-02-19 by sapi01