wolfSSL/wolfssl #3790: openssl compatibility layer additions

RealEnder, 2021-02-20:
Tried to compile https://github.com/ZerBea/hcxtools and https://github.com/ZerBea/hcxdumptool against wolfssl through openssl compatibility layer. There are some unimplemented functions and consts:
For hcxtools:
`EVP_PKEY_new_CMAC_key()`

For hcxdumptool:
```
ERR_FATAL_ERROR
SSL_R_DECRYPTION_FAILED_OR_BAD_RECORD_MAC
SSL_OP_NO_SSLv2
SSL_OP_NO_SSLv3
EVP_PKEY_new_CMAC_key()
SSL_CTX_set_ecdh_auto()
```
Thanks!

embhorn, 2021-02-22:
Hello @RealEnder 

Thanks for the suggestion. We try to work on requested features whenever we get free engineering cycles unless a customer decides to back the effort for a finite implementation timeline. If at any time a feature becomes a progress blocker for a project it can be accelerated. That process can be started by contacting an account business manager.

Are you a developer / maintainer with the hcxtools project?

Kind regards,
Eric @ wolfSSL Support

RealEnder, 2021-02-23:
Hi Eric,
Thanks for the clarification.
I'm a contributor to these repos. The main developer of hcxtools suite is @zerbea.
Regards!

RealEnder, 2021-02-23:
BTW, my tests were with 4.5.0. I see in 4.7.0 at least `SSL_R_DECRYPTION_FAILED_OR_BAD_RECORD_MAC` defined, so I will test with HEAD and see what's left.

ZerBea, 2021-02-23:
Hi guys.
I am the developer of hcxdumptool, and hcxtools.
The whole suite is designed to run on small systems with less resources (e.g. Raspberry Pi, OpenWRT devices, Pineapple).
Purpose of this penetration testing tools is to discover weak points on WiFi devices and 802.11 protocol. Therefore it is mandatory to do an initial check on incoming frames, using some hash and/or crypto-functions. Currently OpenSSL is used for that purpose. Unfortunately the size of this lib is too big to be used on small devices and we have to find a small and fast replacement.
We are talking about an initial check on incoming frames (on the fly as the are received) to determine if the target is running an empty configuration file and the hash results are calculated by zeroed values.
Recovering the password from the hash file is done later on, off-line on big GPU machines, running hashcat or john.
https://hashcat.net/hashcat/
https://openwall.info/wiki/john

hcxdumptool / hcxpcapngtool is the WiFi pre-processor to prepare the data for hashcat and/or john.

Also it would be great to see wolfssl inside Arch Linux repositories:
https://archlinuxarm.org/packages
https://archlinux.org/

danielinux, 2021-02-23:
Hi, @ZerBea thanks for the insight.  Just a comment regarding archLinux, there is an AUR package that looks well maintained:
https://aur.archlinux.org/packages/wolfssl/ . Although I'm not sure about inclusion procedures for the official repositories, this might be a starting point.

Thanks,

-- 
Daniele

ZerBea, 2021-02-23:
Hi Daniele.
Thanks for the fast response and the information about AUR. Problem on AUR packages is that they have to be compiled by the user. Unfortunately, that will take some time on small devices.

BTW:
The project of RealEnder is also involved (fast and small crypto lib):
https://github.com/RealEnder/dwpa
Production version is running here:
https://wpa-sec.stanev.org/?nets

Cheers
Mike