wolfSSL/wolfssl #3811: SP math build fails

guidovranken, 2021-02-24:
This build configuration has been failing for the last 2 days.

Build:

```
export CFLAGS="$CFLAGS -DHAVE_AES_ECB -DWOLFSSL_DES_ECB -DHAVE_ECC_SECPR2 -DHAVE_ECC_SECPR3 -DWOLFSSL_ECDSA_SET_K -DWOLFSSL_ECDSA_SET_K_ONE_LOOP"
git clone --depth 1 https://github.com/wolfSSL/wolfssl.git
cd wolfssl/
autoreconf -ivf
./configure --enable-static --enable-md2 --enable-md4 --enable-ripemd --enable-blake2 --enable-blake2s --enable-pwdbased --enable-scrypt --enable-hkdf --enable-cmac --enable-arc4 --enable-camellia --enable-rabbit --enable-aesccm --enable-aesctr --enable-hc128 --enable-xts --enable-des3 --enable-idea --enable-x963kdf --enable-harden --enable-aescfb --enable-aesofb --enable-aeskeywrap --enable-keygen --enable-curve25519 --enable-curve448 --enable-shake256 --disable-crypttests --disable-examples --enable-compkey --enable-ed448 --enable-ed25519 --enable-xchacha --enable-cryptocb --enable-eccencrypt --enable-smallstack --enable-sp --enable-sp-math
make -j$(nproc)
```

Error message:

```
wolfcrypt/src/.libs/src_libwolfssl_la-rsa.o: In function `_CheckProbablePrime':
rsa.c:(.text+0x31c): undefined reference to `sp_gcd'
wolfcrypt/src/.libs/src_libwolfssl_la-rsa.o: In function `wc_MakeRsaKey':
rsa.c:(.text+0x36ef): undefined reference to `sp_gcd'
/usr/bin/ld: src/.libs/libwolfssl.so.24.4.1: hidden symbol `sp_gcd' isn't defined
/usr/bin/ld: final link failed: Bad value
```

elms, 2021-02-24:
Minimal reproducer `./configure --enable-sp --enable-sp-math --enable-keygen`. With `--enable-keygen` and `--enable-sp-math` I think we should report an error at config time that `--enable-sp-math-all` is required for keygen.

We could fix with 
```
diff --git a/wolfcrypt/src/sp_int.c b/wolfcrypt/src/sp_int.c
index 934037b97..9c047d228 100644
--- a/wolfcrypt/src/sp_int.c
+++ b/wolfcrypt/src/sp_int.c
@@ -13230,7 +13230,8 @@ int sp_prime_is_prime_ex(sp_int* a, int t, int* result, WC_RNG* rng)
 }
 #endif /* WOLFSSL_SP_MATH_ALL || WOLFSSL_HAVE_SP_DH */
 
-#if defined(WOLFSSL_SP_MATH_ALL) && !defined(NO_RSA) && defined(WOLFSSL_KEY_GEN)
+#if defined(WOLFSSL_SP_MATH_ALL) && !defined(NO_RSA) && defined(WOLFSSL_KEY_GEN) || \
+    defined(WOLFSSL_HAVE_SP_DH)
 
 /* Calculates the Greatest Common Denominator (GCD) of a and b into r.
  *
```
but `./configure --enable-sp --enable-sp-math --enable-keygen --disable-dh` will still fail

SparkiDev, 2021-02-25:
Thanks @guidovranken and @elms.

I have a pull request to fix this: https://github.com/wolfSSL/wolfssl/pull/3815

Let me know If there are any issues with the fix!

elms, 2021-02-26:
Closing with merge of  #3815

closed 2021-02-26 by elms