wolfSSL/wolfssl #3830: MP_NOT_INF not consistently defined

elagergren-spideroak, 2021-03-01:
While testing the SP library I noticed that both `integer.h` and `tfm.h` define `MP_NOT_INF`, but `sp_int.h` does not. Admittedly it's a very minor issue: I checked my copies of  v4.4.0 and v4.7.0 and can't find where it's used. I only noticed it because it broke a debugging routine that converts the `MP_*` result codes to strings.

SparkiDev, 2021-03-01:
Hi @elagergren-spideroak,

I've put up a pull request that defines all the errors in integer.h and tfm.h: https://github.com/wolfSSL/wolfssl/pull/3831
MP_NOT_INF is not used explicitly anymore and MP_BUFFER is not an error that will be seen in sp_int.c.

Let me know if this fixes your code.

elagergren-spideroak, 2021-03-01:
> Let me know if this fixes your code.

It does, thank you.

> MP_NOT_INF is not used explicitly anymore and MP_BUFFER is not an error that will be seen in sp_int.c.

Good to know!

SparkiDev, 2021-03-04:
Hi @elagergren-spideroak

The pull request fixes your issue and has been merged, so I'm closing the issue.

closed 2021-03-04 by SparkiDev