wolfSSL/wolfssl #3835: PKCS7 to X509 PEM

lijintv07, 2021-03-02:
is it possible to convert a PKCS7 to X509 PEM format certificate list?

embhorn, 2021-03-02:
Hi @lijintv07,

Yes! You'd need to import the PKCS7 cert, then loop through the list converting and printing the individual certs. We have some great examples of manipulating PKCS7:
https://github.com/wolfSSL/wolfssl-examples/tree/master/pkcs7

Thanks!

cconlon, 2021-03-16:
Hi @lijintv07,

The simple example in the following Pull Request on our wolfssl-examples repository shows how you can use wolfCrypt's PKCS#7 SignedData API to read a .p7b certificate file, extract the DER-encoded certificates, then convert them to PEM format.

https://github.com/wolfSSL/wolfssl-examples/pull/250

Best Regards,
Chris

cconlon, 2021-04-29:
Hi @lijintv07, I'm going to close this issue for now, but feel free to reopen if you have any issues with the example above.

Thanks,
Chris

closed 2021-04-29 by cconlon