wolfSSL/wolfssl #3836: sp_lcm not available anymore in SP math builds

guidovranken, 2021-03-02:
Compile wolfSSL using:

```sh
export CFLAGS="$CFLAGS -DHAVE_AES_ECB -DWOLFSSL_DES_ECB -DHAVE_ECC_SECPR2 -DHAVE_ECC_SECPR3 -DWOLFSSL_ECDSA_SET_K -DWOLFSSL_ECDSA_SET_K_ONE_LOOP"
git clone --depth 1 https://github.com/wolfSSL/wolfssl.git
cd wolfssl/
autoreconf -ivf
./configure --enable-static --enable-md2 --enable-md4 --enable-ripemd --enable-blake2 --enable-blake2s --enable-pwdbased --enable-scrypt --enable-hkdf --enable-cmac --enable-arc4 --enable-camellia --enable-rabbit --enable-aesccm --enable-aesctr --enable-hc128 --enable-xts --enable-des3 --enable-idea --enable-x963kdf --enable-harden --enable-aescfb --enable-aesofb --enable-aeskeywrap --enable-keygen --enable-curve25519 --enable-curve448 --enable-shake256 --disable-crypttests --disable-examples --enable-compkey --enable-ed448 --enable-ed25519 --enable-xchacha --enable-cryptocb --enable-eccencrypt --enable-smallstack --enable-sp --enable-sp-math
make -j$(nproc)
```

Then compile this program:

```c
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/integer.h>

int main(void)
{
    /* Dummy call to demonstrate the linking issue */
    sp_lcm(NULL, NULL, NULL);

    return 0;
}
```

Output:

```
/tmp/p-43e8ab.o: In function `main':
/home/jhg/wolf-sp-math-sp_lcm/wolfssl/p.c:7: undefined reference to `sp_lcm'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

elms, 2021-03-02:
@guidovranken I believe this is intentional. If you compile with `--enable-sp-math-all --enable-keygen --enable-aes` then `sp_lcm` will be included.

@SparkiDev  Can you confirm the intention?

guidovranken, 2021-03-02:
This used to be available in SP math until recently.

SparkiDev, 2021-03-02:
Hi Guido,

The --enable-sp-math option is used to include only as much of the maths routine as is necessary. Most of it happens in sp_c64.c and the like.
In this case, if sp_lcm() is not needed for building the library then the API won't be available.

I was able to compile with the above configuration line and, turning on the tests, resulted in testing easing.

OK?

Sean

guidovranken, 2021-03-03:
Ok, I've removed support for LCM with SP math builds in my fuzzer: https://github.com/guidovranken/cryptofuzz/commit/ebac51a6caa4502d0565a3c572d50d3fbb9cbe4b

closed 2021-03-03 by guidovranken