wolfSSL/wolfssl #3845: SP math rejects valid ECDSA signatures

elagergren-spideroak, 2021-03-05:
We test using [Project Wycheproof](https://github.com/google/wycheproof) test vectors (among others). After switching to SP math from TFM math, we see the following failures:

```
(valid) signature should have been accepted: r and s^-1 have a large Hamming weight (#638)
(valid) signature should have been accepted: r and s^-1 have a large Hamming weight (#639)
(valid) signature should have been accepted: 100-bit r and small s^-1 (#648)
(valid) signature should have been accepted: edge case for signature malleability (#655)
(valid) signature should have been accepted: edge case for signature malleability (#656)
(valid) signature should have been accepted: u2 == n - 1 (#660)
(valid) signature should have been accepted: extreme value for k and s^-1 (#695)
(valid) signature should have been accepted: extreme value for k and s^-1 (#698)
(valid) signature should have been accepted: extreme value for k and s^-1 (#701)
(valid) signature should have been accepted: extreme value for k and s^-1 (#704)
(valid) signature should have been accepted: point duplication during verification (#833)
```

(The numbers in parentheses are the test case IDs.)

I have attached a minimal repro of test case #638 below. Additionally, here's the same test case using the Go standard library: https://play.golang.org/p/2zLRzY0EBE7

```
#include "wolfssl/config.h"
#include "wolfssl/options.h"
#include "wolfssl/wolfcrypt/settings.h"

/* The rest of the headers can come in any order. */
#include <assert.h>
#include <stdarg.h>
#include <stdbool.h>
#include <stdio.h>
#include <stdlib.h>
#include <time.h>

#include "wolfssl/wolfcrypt/ecc.h"
#include "wolfssl/wolfcrypt/error-crypt.h"
#include "wolfssl/wolfcrypt/hash.h"
#include "wolfssl/wolfcrypt/integer.h"
#include "wolfssl/wolfcrypt/random.h"
#include "wolfssl/wolfcrypt/types.h"

__attribute__((format(printf, 1, 2))) static inline void panic(const char* fmt,
                                                               ...) {
    va_list args;
    va_start(args, fmt);
    vfprintf(stderr, fmt, args);
    va_end(args);
    puts("");
    exit(1);
}

int from_hex_char(char c) {
    if (c >= '0' && c <= '9') {
        return c - '0';
    }
    if (c >= 'a' && c <= 'z') {
        return c - 'a' + 10;
    }
    if (c >= 'A' && c <= 'Z') {
        return c - 'A' + 10;
    }
    return -1;
}

bool decode_hex(char* dst, const char* src) {
    int i = 0;
    int j = 1;
    while (src[j - 1] != 0) {
        int a = from_hex_char(src[j - 1]);
        if (a < 0) {
            return false;
        }
        int b = from_hex_char(src[j]);
        if (b < 0) {
            return false;
        }
        dst[i] = (a << 4) | b;
        i++;
        j += 2;
    }
    return true;
}

int main(void) {
    WC_RNG rng;
    int    ret = wc_InitRng(&rng);
    if (ret != 0) {
        panic("wc_RngInit");
    }

    const char hexsig[] =
        "304502207fffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff"
        "fd022100909135bdb6799286170f5ead2de4f6511453fe50914f3df2de54a36383df8d"
        "d4";

    word32 siglen = sizeof(hexsig) / 2;
    byte*  sig    = malloc(siglen);
    if (sig == NULL) {
        abort();
    }
    if (!decode_hex((char*)sig, hexsig)) {
        panic("decode_hex");
    }

    const char hexhash[] =
        "532eaabd9574880dbf76b9b8cc00832c20a6ec113d682299550d7a6e0f345e25";
    word32 hashlen = sizeof(hexhash) / 2;
    byte*  hash    = malloc(hashlen);
    if (hash == NULL) {
        abort();
    }

    const char hexqx[] =
        "8240cd81edd91cb6936133508c3915100e81f332c4545d41189b481196851378";
    byte* qx = malloc(32);
    if (qx == NULL) {
        abort();
    }
    if (!decode_hex((char*)qx, hexqx)) {
        panic("hexqx");
    }

    const char hexqy[] =
        "e05b06e72d4a1bff80ea5db514aa2f93ea6dd6d9c0ae27b7837dc432f9ce89d9";
    byte* qy = malloc(32);
    if (qy == NULL) {
        abort();
    }
    if (!decode_hex((char*)qy, hexqy)) {
        panic("hexqy");
    }

    ecc_key key;
    ret = wc_ecc_init(&key);
    if (ret != 0) {
        panic("wc_ecc_init");
    }
    ret = wc_ecc_import_unsigned(&key, qx, qy, NULL, ECC_SECP256R1);
    if (ret != 0) {
        panic("wc_ecc_import_raw");
    }

    int ok = 0;
    ret    = wc_ecc_verify_hash(sig, siglen, hash, hashlen, &ok, &key);
    if (ret != 0) {
        panic("wc_ecc_verify_hash");
    }
    if (ok == 1) {
        puts("pass");
    } else {
        puts("fail");
    }
}
```

elagergren-spideroak, 2021-03-06:
It appears `sp_ecc_verify_256` does not compute `s^-1 mod N`. If I insert the snippet below, my repro above works. Note that I have not yet tested this with anything other than my single test case. And I am also not particularly familiar the SP library.

```
sp_digit tmp [12];
sp_256_mont_inv_order_avx2_4(s, s, tmp);
```

elagergren-spideroak, 2021-03-06:
`w = 18446744073709551613 18446744073709551615 18446744073709551615 9223653511831486463` but I think it should be `w = 18446744073709551613 18446744073709551615 18446744073709551615 9223372036854775807`. I tentatively fixed it by using the constant-time Montgomery modular inversion, but because `w` isn't what I expected I think my "fix" might not be fixing the root cause (a bug in `sp_256_mod_inv_4`). But I haven't yet had a chance to test it.

SparkiDev, 2021-03-08:
Hi @elagergren-spideroak

The new modinv code was a little optimistic about the adding carries and underflows.
I've put up a pull request that fixes the ASM rather than not use it: https://github.com/wolfSSL/wolfssl/pull/3850

I will close the pull request in favour of the one I put up.

SparkiDev, 2021-03-11:
Hi @elagergren-spideroak

The pull request has been merged and you have indicated the fix works for you.
I will now close the ticket.
Please re-open if there is anything else.

closed 2021-03-11 by SparkiDev

elagergren-spideroak, 2021-03-12:
@SparkiDev thanks. I've a quick question: this will probably be slated for the 4.8.0 release, right? Does wolfSSL follow a particular release cycle? I wasn't able to find that information.

dgarske, 2021-03-12:
Hi @elagergren-spideroak ,

Yes anything merged to master will be in the next release. We usually do quarterly release cycles. The next one isn't scheduled until early June, however that can be accelerated if there is a feature or security issue that merits a release.

Thanks,
David Garske, wolfSSL

elagergren-spideroak, 2021-03-12:
@dgarske thanks! Appreciate it.