wolfSSL/wolfssl #3854: error: 'ECC_MIN_KEY_SZ' was not declared in this scope in line 132 of ecc.h

DrewGlinsman, 2021-03-08:
My compiler recently started throwing this error when trying to compile wolfssl. The file located in wolfssl/wolfssl/wolfcrypt/ecc.h sets the ECC_MINSIZE to ECC_MIN_KEY_SZ/8 but from what I can see this ECC_MIN_KEY_SZ is never declared before this in the file.

I assume I must've done something wrong with the configuration setup to get this error here was the following flags I added to the config script while running initial setup for the wolfssl repository. 

./configure --enable-ipv6 --enable-blake2 --enable-blake2s --enable-keygen --enable-certgen --enable-certreq --enable-ed25519 --enable-eccencrypt --enable-fips && make && sudo make install && mkdir build && cp /usr/local/lib/libwolfssl*

embhorn, 2021-03-08:
Hello @DrewGlinsman 

The FIPS build option is not supported in the open source version of wolfSSL. In fact you should be seeing a config error like:
`configure: error: non-FIPS-v1 source tree is incompatible with --enable-fips=yes`

Please send an email facts@wolfssl.com for more information about our FIPS certified crypto library.

If you do in fact have the FIPS version, please send an email to support@wolfssl.com where we can better address your question.

Thanks,
Eric @ wolfSSL Support

DrewGlinsman, 2021-03-09:
Yea I had missed that issue while setting it up, removing that flag fixed it for me thanks.

closed 2021-03-09 by DrewGlinsman