wolfSSL/wolfssl #3873: Crash when doing wolfSSL_free() on WOLFSSL object created with wolfSSL_write_dup()

rvlad-patrascu, 2021-03-12:
Hello,

I'm getting a segfault when trying to free with `wolfSSL_free()`, a write duplicate WOLFSSL object created with `wolfSSL_write_dup()`. Concurrent reading and writing works well, but as soon as the object is freed my application crashes inside wolfssl:
```
Program terminated with signal SIGSEGV, Segmentation fault.
#0  FreeSuites (ssl=0x7fce5213ff40) at src/internal.c:6343
6343	        wolfSSL_sk_SSL_CIPHER_free(ssl->suites->stack);
(gdb) bt
#0  FreeSuites (ssl=0x7fce5213ff40) at src/internal.c:6343
#1  0x00007fce4f74c239 in SSL_ResourceFree (ssl=0x7fce5213ff40) at src/internal.c:6367
#2  0x00007fce4f74cff1 in FreeSSL (ssl=0x7fce5213ff40, heap=0x0) at src/internal.c:6848
#3  0x00007fce4f78970b in wolfSSL_free (ssl=0x7fce5213ff40) at src/ssl.c:566
...
(gdb) p ssl->suites
$13 = (Suites *) 0x0
```
I have to mention that this happens regardless of setting beforehand the cipher suite for the underlying WOLFSSL_CTX with `wolfSSL_CTX_set_cipher_list()` or for the read / write duplicate WOLFSSL objects with `wolfSSL_set_cipher_list()`. But, if wolfssl is compiled _without_ the OPENSSL compatiblity layer, the crash does not happen (as in that case, `FreeSuites()` just does an `XFREE` on a NULL pointer).

I have tested this with `v4.7.0-stable` and `v4.3.0-stable` versions.

dgarske, 2021-03-12:
Hi @rvlad-patrascu ,

Thank you for this report. We will investigate and see if something happened to break this.

Have you seen the example use of this here?
https://github.com/wolfSSL/wolfssl-examples/blob/master/tls/client-tls-writedup.c

Thanks,
David Garske, wolfSSL

rvlad-patrascu, 2021-03-15:
Hi @dgarske 

Yes, I did. And after you mentioned the example, I've tried to run it too and it crashes the same way (the backtrace looks the same as the one posted above).

dgarske, 2021-03-15:
Hi @rvlad-patrascu ,

Thank you for the feedback. I've assigned this to @kabuobeid to investigate.

Thanks,
David Garske, wolfSSL

kabuobeid, 2021-03-15:
Hi @rvlad-patrascu ,

Thank you for your report, I was able to reproduce this issue in 4.7.0.  I believe I've found and fixed the issue in https://github.com/wolfSSL/wolfssl/pull/3876, can you give this PR a try and let me know if it fixes the issue for you?

Thanks,
Kareem

dgarske, 2021-03-16:
Reopening. End user should comment and mark closed if issue is resolved. 

@kabuobeid please avoid using `Fixes #` in commit comments or PR messages so merging does not auto-close GitHub issue reports.

rvlad-patrascu, 2021-03-18:
Hi @kabuobeid ,

I can confirm the fix, thank you!

commit e668b9b5 (2021-03-16): Merge pull request #3876 from kabuobeid/sslSuitesNullCheck

Fix missing NULL check in FreeSuites().  Fixes #3873.
closed 2021-03-18 by rvlad-patrascu