wolfSSL/wolfssl #3909: -fsanitize=undefined reports __attribute__((nonnull)) violations

dr-m, 2021-03-24:
Compiling MariaDB Server 10.4 or later with `cmake -DWITH_SSL=bundled -DWITH_UBSAN=ON` revealed that WolfSSL is invoking `XMEMCPY()` with a pointer parameter that is null.

You should be aware that invoking functions like `memcpy()`, `memmove()`, `memcmp()` with a null pointer is undefined behavior, even when the length argument is 0. Such code allows the compiler to infer that the pointer arguments are never null, and optimize away subsequent comparisons for the pointers being null. This actually happened for MariaDB Server in the past: the code would crash with `SIGSEGV` when compiled with GCC 8 and `-O2` until MariaDB/server@fc168c3a5e58d8b364a2e87e0d876a261ec7fced was fixed. The reason was that a later check for the pointer being null was optimized away, in some completely different place, with inline code expansion playing a role.

I would recommend you to try `-fsanitize=undefined` in GCC or `clang`, and search for `runtime error` in the output, or set an environment variable when running tests so that undefined behavior will lead to crashes.

I believe that the error that my colleague pointed out in his test build should be fixed by the following, but we did not test it:
```diff
diff --git a/src/ssl.c b/src/ssl.c
index 19d36795e..333338f66 100644
--- a/src/ssl.c
+++ b/src/ssl.c
@@ -41751,8 +41751,10 @@ err:
 
         objBuf[0] = ASN_OBJECT_ID; objSz++;
         objSz += SetLength(oidSz, objBuf + 1);
-        XMEMCPY(objBuf + objSz, oid, oidSz);
-        objSz     += oidSz;
+        if (oidSz) {
+            XMEMCPY(objBuf + objSz, oid, oidSz);
+            objSz += oidSz;
+        }
 
         if (obj->objSz == 0 || objSz != obj->objSz) {
             obj->objSz = objSz;
```

elms, 2021-04-20:
Ran a test case (with gcc) and forgot to post the results.

```
./configure CFLAGS=-fsanitize=undefined --enable-all
rm san.log.*
UBSAN_OPTIONS=log_path=san.log,print_stacktrace=1 make check
```

```
grep -h runtime san.log.* | awk '{print $1}' | sort | uniq -c  | wc -l
23
```

Dominated by `misaligned address` and case hit by far the most is the one @dr-m mentioned.

```
grep -h runtime san.log.* | grep -v "misaligned address" |awk '{print $1}' | sort | uniq
src/internal.c:9757:5:
src/ssl.c:41275:13:
src/ssl.c:41284:13:
src/ssl.c:43637:9:
wolfcrypt/src/eccsi.c:1479:21:
wolfcrypt/src/eccsi.c:2149:21:
wolfcrypt/src/evp.c:2356:13:
wolfcrypt/src/pkcs7.c:225:5:
wolfcrypt/src/pkcs7.c:2591:13:
```

dr-m, 2021-04-21:
Thank you, @elms. While x86 and recent implementations of other ISAs are generally tolerant of unaligned access, SIMD instructions are an exception: https://pzemtsov.github.io/2016/11/06/bug-story-alignment-on-x86.html

To help the compiler emit more efficient code, in MariaDB Server we introduced a `my_assume_aligned` template that is being used in a few wrappers of `<cstring>` functions:
```c++
#ifdef _MSC_VER
template <std::size_t Alignment, class T>
static inline T my_assume_aligned(T ptr)
{
  assert(reinterpret_cast<size_t>(ptr) % Alignment == 0);
  __assume(reinterpret_cast<size_t>(ptr) % Alignment == 0);
  return ptr;
}
#else
template <std::size_t Alignment, class T>
static inline T my_assume_aligned(T ptr)
{
  assert(reinterpret_cast<size_t>(ptr) % Alignment == 0);
  return static_cast<T>(__builtin_assume_aligned(ptr, Alignment));
}
#endif

template <std::size_t Alignment>
inline void *memcpy_aligned(void *dest, const void *src, size_t n)
{
  static_assert(Alignment && !(Alignment & (Alignment - 1)), "power of 2");

  return std::memcpy(my_assume_aligned<Alignment>(dest),
                     my_assume_aligned<Alignment>(src), n);
}
```

elms, 2021-07-01:
@dr-m #4070 has merged to address this. Thanks for this issue. I'm closing, but please reopen if you find any configuration that generate any reports.

closed 2021-07-01 by elms

dr-m, 2021-07-02:
@elms Thank you, I updated the WolfSSL submodule of MariaDB/server@0a67b15a9d3348d10f6df0caae6f2f973295a43a to 197b959916ff794c26201d6289574b4ac3e88256, and all tests (`cmake -DWITH_ASAN=ON -DWITH_UBSAN=ON -DWITH_SSL=bundled`) passed with flying colors. I filed #4178 for a minor compilation issue.