wolfSSL/wolfssl #3964: CMake: wolfcrypt/test/test_paths.h: No such file or directory

cxxboy, 2021-04-17:
When I try to add `vendor/wolfssl` onto my `add_subdirectory(vendor/wolfssl)` and try to build the project, I'm getting 
```
vendor/wolfssl/wolfcrypt/test/test.c:103:18: fatal error: wolfcrypt/test/test_paths.h: No such file or directory
  103 |         #include "wolfcrypt/test/test_paths.h"
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```
As of now, I just remove the `test_paths.h` from `wolfcrypt/test/test_paths.h` manually every time I want to use wolfSSL on my projects, but I would like to know why this error is happening and how I can effectively resolve this.

Environment: Ubuntu 20.04.2 LTS focal
CMake version: 3.16.3

Root CMakelists file:
```cmake
cmake_minimum_required(VERSION 3.10)
project(
	Blueth
	DESCRIPTION "..."
	LANGUAGES CXX
	)
set(CMAKE_CXX_STANDARD 20)
set(CMAKE_CXX_STANDARD_REQUIRED ON)
set(CMAKE_CXX_EXTENSIONS ON)
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)
add_subdirectory(vendor/wolfssl)
```
Command ran from `project_root/build`:  `cmake .. &&  make`

elms, 2021-04-17:
@cxxboy Are you using the the current master? Can you share the result of `git describe` within wolfssl?

This was a known issue but fixed with https://github.com/wolfSSL/wolfssl/pull/3777

The file should be generated ([CMakeLists](https://github.com/wolfSSL/wolfssl/blob/5955603c51caefc7ae6a7d7b229658f0617830de/CMakeLists.txt#L1334-L1335)) and can be checked for with `find . -name test_paths.h` (expected output:
`./vendor/wolfssl/wolfcrypt/test/test_paths.h`)

If you run `cmake .. --trace 2>&1 | grep test_path` you should see a line as below generating the `test_path.h`

```
/tmp/cmake_sub/vendor/wolfssl/CMakeLists.txt(1334):  configure_file(${CMAKE_CURRENT_SOURCE_DIR}/wolfcrypt/test/test_paths.h.in ${CMAKE_CURRENT_BINARY_DIR}/wolfcrypt/test/test_paths.h )
```

If it's still an issue, can you share the cmake_trace.log file after running `cmake .. --trace-redirect=cmake_trace.log`

cxxboy, 2021-04-18:
@elms I actually got the wolfSSL source code from `wolfSSL Release 4.7.0 (February 16, 2021)` on the Releases and placed it into the vendors folder. 

So I guess since the `vendor/wolfSSL` is not a git repo, I'm getting `fatal: No names found, cannot describe anything.` when I run `git describe`. 

And `cmake .. --trace 2>&1 | grep test_path` returns nothing, the result of the commend is empty. 

And [here](https://termbin.com/veux) is my cmake_trace.log

elms, 2021-04-19:
@cxxboy Sorry you hit this issue. This is a known issue and fixed in 4.7.1 so I'll mark this as closed. Please feel free to contact support@wolfssl.com if you need 4.7.1 (you can reference this issue).

closed 2021-04-19 by elms