wolfSSL/wolfssl #3971: atcab_verify_extern  in atmel.c casts int* to bool*, which does NOT conform to ANSI C.

thegoodhen, 2021-04-21:
The atmel_ecc_verify function casts the input argument int* verified to (bool*) verified when passing it into the atcab_verify_extern function.

Dereferencing this pointer causes a critical runtime failure on platforms where sizeof(bool)!=sizeof(int). 

Tested on ESP32, causes a memory access exception and reset.

dgarske, 2021-04-21:
Hi @thegoodhen , thanks for the report. We will take a look. Thanks, David Garske, wolfSSL

thegoodhen, 2021-04-21:
> Hi @thegoodhen , thanks for the report. We will take a look. Thanks, David Garske, wolfSSL

Wow, that was super fast! I should note that I am not absolutely sure whether some other libraries I have included haven't changed the definition of bool on my platform, but in either case, it caused problems and the code does not comply to ANSI. Thank you very much!

dgarske, 2021-04-23:
Hi @thegoodhen , I put up a fix in PR #3982 . Let me know if that resolves your runtime failure.

thegoodhen, 2021-04-27:
> Hi @thegoodhen , I put up a fix in PR #3982 . Let me know if that resolves your runtime failure.

Sadly, I am unable to confirm this. I am positive that the original implementation was incorrect, but unfortunately, it seems like this was not the reason why I experienced problems. When I tried fixing the issue in my code, it seemed to help, but then it failed later on... I don't know. Right now I am just trying to find at least some way to get the client TLS example with secure element to work, but I am struggling a lot, unfortunately.

closed 2021-06-14 by dgarske