wolfSSL/wolfssl #3984: why aren't doxygen comments not in production files

mcr, 2021-04-24:
I was trying to update the documentation for wc_SetAltNamesBuffer() [parameters and example are copy and pasted and are wrong], and I found the doxygen in doc/dox_comments/header_files/asn_public.h rather than in wolfssl/wolfcrypt/asn_public.h.
Is there a reason for this?  Are there some compilers, IDEs, etc? which can't cope with Doyxgen *comments*?

embhorn, 2021-04-26:
When we originally added the Doxygen docs, we didn't want to double the size of our existing headers by pulling all the additional doxygen comments in. That's what drove the duplication of headers into the `doc/dox_comments/header_files` location.

mcr, 2021-04-27:
> When we originally added the Doxygen docs, we didn't want to double the size of our existing headers by pulling all the additional doxygen comments in. That's what drove the duplication of headers into the `doc/dox_comments/header_files` location.

I'm from the 8-bit world and 20Mhz 68020 workstation world, and sure, header size mattered then.
Are there compilers where the size of the headers really matters more than having the documentation right next to the definition where it can be updated easily?

embhorn, 2021-12-17:
I am closing this issue, but please note that this concern was reviewed by the wolfSSL team.

closed 2021-12-17 by embhorn