wolfSSL/wolfssl #3988: Problems compiling the tls_client example on ESP-IDF on ATECC608A

thegoodhen, 2021-04-27:
Hello,
I have been trying to get the atecc support to work under PlatformIO for ESP32 for quite some time now, but only succeeded to some extent. I am getting random crashes that I can't really explain and therefore I decided to instead try a more standard platform.

I am trying to get the wolfssl_client example to work with ATECC608A, but I am struggling heavily. 

First, I followed the setup here: 
https://github.com/wolfSSL/wolfssl/blob/master/IDE/Espressif/ESP-IDF/README_32se.md

This links to a special version of cryptoauthlib, which I attempted to install, as described here:

https://github.com/miyazakh/cryptoauthlib_esp_idf

(I copied the library files as described)

I also edited
(...)\esp\esp-idf\components\wolfssl\wolfssl\wolfcrypt\settings.h 

to say
`
 #define WOLFSSL_ESPIDF 
(...)
#define WOLFSSL_ESPWROOM32SE
`

The build process then fails, complaining about missing CMakeLists.txt in cryptoauthlib directory, skipping the actual components, then failing on a missing include of <cryptoauthlib.h>....

I found a project that managed to integrate cryptoauthlib with esp-idf, though! Hooray! I compiled it succesfully (Well, it didn't work, for some reason the I2C config is totally messed up, but I managed to fix it) and it started to communicate! Success!

https://github.com/PBearson/eccx08-with-esp32

So I took the library component from this project and just straight up copied it into 

esp-idf/components/(cryptoauthlib, cryptoauthlib_hal, cryptoauthlib_basic).

Well, I restarted my IDE and now the compilation fails on something else! 

`
Executing action: all (aliases: build)
Running ninja in directory c:\users\thego\onedrive\dokumenty\kokon\wolfssl_client\build
Executing "ninja all"...
[1/95] Performing build step for 'bootloader'
ninja: no work to do.
[43/93] Building C object esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/ecc.c.obj
FAILED: esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/ecc.c.obj 
ccache C:\Users\thego\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -Iconfig -IC:/Users/thego/esp/esp-idf/components/wolfssl -IC:/Users/thego/esp/esp-idf/components/wolfssl/include -IC:/Users/thego/esp/esp-idf/components/freertos/include/freertos -IC:/Users/thego/esp/esp-idf/components/cryptoauthlib/lib -IC:/Users/thego/esp/esp-idf/components/newlib/platform_include -IC:/Users/thego/esp/esp-idf/components/freertos/include -IC:/Users/thego/esp/esp-idf/components/freertos/xtensa/include -IC:/Users/thego/esp/esp-idf/components/heap/include -IC:/Users/thego/esp/esp-idf/components/log/include -IC:/Users/thego/esp/esp-idf/components/lwip/include/apps -IC:/Users/thego/esp/esp-idf/components/lwip/include/apps/sntp -IC:/Users/thego/esp/esp-idf/components/lwip/lwip/src/include -IC:/Users/thego/esp/esp-idf/components/lwip/port/esp32/include -IC:/Users/thego/esp/esp-idf/components/lwip/port/esp32/include/arch -IC:/Users/thego/esp/esp-idf/components/soc/src/esp32/. -IC:/Users/thego/esp/esp-idf/components/soc/src/esp32/include -IC:/Users/thego/esp/esp-idf/components/soc/include -IC:/Users/thego/esp/esp-idf/components/esp_rom/include -IC:/Users/thego/esp/esp-idf/components/esp_common/include -IC:/Users/thego/esp/esp-idf/components/esp_system/include -IC:/Users/thego/esp/esp-idf/components/xtensa/include -IC:/Users/thego/esp/esp-idf/components/xtensa/esp32/include -IC:/Users/thego/esp/esp-idf/components/esp32/include -IC:/Users/thego/esp/esp-idf/components/driver/include -IC:/Users/thego/esp/esp-idf/components/driver/esp32/include -IC:/Users/thego/esp/esp-idf/components/esp_ringbuf/include -IC:/Users/thego/esp/esp-idf/components/efuse/include -IC:/Users/thego/esp/esp-idf/components/efuse/esp32/include -IC:/Users/thego/esp/esp-idf/components/espcoredump/include -IC:/Users/thego/esp/esp-idf/components/esp_timer/include -IC:/Users/thego/esp/esp-idf/components/esp_ipc/include -IC:/Users/thego/esp/esp-idf/components/soc/soc/esp32/include -IC:/Users/thego/esp/esp-idf/components/soc/soc/esp32/../include -IC:/Users/thego/esp/esp-idf/components/soc/soc/esp32/private_include -IC:/Users/thego/esp/esp-idf/components/vfs/include -IC:/Users/thego/esp/esp-idf/components/esp_wifi/include -IC:/Users/thego/esp/esp-idf/components/esp_wifi/esp32/include -IC:/Users/thego/esp/esp-idf/components/esp_event/include -IC:/Users/thego/esp/esp-idf/components/esp_netif/include -IC:/Users/thego/esp/esp-idf/components/esp_eth/include -IC:/Users/thego/esp/esp-idf/components/tcpip_adapter/include -IC:/Users/thego/esp/esp-idf/components/app_trace/include -mlongcalls -Wno-frame-address -DWOLFSSL_USER_SETTINGS   -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.2-dirty\" -DESP_PLATFORM -MD -MT esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/ecc.c.obj -MF esp-idf\wolfssl\CMakeFiles\__idf_wolfssl.dir\wolfcrypt\src\ecc.c.obj.d -o esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/ecc.c.obj   -c C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/ecc.c
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/ecc.c: In function 'ecc_projective_add_point_safe':
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/ecc.c:5923:19: error: implicit declaration of function 'ecc_projective_dbl_point'; did you mean 'ecc_projective_add_point_safe'? [-Werror=implicit-function-declaration]
             err = ecc_projective_dbl_point(B, R, a, modulus, mp);
                   ^~~~~~~~~~~~~~~~~~~~~~~~
                   ecc_projective_add_point_safe
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/ecc.c:5937:15: error: implicit declaration of function 'ecc_projective_add_point'; did you mean 'ecc_projective_add_point_safe'? [-Werror=implicit-function-declaration]
         err = ecc_projective_add_point(A, B, R, a, modulus, mp);
               ^~~~~~~~~~~~~~~~~~~~~~~~
               ecc_projective_add_point_safe
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/ecc.c: In function 'wc_ecc_import_raw_private':
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/ecc.c:8620:9: error: 'keySz' undeclared (first use in this function); did you mean 'key'?
         keySz = key->dp->size;
         ^~~~~
         key
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/ecc.c:8620:9: note: each undeclared identifier is reported only once for each function it appears in
cc1.exe: some warnings being treated as errors
[52/93] Building C object esp-idf/wolfssl/CMakeFiles/__idf_wolfssl.dir/wolfcrypt/src/pkcs12.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

`

I vaguely recall that I had to fix the keySz problem before, it was just a matter of a wrongly placed #ifdef , but... I mean, I sort of expected the examples to at least compile (if not run correctly) straight out of the box. Can you please please help me? I am a bit lost at this stage.

thegoodhen, 2021-04-27:
```
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/ecc.c:5923:19: error: implicit declaration of function 'ecc_projective_dbl_point'; did you mean 'ecc_projective_add_point_safe'? [-Werror=implicit-function-declaration]
             err = ecc_projective_dbl_point(B, R, a, modulus, mp);
                   ^~~~~~~~~~~~~~~~~~~~~~~~
```

I just checked and the function is defined on line 1973 IN THE SAME FILE?!! What is going on...

thegoodhen, 2021-04-27:
There are include guards in ecc.h to prevent the function prototypes from being generated; this suggests that the functions are not even used when ATECC is used... how curious!!

thegoodhen, 2021-04-27:
Okay, disabling the #ifdef guards in ecc.h "fixes" the first problem... But probably at a cost of compiling something that doesn't need to be compiled (?) . Now to the second problem!!

thegoodhen, 2021-04-27:
So apparently, in ecc.c on lines 8550 to 8554 there are include guards that prevent keySz from being defined when ATECC is used, even though the variable is still acessed in such a case? Oof. 

```
#if (defined(WOLFSSL_CRYPTOCELL) && !defined(WOLFSSL_ATECC508A) &&      \
     !defined(WOLFSSL_ATECC608A))  ||                                   \
  defined(WOLFSSL_SILABS_SE_ACCEL)
    word32 keySz = 0;
#endif
```
Okay, if I define the keySz no matter what the defines say, the compilation (drumroll) crashes on something else. Oh well.

thegoodhen, 2021-04-27:
@BrianAker  does anyone have a minimum example for ESP32 that would compile and actually work with the atecc608a?

BrianAker, 2021-04-27:
Sorry, I don’t have anything that specific. 

乌

embhorn, 2021-04-27:
@miyazakh Could you please review this report?

thegoodhen, 2021-04-28:
I navigated to esp-idf\components and git-pulled https://github.com/espressif/esp-cryptoauthlib

idf.py menuconfig gives me the options to configure esp-cryptoauthlib and it gets built (at least partially)

```
[663/1089] Building C object esp-idf/esp_cryptoauthlib/CMakeFiles/__idf_esp_cryptoauthlib.dir/cryptoauthlib/lib/atcacert/atcacert_date.c.obj
[664/1089] Building C object esp-idf/esp_cryptoauthlib/CMakeFiles/__idf_esp_cryptoauthlib.dir/cryptoauthlib/lib/atcacert/atcacert_def.c.obj
[665/1089] Building C object esp-idf/esp_cryptoauthlib/CMakeFiles/__idf_esp_cryptoauthlib.dir/cryptoauthlib/lib/atcacert/atcacert_der.c.obj
[666/1089] Building C object esp-idf/esp_cryptoauthlib/CMakeFiles/__idf_esp_cryptoauthlib.dir/cryptoauthlib/lib/atcacert/atcacert_client.c.obj
[667/1089] Building C object esp-idf/esp_cryptoauthlib/CMakeFiles/__idf_esp_cryptoauthlib.dir/cryptoauthlib/lib/atcacert/atcacert_host_sw.c.obj

```

But then the compilation fails on 

```

C:/Users/thego/esp/esp-idf/components/wolfssl/wolfssl/wolfcrypt/port/atmel/atmel.h:33:14: fatal error: cryptoauthlib.h: 
No such file or directory
     #include <cryptoauthlib.h>
              ^~~~~~~~~~~~~~~~~
compilation terminated.
```

thegoodhen, 2021-04-28:
idf.py menuconfig -> compatibility options -> include headers across components does NOT fix anything. Running idf.py fullclean before does not help either.

thegoodhen, 2021-04-28:
Okay, the reason for the last problem is that the wolfssl expects a different directory naming and structure; I added esp_cryptoauthlib as a requirement for wolfssl (which will likely break compilation of everything if you don't have cryptoauthlib installed :D ) 

`set(COMPONENT_REQUIRES lwip esp_cryptoauthlib)` (in CMakeLists.txt in the wolfssl directory.

Now it stopped complaining about missing cryptoauthlib.h and instead tells me the following:

```
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/port/atmel/atmel.c: In function 'atmel_ecc_create_pms':     
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/port/atmel/atmel.c:413:11: error: too few arguments to function 'atcab_ecdh_enc'
     ret = atcab_ecdh_enc(slotId, peerKey, pms, read_key, slotIdEnc);
           ^~~~~~~~~~~~~~
In file included from C:/Users/thego/esp/esp-idf/components/esp_cryptoauthlib/cryptoauthlib/lib/cryptoauthlib.h:39,     
                 from C:/Users/thego/esp/esp-idf/components/wolfssl/wolfssl/wolfcrypt/port/atmel/atmel.h:33,
                 from C:/Users/thego/esp/esp-idf/components/wolfssl/wolfssl/wolfcrypt/ecc.h:54,
                 from C:/Users/thego/esp/esp-idf/components/wolfssl/wolfssl/internal.h:103,
                 from C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/port/atmel/atmel.c:34:
C:/Users/thego/esp/esp-idf/components/esp_cryptoauthlib/cryptoauthlib/lib/basic/atca_basic.h:130:13: note: declared here ATCA_STATUS atcab_ecdh_enc(uint16_t key_id, const uint8_t* public_key, uint8_t* pms, const uint8_t* read_key, uint16_t 
read_key_id, const uint8_t num_in[NONCE_NUMIN_SIZE]);
             ^~~~~~~~~~~~~~
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/port/atmel/atmel.c: In function 'atcatls_set_certificates': 
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/port/atmel/atmel.c:936:14: error: implicit declaration of function 'tng_atcacert_read_signer_cert' [-Werror=implicit-function-declaration]
     status = tng_atcacert_read_signer_cert(&certBuffer[ATCATLS_TNGTLS_DEVICE_CERT_SIZE],
              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
C:/Users/thego/esp/esp-idf/components/wolfssl/wolfcrypt/src/port/atmel/atmel.c:951:14: error: implicit declaration of function 'tng_atcacert_read_device_cert' [-Werror=implicit-function-declaration]
     status = tng_atcacert_read_device_cert(certBuffer, &deviceCertSize,\
```

thegoodhen, 2021-04-28:
tng_atcacert_read_device_cert and tng_atcacert_read_signer_cert both exist inside (...)\esp\esp-idf\components\esp_cryptoauthlib\cryptoauthlib\app\tng\tng_atcacert_client.c

So I am not sure what the deal is there...

thegoodhen, 2021-04-28:
WOLFSSL_ATECC_TNGTLS is undefined in atmel.c , which is preventing "tng_atcacert_client.h" from getting included; defining it still doesn't help, since the path doesn't get resolved;

This change removes the errors related to missing definitions of the functions:

(in atmel.c)

```
//#ifdef WOLFSSL_ATECC_TNGTLS
//#include "tng/tng_atcacert_client.h"
//#endif

#include <tng_atcacert_client.h>
```

thegoodhen, 2021-04-28:
Okay, so adding the define into atca_basic.h as such:

```
#define ATCA_USE_CONSTANT_HOST_NONCE
#if defined(ATCA_USE_CONSTANT_HOST_NONCE)
ATCA_STATUS atcab_ecdh_enc(uint16_t key_id, const uint8_t* public_key, uint8_t* pms, const uint8_t* read_key, uint16_t read_key_id);
#else
ATCA_STATUS atcab_ecdh_enc(uint16_t key_id, const uint8_t* public_key, uint8_t* pms, const uint8_t* read_key, uint16_t read_key_id, const uint8_t num_in[NONCE_NUMIN_SIZE]);
#endif
```

Finally allows the project to compile, with a rather ominous warning stating

` warning: #warning "Using a constant host nonce with atcab_read_enc, atcab_write_enc, etcc., can allow spoofing of a device by replaying previously recorded messages" [-Wcpp]`

Probably not something I should just ignore, but hey, it compiles! I don't really have the courage to test if it runs, but hey, it compiles.

thegoodhen, 2021-04-28:
To anyone reading: I have taken desperate measures to get the darn thing to compile; the changes I have made allow that and only that; it is not recommended you follow what I did.

thegoodhen, 2021-04-28:
Even when I enable  #define WOLFSSL_ESPWROOM32 and undefine  #define WOLFSSL_ESPWROOM32SE in settings.h, I get a runtime error:

E (2708) tls_client: ERROR: failed to load -150, please check the file.

I tried enabling debug by defining -DDEBUG_WOLFSSL, but then the compilation crashes on an undefined reference to WOLFSSL_MSG...

...Okay, so if I add #define DEBUG_WOLFSSL to settings.h, it works for some reason...

So here is what it says.

(Probably irrelevant):
```

--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM4 instead...
--- idf_monitor on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
 � ���x��x�x�ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7100
load:0x40078000,len:13212
ho 0 tail 12 room 4
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (30) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (30) boot: compile time 14:10:26
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x285b0 (165296) map
I (176) esp_image: segment 1: paddr=0x000385d8 vaddr=0x3ffb0000 size=0x03820 ( 14368) load
I (183) esp_image: segment 2: paddr=0x0003be00 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/thego/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (184) esp_image: segment 3: paddr=0x0003c20c vaddr=0x40080404 size=0x03e0c ( 15884) load
I (199) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0xa93bc (693180) map
0x400d0020: _stext at ??:?

I (465) esp_image: segment 5: paddr=0x000e93e4 vaddr=0x40084210 size=0x113ac ( 70572) load
0x40084210: spi_flash_mmap at C:/Users/thego/esp/esp-idf/components/spi_flash/flash_mmap.c:153

I (508) boot: Loaded app from partition at offset 0x10000
I (508) boot: Disabling RNG early entropy source...
I (508) cpu_start: Pro cpu up.
I (512) cpu_start: Application information:
I (517) cpu_start: Project name:     wolfssl_client
I (522) cpu_start: App version:      1
I (527) cpu_start: Compile time:     Apr 28 2021 14:10:11
I (533) cpu_start: ELF file SHA256:  26c483f1872d723e...
I (539) cpu_start: ESP-IDF:          v4.2-dirty
I (544) cpu_start: Starting app cpu, entry point is 0x400818a8
0x400818a8: call_start_cpu1 at C:/Users/thego/esp/esp-idf/components/esp32/cpu_start.c:287

[... 157 lines cut ...]
I (21903) wolfssl:      General error
I (21913) wolfssl: wolfSSL error occurred, error = -308
I (21913) wolfssl: wolfSSL Leaving wolfSSL_negotiate, return -1
I (21923) wolfssl: wolfSSL Leaving SSL_write(), return -1
E (21933) tls_client: ERROR: failed to write

I (21933) wolfssl: wolfSSL Entering wolfSSL_read()
I (21943) wolfssl: wolfSSL Entering wolfSSL_read_internal()
I (21943) wolfssl: wolfSSL Entering ReceiveData()
I (21953) wolfssl: User calling wolfSSL_read in error state, not allowed
I (21963) wolfssl: wolfSSL Leaving wolfSSL_read_internal(), return -308
E (21963) tls_client: ERROR: failed to read

Server:I (21973) wolfssl: wolfSSL Entering SSL_free
I (21973) wolfssl: CTX ref count not 0 yet, no free
I (21983) wolfssl: Shrinking output buffer

I (21983) wolfssl: wolfSSL Leaving SSL_free, return 0
I (21993) wolfssl: wolfSSL Entering SSL_CTX_free
I (21993) wolfssl: CTX ref count down to 0, doing full free
I (22003) wolfssl: wolfSSL Entering wolfSSL_CertManagerFree
I (22013) wolfssl: wolfSSL Leaving SSL_CTX_free, return 0
I (22013) wolfssl: wolfSSL Entering wolfSSL_Cleanup
I (22023) wolfssl: wolfSSL Entering wolfCrypt_Cleanup
```

thegoodhen, 2021-04-28:
I think I give up for now, my objective was to get something that's proven to work and move on from there, which doesn't seem possible. Hit me up if you want me to perform additional tests to help you debug the problem. I will go try mbedTLS instead of wolfSSL and I'll see where that gets me.

miyazakh, 2021-04-29:
Hi @thegoodhen 

Thank you very much for reporting this. 

32se support seems relatively obsolete.  I verified those examples at the following conditions:
ESP-IDF : v3.3-beta1-39-g6cb37ecc5(commit hash : 6cb37ecc5)
CryptAuthLib: commit hash : c6b176e
and, wolfSSL v4.0.0. 
Unfortunately, I cannot find esp32 se board here. Once I find esp32se or set up sp32 plus atecc608A, I will update those examples and related code if needed.

Regarding "tls_client: ERROR: failed to load -150, please check the file.". the failure is because of expiration date. I have update the time in examples at [PR3994](https://github.com/wolfSSL/wolfssl/pull/3994).

thegoodhen, 2021-04-29:
> Hi @thegoodhen
>
> Thank you very much for reporting this.
>
> 32se support seems relatively obsolete. I verified those examples at the following conditions:
> ESP-IDF : v3.3-beta1-39-g6cb37ecc5(commit hash [...]

Aha! That makes a lot of sense.  Thank you for your prompt response. Should you need to install esp-idf, I shall warn you that version 4.0.2 (current version) of esp-idf plugin for Visual Code studio has a bug that makes it impossible to compile anything.
Maybe it can be fixed, but I just downgraded to 4.0.1 ...

I have also encountered a problem at one point, where the communication with ATECC didn't work at all. I fixed it by lowering the I2C frequency to 100kHz for waking up the device, then it was okay to communicate at much higher speeds. 
Sadly, I don't recall the conditions under which this happened, but it might be something to watch out for in case it doesn't work for you...

embhorn, 2022-08-23:
Closing this issue as resolved

closed 2022-08-23 by embhorn