wolfSSL/wolfssl #3989: Could you provide deb package for ubuntu 18.04 ?

hustliyilin, 2021-04-27:
I'm a user of ubuntu, could you provide the wolfssl deb package? Thanks a  lot.

dgarske, 2021-04-27:
@lechner, can you answer this?

These links might be useful:
https://packages.debian.org/search?keywords=wolfssl
https://repology.org/project/wolfssl/versions
https://tracker.debian.org/pkg/wolfssl

Thanks, David Garske, wolfSSL

hustliyilin, 2021-04-28:
> @lechner, can you answer this?
> 
> These links might be useful:
> https://packages.debian.org/search?keywords=wolfssl
> https://repology.org/project/wolfssl/versions
> https://tracker.debian.org/pkg/wolfssl
> 
> Thanks, David Garske, wolfSSL

Thank you for your quick reply. The information you provided is enough for me, thanks again.

closed 2021-04-28 by hustliyilin