wolfSSL/wolfssl #3998: MariaDB build is broken with recent wolfssl

vaintroub, 2021-04-30:
Suspect that WOLFSSL_USER_IO is broken . It had also been broken in 4.7.1r

commit 385e0bedaac935a1fc86766fbb2fcc47c9b82386 (HEAD -> master, origin/master, origin/HEAD)
Merge: edb0beb9b 73076940a
Author: David Garske <david@wolfssl.com>
Date:   Tue Apr 27 17:07:58 2021 -0700

============================== user_settings.h
#ifndef WOLFSSL_USER_SETTINGS_H
#define WOLFSSL_USER_SETTINGS_H

#define HAVE_CRL
#define WOLFSSL_MYSQL_COMPATIBLE
#define HAVE_ECC
#define ECC_TIMING_RESISTANT
#define HAVE_HASHDRBG
#define WOLFSSL_AES_DIRECT
#define WOLFSSL_SHA384
#define WOLFSSL_SHA512
#define WOLFSSL_SHA224
#define SESSION_CERT
#define KEEP_OUR_CERT
#define WOLFSSL_STATIC_RSA
#define WOLFSSL_USER_IO
#define WC_RSA_BLINDING
#define HAVE_TLS_EXTENSIONS
#define HAVE_AES_ECB
#define HAVE_AESGCM
#define WOLFSSL_AES_COUNTER
#define NO_WOLFSSL_STUB
#define OPENSSL_ALL
#undef WOLFSSL_ALLOW_TLSV10 /* see https://github.com/wolfSSL/wolfssl/issues/2960 */
#define NO_OLD_TIMEVAL_NAME
/*
  FP_MAX_BITS is set high solely to satisfy ssl_8k_key.test
  WolfSSL will use more stack space with it, with fastmath
*/
#define FP_MAX_BITS 16384

#define WOLFSSL_AESNI
#define USE_FAST_MATH
#define TFM_TIMING_RESISTANT
#define HAVE_INTEL_RDSEED
#define HAVE_INTEL_RDRAND
/* #undef USE_INTEL_SPEEDUP */
#define USE_FAST_MATH
#define WOLFSSL_X86_64_BUILD

#endif  /* WOLFSSL_USER_SETTINGS_H */
=======================================

Errors

C:\work\10.4\extra\wolfssl\wolfssl\src\wolfio.c(150,36): warning C4013: 'wolfSSL_LastError' undefined; assuming extern returning int [C:\work\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src\wolfio.c(151,43): error C2065: 'SOCKET_EWOULDBLOCK': undeclared identifier [C:\work\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src\wolfio.c(151,66): error C2065: 'SOCKET_EAGAIN': undeclared identifier [C:\work\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src\wolfio.c(155,46): error C2065: 'SOCKET_ECONNRESET': undeclared identifier [C:\work\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src\wolfio.c(159,41): error C2065: 'SOCKET_EINTR': undeclared identifier [C:\work\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src\wolfio.c(163,48): error C2065: 'SOCKET_ECONNABORTED': undeclared identifier [C:\work\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src\wolfio.c(207,43): error C2065: 'SOCKET_EWOULDBLOCK': undeclared identifier [C:\work\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src\wolfio.c(207,66): error C2065: 'SOCKET_EAGAIN': undeclared identifier [C:\work\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src\wolfio.c(211,46): error C2065: 'SOCKET_ECONNRESET': undeclared identifier [C:\work\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src\wolfio.c(215,41): error C2065: 'SOCKET_EINTR': undeclared identifier [C:\work\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src\wolfio.c(219,41): error C2065: 'SOCKET_EPIPE': undeclared identifier [C:\work\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src/bio.c(264,34): warning C4013: 'wolfIO_Recv' undefined; assuming extern returning int (compiling source file C:\work\10.4\extra\wolfssl\wolfssl\src\ssl.c) [C:\w
ork\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]
C:\work\10.4\extra\wolfssl\wolfssl\src/bio.c(638,34): warning C4013: 'wolfIO_Send' undefined; assuming extern returning int (compiling source file C:\work\10.4\extra\wolfssl\wolfssl\src\ssl.c) [C:\w
ork\10.4\xxx\extra\wolfssl\wolfssl.vcxproj]

dgarske, 2021-04-30:
Thanks @vaintroub for the report. I can reproduce. I believe this was broken in PR #3824 with a BIO layer refactor. I will push up a fix and reference this issue. David Garske, wolfSSL

dgarske, 2021-04-30:
Hi @vaintroub a fix has been pushed to #3999. Let me know if I missed anything. Thanks, David Garske, wolfSSL

closed 2021-06-14 by dgarske