wolfSSL/wolfssl #4004: Errors in compiling sp_cortexm.c  (__asm__ __volatile__)

MTKNTS, 2021-05-02:
I am going to use WolfCrypto to implement AES or CHACHA20 encryptions on STM32F407VGT6. I followed your guidance and add wolfcrypto patch to STM32CubeMX and after configuring, hit the Generate Code button. Now, when I try to compile it I get errors related to `__asm__ __volatile__`.
First error:
> sp_cortexm.c(2085): error:  #130: expected a "{"
>       __asm  __volatile__(
> 

```
Second error:
sp_cortexm.c(2107): error:  #18: expected a ")"
          : [c] "+r" (c), [r] "+r" (r), [a] "+r" (a), [b] "+r" (b)
```

By changing `__volatile__` to `volatile` the first error will be removed. However, the second one will remain.
I have no idea how to get rid of it. Please help me. Thanks in advance.

dgarske, 2021-05-02:
Please let us know your compiler and version. Thanks

MTKNTS, 2021-05-03:
Sorry to be late. I am using Keil uVision 5.25. Details have been provided in attached image.
[image]

dgarske, 2021-05-03:
Hi @MTKNTS ,

I recall we implemented a fix for this in PR https://github.com/wolfSSL/wolfssl/pull/2477/files

The top of sp_cortexm.c has a section for handling this:

```
#ifdef __KEIL__
 #define __asm__        __asm
 #define __volatile__   volatile
 #endif
```

Is that section not being included with that macro logic? What version of wolfSSL is being used? 

Thanks,
David Garske, wolfSSL

MTKNTS, 2021-05-05:
I used this file `wolfSSL.I-CUBE-wolfSSL.4.7.0.pack` to add wolfCrypt software pack to my STM32CubeMX. The version of wolfSSL employed by me is `4.7.0`. The first thing I did was to change `__asm__` to `__asm` and `__volatile__` to `volatile`. Nevertheless, as I mentioned before, only the first error would be gone.
Is there any advantage to use STM32CubeIDE rather than Keil when using wolfSSL softawre pack?
[image]

SparkiDev, 2021-05-06:
Hi @MTKNTS

Could I get you to try:
#define __volatile__
That is mapping it to nothing.

Thanks,
Sean

SparkiDev, 2021-05-06:
Hi @MTKNTS,

Could you please try removing the register list and see if that compiles? (That is the lines with ':')
Looking at the Keil manuals I can't see the register list as part of the syntax.
We have had it work with other versions of Keil.

MTKNTS, 2021-05-08:
I commented lines beginning with `:` then error changed to this:

> /sp_cortexm.c(2086): error:  #29: expected an expression

[image]

SparkiDev, 2021-05-09:
It was a bit of long shot actually but I don't think there is anything else that can be done here.
The compiler doesn't appear to handle a register list like other C compilers.
Looking at the documentation online, there are no examples I can find with the register list - they assume the use of variables in the asm directly.
In order to support Keil, at the version you are using, the assembly code would have to be rewritten.
While the assembly code is generated, it would still require a lot of work and would need to be paid for to have any priority.

Sorry I can't be of more help.
Sean

dgarske, 2021-05-10:
Hi @MTKNTS , 

We did get the SP Cortex-M assembly working with a more recent Keil compiler (v5.31). Is is possible you can upgrade Keil? It looks like they have release v5.34 now.

Thanks,
David Garske, wolfSSL

dgarske, 2021-06-14:
Hi @MTKNTS ,

Did you try updating your Keil compiler to see if that resolved the SP Cortex-M assembly errors? Should we consider this issue solved or is there more we can do to assist your efforts?

Thanks,
David Garske, wolfSSL

MTKNTS, 2021-06-15:
Hi @dgarske
After years of using Keil uVision, I switched to STM32CubeIDE and it worked fine.
Thank you.

dgarske, 2021-06-15:
Hi @MTKNTS ,

Thank you for the followup and glad you were able to get things working with STM32CubeIDE (which uses GCC).

We have noted the issue with older Keil and SP ASM (inline assembly), but have no immediate plans to work on it.

I am marking this issue solved for now.

Thanks,
David Garske, wolfSSL

closed 2021-06-15 by dgarske