wolfSSL/wolfssl #4009: Conflicting declarations

Haag-se, 2021-05-04:
Hi,

I get a lot of errors when compiling my C++ code that includes wolfSSL, Boost.Asio and CppRestSDK. Most errors concerns conflicting declarations in different header files.

Removing wolfSSL solves the conflicts. I have tried both installing wolfSSL with apt-get and from source with the following flags: 
```
./configure --enable-opensslall --enable-asio --enable-fastmath --enable-trustedca --enable-intelasm --enable-aesni --enable-sp  --enable-sp-asm  
```

To compile the code:
```
g++-10 -g -std=c++20 -march=native -O3 alpha.cpp -lmdf -ltbb -ltensorflow-lite -lsqlite3 -lwolfssl -lcpprest -lssl -lcrypto -lboost_system -lpthread -lm -ldl -licuio -licuuc -o alpha
```

This is part of the compiler output:

```
In file included from /usr/local/include/wolfssl/ssl.h:89,
                 from alpha.cpp:62:
/usr/local/include/wolfssl/openssl/bn.h:137:24: error: conflicting declaration ‘typedef WOLFSSL_BIGNUM BIGNUM’
  137 | typedef WOLFSSL_BIGNUM BIGNUM;
      |                        ^~~~~~
In file included from /usr/include/openssl/crypto.h:25,
                 from /usr/include/openssl/bio.h:20,
                 from /usr/include/openssl/conf.h:13,
                 from /usr/include/boost/asio/ssl/detail/openssl_types.hpp:23,
                 from /usr/include/boost/asio/ssl/context_base.hpp:19,
                 from /usr/include/boost/asio/ssl/context.hpp:23,
                 from /usr/include/boost/asio/ssl.hpp:18,
                 from /usr/include/cpprest/http_client.h:68,
                 from alpha.cpp:52:
/usr/include/openssl/ossl_typ.h:80:26: note: previous declaration as ‘typedef struct bignum_st BIGNUM’
   80 | typedef struct bignum_st BIGNUM;
      |                          ^~~~~~
In file included from /usr/local/include/wolfssl/ssl.h:89,
                 from alpha.cpp:62:
/usr/local/include/wolfssl/openssl/bn.h:138:24: error: conflicting declaration ‘typedef WOLFSSL_BN_CTX BN_CTX’
  138 | typedef WOLFSSL_BN_CTX BN_CTX;
      |                        ^~~~~~
In file included from /usr/include/openssl/crypto.h:25,
                 from /usr/include/openssl/bio.h:20,
                 from /usr/include/openssl/conf.h:13,
                 from /usr/include/boost/asio/ssl/detail/openssl_types.hpp:23,
                 from /usr/include/boost/asio/ssl/context_base.hpp:19,
                 from /usr/include/boost/asio/ssl/context.hpp:23,
                 from /usr/include/boost/asio/ssl.hpp:18,
                 from /usr/include/cpprest/http_client.h:68,
                 from alpha.cpp:52:
/usr/include/openssl/ossl_typ.h:81:27: note: previous declaration as ‘typedef struct bignum_ctx BN_CTX’
   81 | typedef struct bignum_ctx BN_CTX;
      |                           ^~~~~~
In file included from /usr/local/include/wolfssl/ssl.h:89,
                 from alpha.cpp:62:
/usr/local/include/wolfssl/openssl/bn.h:139:26: error: conflicting declaration ‘typedef WOLFSSL_BN_GENCB BN_GENCB’
  139 | typedef WOLFSSL_BN_GENCB BN_GENCB;
      |                          ^~~~~~~~
In file included from /usr/include/openssl/crypto.h:25,
                 from /usr/include/openssl/bio.h:20,
                 from /usr/include/openssl/conf.h:13,
                 from /usr/include/boost/asio/ssl/detail/openssl_types.hpp:23,
                 from /usr/include/boost/asio/ssl/context_base.hpp:19,
                 from /usr/include/boost/asio/ssl/context.hpp:23,
                 from /usr/include/boost/asio/ssl.hpp:18,
                 from /usr/include/cpprest/http_client.h:68,
                 from alpha.cpp:52:
/usr/include/openssl/ossl_typ.h:85:28: note: previous declaration as ‘typedef struct bn_gencb_st BN_GENCB’
   85 | typedef struct bn_gencb_st BN_GENCB;
      |                            ^~~~~~~~
In file included from /usr/include/openssl/x509.h:30,
                 from /usr/include/openssl/ssl.h:20,
                 from /usr/include/boost/asio/ssl/detail/openssl_types.hpp:24,
                 from /usr/include/boost/asio/ssl/context_base.hpp:19,
                 from /usr/include/boost/asio/ssl/context.hpp:23,
                 from /usr/include/boost/asio/ssl.hpp:18,
                 from /usr/include/cpprest/http_client.h:68,
                 from alpha.cpp:52:
/usr/local/include/wolfssl/openssl/sha.h:62:5: error: expected identifier before numeric constant
   62 |     SHA_DIGEST_LENGTH = 20
      |     ^~~~~~~~~~~~~~~~~
/usr/local/include/wolfssl/openssl/sha.h:62:5: error: expected ‘}’ before numeric constant
In file included from /usr/local/include/wolfssl/openssl/evp.h:45,
                 from /usr/local/include/wolfssl/openssl/hmac.h:38,
                 from /usr/local/include/wolfssl/ssl.h:90,
                 from alpha.cpp:62:
/usr/local/include/wolfssl/openssl/sha.h:61:6: note: to match this ‘{’
   61 | enum {
      |      ^
In file included from /usr/include/openssl/x509.h:30,
                 from /usr/include/openssl/ssl.h:20,
                 from /usr/include/boost/asio/ssl/detail/openssl_types.hpp:24,
                 from /usr/include/boost/asio/ssl/context_base.hpp:19,
                 from /usr/include/boost/asio/ssl/context.hpp:23,
                 from /usr/include/boost/asio/ssl.hpp:18,
                 from /usr/include/cpprest/http_client.h:68,
                 from alpha.cpp:52:
/usr/local/include/wolfssl/openssl/sha.h:62:5: error: expected unqualified-id before numeric constant
   62 |     SHA_DIGEST_LENGTH = 20
      |     ^~~~~~~~~~~~~~~~~
In file included from /usr/local/include/wolfssl/openssl/evp.h:45,
                 from /usr/local/include/wolfssl/openssl/hmac.h:38,
                 from /usr/local/include/wolfssl/ssl.h:90,
                 from alpha.cpp:62:
                 ```

elms, 2021-05-04:
@Haag-se It appears that it is including openssl headers as well as wolfSSL headers. I think this occurs when `BOOST_ASIO_USE_WOLFSSL` is not defined. You can resolve this by adding `#include <wolfssl/options.h>` before the other includes for asio or wolfSSL

Haag-se, 2021-05-05:
I tried your solution, but the errors remains. It seems like the combination wolfSSL and Boost.Asio is not the problem. Removing the inclusion of Boost.Asio still gives compilation errors. 

Instead, it seems like the combination of wolfSSL and CppRestSDK raises errors.

Boost.Asio + CppRestSDK WORKS
Boost.Asio + wolfSSL WORKS
CppRestSDK + wolfSSL ERRORS

Here is my includes:
```
#define BOOST_ASIO_USE_WOLFSSL
#include <wolfssl/options.h>
#include <wolfssl/ssl.h>

#include <sys/socket.h>
#include <arpa/inet.h>
#include <netinet/in.h>
#include <unistd.h>
#include <netinet/tcp.h>

#include <boost/asio/post.hpp>
#include <boost/asio/thread_pool.hpp>
#include <boost/algorithm/string.hpp>

#include <cpprest/http_client.h>
#include <cpprest/json.h>
#include <cpprest/ws_client.h>```

Haag-se, 2021-05-05:
I found a solution!

Building wolfSSl with `--enable-opensslcoexist` did the trick. I had to remove `--enable-asio`, `--enable-opensslall` and `--enable-trustedca` but that's ok.

closed 2021-05-05 by Haag-se