wolfSSL/wolfssl #4031: DTLS handshake started with invalid message sequence numbers

pfg666, 2021-05-12:
### Description
- Type: Bug
- Priority: Minor 

## Non-conformance Bug
**OS**  
Linux, Ubuntu 20

**build:**  
Version: 4.7.1r

**Affected Versions**
DTLS 1.2

**Related issues**
This issue is similar to the one I posted earlier,  [issue 4029](https://github.com/wolfSSL/wolfssl/issues/4029). The cause may very well be the same.

**Expected behavior**
According to the [DTLS 1.2. RFC](https://www.rfc-editor.org/rfc/rfc6347.html#page-16):

> The first message each side transmits in each handshake always has message_seq = 0. Whenever each new message is  generated, the message_seq value is incremented by one.

This means the server should only carry out the handshake if the first and second ClientHello messages have message_seq 0 and 1, respectively, assuming that cookie exchange is performed.

**Actual Behavior**
Our testing shows that the WolfSSL server does not properly validate the message_seq field. 
It completes the cookie exchange and generates the ServerHello flight following a ClientHello message with invalid message_seq (2 instead of 1).

**Steps to Reproduce**
I attached files necessary for reproduction (see reproduction.tar.gz) using [DTLS-Fuzzer](https://github.com/tls-attacker/TLS-Attacker), a Java-based tool for testing DTLS libraries based on TLS-Attacked. Also included in the archive is a capture of the interaction,  generated on my machine. DTLS-Fuzzer requires  the JDK for Java 8. On Ubuntu, this can be installed  by running:
`sudo apt-get install openjdk-8-jdk`

Unpack the archive, `cd` to resulting folder and run `bash reproduce.sh`, while running an instance of Wireshark on the side. The reproduction script will: 
* setup WolfSSL 4.7.1r and DTLS-Fuzzer;
* launch an WolfSSL server
* launch DTLS-Fuzzer, execute test.
 
If everything works as planned, Wireshark should show an interaction similar to that in the image below:

[image]

Notice that the second ClientHello prompts the server to generate a new ServerHello flight. Looking at the contents of this ClientHello, we see that its message_seq field is 2 instead of 1. Further testing reveals that message_seq can take an even higher number (e.g. 4) and the handshake will still proceed.

[image]

Thanks!
[reproduction.tar.gz](https://github.com/wolfSSL/wolfssl/files/6465926/reproduction.tar.gz)

pfg666, 2021-05-12:
I opened a similar [issue for Scandium](https://github.com/eclipse/californium/issues/1620), which also exhibited this behavior. From the discussions I had there, it appears the DTLS 1.2 RFC may provide some ground for accepting ClientHello messages with increased message sequence numbers based on the [quote](https://www.rfc-editor.org/rfc/rfc6347.html#page-18):

> If a server receives a ClientHello with an invalid cookie, it SHOULD
> treat it the same as a ClientHello with no cookie. This avoids
> race/deadlock conditions if the client somehow gets a bad cookie
> (e.g., because the server changes its cookie signing key).

> Note to implementors: This may result in clients receiving multiple
> HelloVerifyRequest messages with different cookies. Clients SHOULD
> handle this by sending a new ClientHello with a cookie in response to
> the new HelloVerifyRequest.

If this is indeed the reason for accepting such ClientHellos, then I think the issue can be closed. I am leaving this issue open to to ensure the behavior is intended (a dev may close it if that is the case), as it does appear to be a bit of an edge case.  The same applies for [the related issue](https://github.com/wolfSSL/wolfssl/issues/4029) I posted.

ejohnstown, 2023-01-13:
I believe this is not an issue at this point for the same reasons as for issues #4029 and #5224. Thank you for the bug report.

closed 2023-01-13 by ejohnstown