wolfSSL/wolfssl #4045: Won't compile on Debian 10, Ubuntu 20.04, or Fedora 34

BrianAker, 2021-05-18:
This is the output, the regression hit GitHub on April 24th. 

  CC       examples/server/server-server.o
In file included from examples/server/server.c:30:
./wolfssl/wolfcrypt/settings.h:2278:14: error: #warning "For timing resistance / side-channel attack prevention consider using harden options" [-Werror=cpp]
 2278 |             #warning "For timing resistance / side-channel attack prevention consider using harden options"
      |              ^~~~~~~
  CC       wolfcrypt/test/testsuite_testsuite_test-test.o
  CC       examples/client/testsuite_testsuite_test-client.o
cc1: all warnings being treated as errors
make[2]: *** [Makefile:5544: examples/server/server-server.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/home/build/workspace/wolfssl/label/ubuntu'
make[1]: *** [Makefile:5907: all-recursive] Error 1
make[1]: Leaving directory '/home/build/workspace/wolfssl/label/ubuntu'
make: *** [Makefile:3552: all] Error 2

elms, 2021-05-18:
Hi @BrianAker,

I've seen this before when jumping between revisions, and it can be corrected by running `make distclean`.

An example to reproduce the error
```
git checkout v4.7.0
make distclean
./configure && make
git checkout master
make
```

```
git describe 
v4.7.1r-272-gd8312a2e6
```

I'll look somemore to understand what is happening to need `make distclean`

BrianAker, 2021-05-18:
After doing a “make distclean”, get the following error:

  CC       tests/unit_test-srp.o
In function ‘DeriveKey’,
    inlined from ‘DeriveFinishedSecret’ at src/tls13.c:883:12:
src/tls13.c:438:12: error: ‘hash’ may be used uninitialized [-Werror=maybe-uninitialized]
  438 |     return HKDF_Expand_Label(output, outputLen, secret, hashSz,
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  439 |                              protocol, protocolLen, label, labelLen,
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  440 |                              hash, hashOutSz, digestAlg);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/tls13.c: In function ‘DeriveFinishedSecret’:
src/tls13.c:224:12: note: by argument 7 of type ‘const byte *’ {aka ‘const unsigned char *’} to ‘HKDF_Expand_Label.constprop’ declared here
  224 | static int HKDF_Expand_Label(byte* okm, word32 okmLen,
      |            ^~~~~~~~~~~~~~~~~
src/tls13.c:388:17: note: ‘hash’ declared here
  388 |     byte        hash[WC_MAX_DIGEST_SIZE];
      |                 ^~~~
In function ‘DeriveKey’,
    inlined from ‘DeriveTrafficSecret’ at src/tls13.c:902:12,
    inlined from ‘DeriveTls13Keys’ at src/tls13.c:1201:23:
src/tls13.c:438:12: error: ‘hash’ may be used uninitialized [-Werror=maybe-uninitialized]
  438 |     return HKDF_Expand_Label(output, outputLen, secret, hashSz,
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  439 |                              protocol, protocolLen, label, labelLen,
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  440 |                              hash, hashOutSz, digestAlg);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/tls13.c: In function ‘DeriveTls13Keys’:
src/tls13.c:224:12: note: by argument 7 of type ‘const byte *’ {aka ‘const unsigned char *’} to ‘HKDF_Expand_Label.constprop’ declared here
  224 | static int HKDF_Expand_Label(byte* okm, word32 okmLen,
      |            ^~~~~~~~~~~~~~~~~
src/tls13.c:388:17: note: ‘hash’ declared here
  388 |     byte        hash[WC_MAX_DIGEST_SIZE];
      |                 ^~~~
In function ‘DeriveKey’,
    inlined from ‘DeriveFinishedSecret’ at src/tls13.c:883:12,
    inlined from ‘DoTls13Finished’ at src/tls13.c:6206:15:
src/tls13.c:438:12: error: ‘hash’ may be used uninitialized [-Werror=maybe-uninitialized]
  438 |     return HKDF_Expand_Label(output, outputLen, secret, hashSz,
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  439 |                              protocol, protocolLen, label, labelLen,
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  440 |                              hash, hashOutSz, digestAlg);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/tls13.c: In function ‘DoTls13Finished’:
src/tls13.c:224:12: note: by argument 7 of type ‘const byte *’ {aka ‘const unsigned char *’} to ‘HKDF_Expand_Label.constprop’ declared here
  224 | static int HKDF_Expand_Label(byte* okm, word32 okmLen,
      |            ^~~~~~~~~~~~~~~~~
src/tls13.c:388:17: note: ‘hash’ declared here
  388 |     byte        hash[WC_MAX_DIGEST_SIZE];
      |                 ^~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:5412: src/libwolfssl_la-tls13.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
src/ssl.c: In function ‘wolfSSL_writev’:
src/ssl.c:14460:19: error: ‘staticBuffer’ may be used uninitialized [-Werror=maybe-uninitialized]
14460 |             ret = wolfSSL_write(ssl, myBuffer, sending);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/ssl.c:1980:5: note: by argument 2 of type ‘const void *’ to ‘wolfSSL_write’ declared here
 1980 | int wolfSSL_write(WOLFSSL* ssl, const void* data, int sz)
      |     ^~~~~~~~~~~~~
src/ssl.c:14432:20: note: ‘staticBuffer’ declared here
14432 |             byte   staticBuffer[FILE_BUFFER_SIZE];
      |                    ^~~~~~~~~~~~

乌

elms, 2021-05-18:
Can you share the revision and configure arguments you are using?

I think I just identifying the PR that causes the original issue.

elms, 2021-05-18:
@BrianAker For the original issue I think I address it in #4047

elms, 2021-05-20:
Closing since #4047 is merged. Please reopen if you encounter this again.

closed 2021-05-20 by elms