wolfSSL/wolfssl #4048: Won't compile on Fedora 34

BrianAker, 2021-05-18:
Pulled from GitHub yesterday/today:

autoreconf -I
./configure
make

CC       wolfcrypt/test/test.o
src/ssl.c: In function ‘wolfSSL_writev’:
src/ssl.c:14460:19: error: ‘staticBuffer’ may be used uninitialized [-Werror=maybe-uninitialized]
14460 |             ret = wolfSSL_write(ssl, myBuffer, sending);
      |                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/ssl.c:1980:5: note: by argument 2 of type ‘const void *’ to ‘wolfSSL_write’ declared here
 1980 | int wolfSSL_write(WOLFSSL* ssl, const void* data, int sz)
      |     ^~~~~~~~~~~~~
src/ssl.c:14432:20: note: ‘staticBuffer’ declared here
14432 |             byte   staticBuffer[FILE_BUFFER_SIZE];
      |                    ^~~~~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:5398: src/libwolfssl_la-ssl.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
In function ‘DeriveKey’,
    inlined from ‘DeriveFinishedSecret’ at src/tls13.c:883:12:
src/tls13.c:438:12: error: ‘hash’ may be used uninitialized [-Werror=maybe-uninitialized]
  438 |     return HKDF_Expand_Label(output, outputLen, secret, hashSz,
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  439 |                              protocol, protocolLen, label, labelLen,
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  440 |                              hash, hashOutSz, digestAlg);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/tls13.c: In function ‘DeriveFinishedSecret’:
src/tls13.c:224:12: note: by argument 7 of type ‘const byte *’ {aka ‘const unsigned char *’} to ‘HKDF_Expand_Label.constprop’ declared here
  224 | static int HKDF_Expand_Label(byte* okm, word32 okmLen,
      |            ^~~~~~~~~~~~~~~~~
src/tls13.c:388:17: note: ‘hash’ declared here
  388 |     byte        hash[WC_MAX_DIGEST_SIZE];
      |                 ^~~~
In function ‘DeriveKey’,
    inlined from ‘DeriveTrafficSecret’ at src/tls13.c:902:12,
    inlined from ‘DeriveTls13Keys’ at src/tls13.c:1201:23:
src/tls13.c:438:12: error: ‘hash’ may be used uninitialized [-Werror=maybe-uninitialized]
  438 |     return HKDF_Expand_Label(output, outputLen, secret, hashSz,
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  439 |                              protocol, protocolLen, label, labelLen,
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  440 |                              hash, hashOutSz, digestAlg);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/tls13.c: In function ‘DeriveTls13Keys’:
src/tls13.c:224:12: note: by argument 7 of type ‘const byte *’ {aka ‘const unsigned char *’} to ‘HKDF_Expand_Label.constprop’ declared here
  224 | static int HKDF_Expand_Label(byte* okm, word32 okmLen,
      |            ^~~~~~~~~~~~~~~~~
src/tls13.c:388:17: note: ‘hash’ declared here
  388 |     byte        hash[WC_MAX_DIGEST_SIZE];
      |                 ^~~~
In function ‘DeriveKey’,
    inlined from ‘DeriveFinishedSecret’ at src/tls13.c:883:12,
    inlined from ‘DoTls13Finished’ at src/tls13.c:6206:15:
src/tls13.c:438:12: error: ‘hash’ may be used uninitialized [-Werror=maybe-uninitialized]
  438 |     return HKDF_Expand_Label(output, outputLen, secret, hashSz,
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  439 |                              protocol, protocolLen, label, labelLen,
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  440 |                              hash, hashOutSz, digestAlg);
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/tls13.c: In function ‘DoTls13Finished’:
src/tls13.c:224:12: note: by argument 7 of type ‘const byte *’ {aka ‘const unsigned char *’} to ‘HKDF_Expand_Label.constprop’ declared here
  224 | static int HKDF_Expand_Label(byte* okm, word32 okmLen,
      |            ^~~~~~~~~~~~~~~~~
src/tls13.c:388:17: note: ‘hash’ declared here
  388 |     byte        hash[WC_MAX_DIGEST_SIZE];
      |                 ^~~~
cc1: all warnings being treated as errors

BrianAker, 2021-05-18:
[config.log](https://github.com/wolfSSL/wolfssl/files/6503803/config.log)

elms, 2021-05-18:
I've been able to reproduce. New warnings are probably due to gcc 11. At least one of the errors is the compiler being overly cautious. Looking at the best way to suppress warnings in select cases.

kaleb-himes, 2021-05-18:
Just my 2¢, We have yet to add any version of gcc 9, 10, or 11 to our Jenkins slave network due to what we believe are a large number of false positives IE overly zealous compiler warnings when testing with and without fsanitize (libasan).

elms, 2021-05-19:
@kaleb-himes @BrianAker I've put up PR https://github.com/wolfSSL/wolfssl/pull/4052 I don't love adding the `pragma`s to disable the warnings on those lines. The other option was adding `memset`. Other issues were related to inconsistent whitespace.

BrianAker, 2021-05-20:
I believe the compiler is correct.  If NO_SHA256, WOLFSSL_SHA384, and WOLFSSL_SHA384, are not defined then ‘hash’ may be used uninitialized.
If for some reason you end up with the previous true, I would force the compiler to throw a compile time error  ( assuming that having the hash be zeroed out is not an appropriate state ).

elms, 2021-05-20:
@BrianAker  I did notice that with those define options I added an error to the default case  [see change](https://github.com/wolfSSL/wolfssl/pull/4052/files#diff-927d62a3fce45ef1b7d9ba70df049ab4dc21ea53fc2491b5c006fc96c5af680eR424). I believe this is the correct way to handle it, but I get other errors manually disabling SHA256... don't see a way to set the flag via `configure` and becomes a little bit of a rabbit hole to disable them all.

elms, 2021-06-10:
Closing as #4052 merged. Let me know if you find any other errors

closed 2021-06-10 by elms