wolfSSL/wolfssl #4051: Dead mp_toraw and mp_read_raw macros

jacobboertjes, 2021-05-19:
In `wolfssl/wolfcrypt/integer.h`, the macros `mp_toraw` & `mp_read_raw` are defined to point to `mp_to_signed_bin` & `mp_read_signed_bin` respectively, but no implementations for these function are available in the library.

SparkiDev, 2021-05-20:
Hi @jacobboertjes,

That is correct. We do not support those APIs anymore.
Do you have concerns about this?

jacobboertjes, 2021-05-27:
I don't have concerns other than causing a bit of confusion. For example, a function like `wc_ecc_import_raw` along with `mp_toraw`  seem like they should be used together when you have raw key values stored in `mp_int` format. If there is no concerns on your end then this can be closed. I just figured I'd point it out incase it wasn't known.

SparkiDev, 2021-05-27:
Hi @jacobboertjes

Thanks for pointing this out to us!
I agree that it is confusing and appreciate you bring it to our notice.
There is a pull request up now that removes the APIs from the header file.

Thanks,
Sean :-)

closed 2021-05-27 by SparkiDev