wolfSSL/wolfssl #4065: Upstream, addition of an extended version of wc_RsaKeyToPublicDer

ghs-sec, 2021-05-25:
Good afternoon,

When using wolf RSA API, I once needed to export an rsa key without the DER header. This functionality is offered by the static function SetRsaPublicKey, however not accessible for the user. I therefore added an extended version of wc_RsaKeyToPublicDer offering the possibility to chose the header or not.
Therefore, I would like to send it upstream to you, to be eventually included in wolf code.

I join to this issue a zip with a patch file for this functionality.

Best regards
Blaise Plagnol

[RsaKeyToPublicDer_ex.zip](https://github.com/wolfSSL/wolfssl/files/6540056/RsaKeyToPublicDer_ex.zip)

embhorn, 2021-05-25:
Hi Blaise,

Thank you for the contribution! Could you please open a pull request containing the changes? This will begin the contribution approval process.

Thanks,
Eric @ wolfSSL Support

ghs-sec, 2021-05-26:
Done, thanks for the guidance

embhorn, 2021-05-27:
This issue will be fixed by #4068

embhorn, 2021-06-14:
Being tracked by PR #4068, so closing this issue.

closed 2021-06-14 by embhorn