wolfSSL/wolfssl #4082: Is there any user_settings.h template?

lygstate, 2021-06-02:
I want a user_settings.h template with full description.

dgarske, 2021-06-02:
Hi @lygstate ,

See these:
https://github.com/wolfSSL/wolfssl/tree/master/examples/configs
https://github.com/wolfSSL/wolfssl/blob/master/IDE/GCC-ARM/Header/user_settings.h

Thanks,
David Garske, wolfSSL

lygstate, 2021-06-03:
> Hi @lygstate ,
> 
> See these:
> https://github.com/wolfSSL/wolfssl/tree/master/examples/configs
> https://github.com/wolfSSL/wolfssl/blob/master/IDE/GCC-ARM/Header/user_settings.h
Thanks, https://github.com/wolfSSL/wolfssl/blob/master/IDE/GCC-ARM/Header/user_settings.h

 looks better, but hard to found,
can we update https://github.com/wolfSSL/wolfssl/tree/master/examples/configs/user_settings.h to match
https://github.com/wolfSSL/wolfssl/blob/master/IDE/GCC-ARM/Header/user_settings.h？
I've seen https://github.com/wolfSSL/wolfssl/tree/master/examples/configs/user_settings.h before but I gived it up
> 
> Thanks,
> David Garske, wolfSSL

dgarske, 2021-06-04:
Hi @lygstate ,

So you would like to see the IDE/GCC-ARM/Header/user_settings.h in the `examples/configs` dir? That sounds good to me and I will put up something soon with that change soon.

Are you having any build issues or were the example user_settings.h sufficient?

Note: we do have most build macros documented in the user manual in chapter 2 here:
https://www.wolfssl.com/docs/wolfssl-manual/ch2/

Thanks,
David Garske, wolfSSL

dgarske, 2021-06-09:
I put up a PR with some additional user_settings.h examples here:
https://github.com/wolfSSL/wolfssl/pull/4110

embhorn, 2022-08-23:
Closing this older issue as resolved.

closed 2022-08-23 by embhorn