wolfSSL/wolfssl #4092: TLS 1.3 with ESP8266

oaugustocezar, 2021-06-04:
I'm trying to run a client with wolfssl on ESP8266. I've configured my callback functions to send and receive data, but when I call the wolfSSL_connect(), I send the CLIENT_HELLO_SENT, but i'm not receiving any data from the server.

**Here's my code:**

void setup() {
  int err            = 0;
  char errBuf[80];
  WiFi.begin(SSID, PASS);
  Serial.begin(9600); 
  //wolfSSL_Debugging_ON();
  wolfSSL_SetLoggingCb(log_function);
  Serial.printf("Connecting to %s ", SSID); 
  while (WiFi.status() != WL_CONNECTED)
  {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" connected");  

  if(client.connect(HOST,PORT)){
    client.setSync(true);
    Serial.println("Server connected");   
    wolfSSL_Init();
    method = wolfTLSv1_2_client_method(); /* use TLS v1.3 */    
    if((ctx = wolfSSL_CTX_new(method)) == NULL){
      Serial.println("wolfSSL_CTX_new error");      
    }

    err = wolfSSL_CTX_load_verify_buffer(ctx,(const byte*)cert_fog_der_2048,sizeof_cert_fog_der_2048,SSL_FILETYPE_ASN1); 
    if(err != SSL_SUCCESS){
      Serial.println("Error loading certs");
      Serial.println(err);
    }
    
  // initialize wolfSSL using callback functions
    wolfSSL_CTX_set_verify(ctx, SSL_VERIFY_NONE,0);
    wolfSSL_SetIOSend(ctx, WiFiSend);
    wolfSSL_SetIORecv(ctx, WiFiReceive);    
    return;
  }
}

void loop() {
  int err            = 0;
  int input          = 0;
  int total_input    = 0;
  int ret;
  char msg[500];
  sprintf(msg,"GET / HTTP/1.1\nHost: %s\r\n\r\n",HOST);
  int msgSz          = (int)strlen(msg);
  char errBuf[80];
  char reply[500];
  const char* cipherName;
  int flagWrite;
  Serial.println(msg);
  Serial.print("msg size: ");
  Serial.println(msgSz);

  err = wolfSSL_CTX_UseSNI(ctx, WOLFSSL_SNI_HOST_NAME, HOST, XSTRLEN(HOST));    
  if (err != WOLFSSL_SUCCESS) {
      sprintf(errBuf,"Setting host name failed with error condition: %d and reason %s\n", ret, wolfSSL_ERR_error_string(ret, reply)); 
      Serial.print(errBuf);
  }
    
  Serial.print("Connected to ");
  Serial.println(HOST);
  
  ssl = wolfSSL_new(ctx);
  if (ssl == NULL) {
    Serial.println("Unable to allocate SSL object");
    return;
  }else{
    Serial.println("SSL object allocate");
  }
  err = wolfSSL_connect(ssl);
  Serial.println("After connect");
  Serial.println(err);
  if (err != WOLFSSL_SUCCESS) {
    err = wolfSSL_get_error(ssl, 0);
    wolfSSL_ERR_error_string(err, errBuf);
    Serial.print("TLS Connect Error: ");
    Serial.println(errBuf);
  }
  Serial.print("SSL version is ");
  Serial.println(wolfSSL_get_version(ssl));  
  cipherName = wolfSSL_get_cipher(ssl);
  Serial.print("SSL cipher suite is ");
  Serial.println(cipherName);
  
  flagWrite = wolfSSL_write(ssl, msg, msgSz); 
  err = wolfSSL_get_error(ssl, 0);
  wolfSSL_ERR_error_string(err, errBuf);
  Serial.print("Write Error: ");
  Serial.println(errBuf);
  if (flagWrite > 0 ) {    
    Serial.print("Server response: ");
    while (client.available() || wolfSSL_pending(ssl)) {
     input = wolfSSL_read(ssl, reply, sizeof(reply) - 1);
      total_input += input;
     if (input < 0) {
        err = wolfSSL_get_error(ssl, 0);
        wolfSSL_ERR_error_string(err, errBuf);
        Serial.print("TLS Read Error: ");
        Serial.println(errBuf);
        break;
      } else if (input > 0) {
        reply[input] = '\0';
        Serial.print(reply);
      } else {
        Serial.println();
      }
    } 
  } else {
    err = wolfSSL_get_error(ssl, 0);
    wolfSSL_ERR_error_string(err, errBuf);
    Serial.print("TLS Write Error: ");
    Serial.println(errBuf);
  }
  
  wolfSSL_shutdown(ssl);
  wolfSSL_free(ssl);

  client.stop();
  Serial.println("Connection complete.");
  
  delay(1000);
}

int WiFiSend(WOLFSSL* ssl_cli, char* msg, int sz, void* ctx_cli) {
  int sent = 0;
  Serial.println("WiFiSend");
  Serial.println(sz);
  sent = client.write(msg,sz);  
  Serial.println(sent); 
  client.flush(); 
  return sent;
}

int WiFiReceive(WOLFSSL* ssl_cli, char* reply, int sz, void* ctx_cli) {
  int ret = 0;
  Serial.println("WiFiReceive");
  Serial.println(sz);
  while(!client.available()){}
  while(client.available() > 0 && ret < sz) {
    reply[i++] = client.read();    
    Serial.print("Ret: ");
    Serial.println(ret);
  }  
  return ret;
}

**And here's my user_settings.h:**

/* Generated wolfSSL user_settings.h file for Arduino */
#ifndef ARDUINO_USER_SETTINGS_H
#define ARDUINO_USER_SETTINGS_H

/* Platform */
#define WOLFSSL_ARDUINO

/* Math library (remove this to use normal math)*/
#define USE_FAST_MATH
#define TFM_NO_ASM
#define TFM_TIMING_RESISTANT

/* RNG DEFAULT !!FOR TESTING ONLY!! */
/* comment out the error below to get started w/ bad entropy source
 * This will need fixed before distribution but is OK to test with */
/* #error "needs solved, see: https://www.wolfssl.com/docs/porting-guide/" */
#define WOLFSSL_GENSEED_FORTEST

#define USE_CERT_BUFFERS_2048
//#define DEBUG_WOLFSSL
#define NO_ASN_TIME
#define SINGLE_THREADED 

#define XTIME fnSecondsSinceEpoch
#define XGMTIME
#define ALT_ECC_SIZE
#define WOLFSSL_USER_IO
#define STATIC_CHUNKS_ONLY
#define NO_FILESYSTEM
#define WOLFSSL_TLS13
#define HAVE_TLS_EXTENSIONS
#define HAVE_SUPPORTED_CURVES
#define HAVE_ECC
#define HAVE_HKDF
#define WC_RSA_PSS
#define NO_DH 
#define HAVE_SNI
#define SMALL_SESSION_CACHE
#define HAVE_CHACHA
//#define HAVE_POLY1305
#define WOLFSSL_STATIC_MEMORY
#define HAVE_AESCCM
#define WOLFSSL_SMALL_CERT_VERIFY
#define GCM_SMALL
#define USE_SLOW_SHA256
#define HAVE_AESGCM
#endif /* ARDUINO_USER_SETTINGS_H */

dgarske, 2021-06-07:
Hi @oaugustocezar ,

Your code and build settings look fine. What is the TLS server you are connecting to and do you have the logs from it? I don't see any sockets being used here, instead it looks like you are sending the client_hello over UART. I assume your `WiFiReceive` is called after the client_hello is sent? Our flow is to first ask for 5 bytes (the TLS header) then `WiFiReceive` is called again for the remainder. And issue I have seen with using UART is no flow control, so you may need to implement an IRQ and global buffer to capture the UART bytes and then have your `WiFiReceive` pull from the global buffer.

Thanks,
David Garske, wolfSSL

dgarske, 2021-06-14:
Hi @oaugustocezar ,

I wanted to check in and see if you were able to resolve the Arduino ESP8266 issues? Let us know if you need any further assistance.

Thanks,
David Garske, wolfSSL