wolfSSL/wolfssl #4104: Installing fails at the "make check" step, when the folder path has spaces in it

Eraz19, 2021-06-08:
On linux (in my case, Pop OS) if you clone the wolfssl repo in a folder path which contains whitespaces in it, the "make check" test suite fails.

elms, 2021-06-08:
@Eraz19 Thanks for the report. I was able to reproduce on Ubuntu 20.04

<details>
<summary>resume.log</summary>

```
ready file /home/elms/work/wolf/wolfssl space/wolfssl/wolfssl_resume_ready1363923
-r          Resume session
-e 
Starting example server for resume test...

-e 
Resume test supported
./scripts/resume.test: 32: test: /home/elms/work/wolf/wolfssl: unexpected operator
./scripts/resume.test: 75: [: /home/elms/work/wolf/wolfssl: unexpected operator
./scripts/resume.test: 81: test: /home/elms/work/wolf/wolfssl: unexpected operator
-e NO ready file ending test...
in cleanup
killing server
./scripts/resume.test: 32: test: /home/elms/work/wolf/wolfssl: unexpected operator
FAIL scripts/resume.test (exit status: 1)
```

</details>

<details>
<summary>tls13.log</summary>

```
ready file /home/elms/work/wolf/wolfssl space/wolfssl/wolfssl_tls13_ready1363912
-e 

TLS v1.3 server with TLS v1.3 client
./scripts/tls13.test: 36: [: /home/elms/work/wolf/wolfssl: unexpected operator
./scripts/tls13.test: 45: [: /home/elms/work/wolf/wolfssl: unexpected operator
-e NO ready file ending test...
in cleanup
killing server
./scripts/tls13.test: 60: [: /home/elms/work/wolf/wolfssl: unexpected operator
./scripts/tls13.test: 76: [: /home/elms/work/wolf/wolfssl: unexpected operator
./scripts/tls13.test: 80: [: /home/elms/work/wolf/wolfssl: unexpected operator
./scripts/tls13.test: 84: [: /home/elms/work/wolf/wolfssl: unexpected operator
tee: /home/elms/work/wolf/wolfssl: Is a directory
tee: space/wolfssl/wolfssl_tls13_client1363912: No such file or directory
wolfSSL error: port number cannot be 0
./scripts/tls13.test: 60: [: /home/elms/work/wolf/wolfssl: unexpected operator
-e 

TLS v1.3 not enabled
in cleanup
./scripts/tls13.test: 60: [: /home/elms/work/wolf/wolfssl: unexpected operator
./scripts/tls13.test: 76: [: /home/elms/work/wolf/wolfssl: unexpected operator
./scripts/tls13.test: 80: [: /home/elms/work/wolf/wolfssl: unexpected operator
./scripts/tls13.test: 84: [: /home/elms/work/wolf/wolfssl: unexpected operator
FAIL scripts/tls13.test (exit status: 1)
```

</details>

elms, 2021-06-09:
@Eraz19 I opened #4108 but will have to sort out failures I caused for other configs.

elms, 2021-06-15:
@Eraz19 #4108 has merged and fixes this. Thanks for the report.

closed 2021-06-15 by elms