wolfSSL/wolfssl #4179: SP math: Functions declared but not exported

guidovranken, 2021-07-02:
When I compile wolfCrypt with SP math using these commands: https://github.com/guidovranken/cryptofuzz/blob/master/docs/wolfcrypt.md#sp-math

The following program compiles but fails to link:

```c
#include <wolfssl/options.h>
#include <wolfssl/wolfcrypt/ecc.h>

int main(void)
{
    printf("%p, %p\n", ecc_projective_add_point_safe, ecc_map);
    return 0;
}
```

```
p.c:6: undefined reference to `ecc_projective_add_point_safe'
p.c:6: undefined reference to `ecc_map'
```

This suggests that these functions are declared in the header files but not compiled into the library.

dgarske, 2021-07-02:
@guidovranken , if you build with `WOLFSSL_PUBLIC_ECC_ADD_DBL` defined the `ecc_map` is made public. I think we should also change `ecc_projective_add_point_safe` to `ECC_API`. @JacobBarthelmeh perhaps the release PR put up for the release can add this?

dgarske, 2022-12-08:
@guidovranken going back through old issues. I fixed this in PR https://github.com/wolfSSL/wolfssl/pull/5868
Hopefully you can add some more ECC math tests.
Thanks, David Garske, wolfSSL

guidovranken, 2022-12-08:
Thanks but I was already testing these functions by compiling with `WOLFSSL_PUBLIC_ECC_ADD_DBL`.

Off-topic but I'm currently fuzzing these different bignum configurations: https://github.com/guidovranken/cryptofuzz/blob/master/docs/wolfcrypt.md

Anything I should add?

SparkiDev, 2022-12-08:
Hi Guido,

I can't see a configuration with --enable-sp-asm.
Could you add a configuration that has --enable-sp --enable-sp-asm --enable-sp-math?

Thanks,
Sean

dgarske, 2022-12-12:
> Hi Guido,
> 
> I can't see a configuration with --enable-sp-asm. Could you add a configuration that has --enable-sp --enable-sp-asm --enable-sp-math?
> 
> Thanks, Sean

@guidovranken , making sure you saw this feedback from Sean?  We'd like SP math with assembly also tested. Thanks

closed 2022-12-12 by dgarske

guidovranken, 2022-12-12:
@dgarske @SparkiDev is there any difference between adding --enable-sp-asm and not? It seems that `wolfcrypt/src/sp_x86_64_asm.S` is compiled in both cases.

dgarske, 2022-12-12:
> @dgarske @SparkiDev is there any difference between adding --enable-sp-asm and not? It seems that `wolfcrypt/src/sp_x86_64_asm.S` is compiled in both cases.

I depends on the platform. If you are building on Intel host (`x86_64`, `aarch64` and `amd64`) it does enable SP math assembly `WOLFSSL_SP_X86_64_ASM` by default. Perhaps it might be worth specifically testing with `--disable-sp-asm`, so `sp_c64.c` is used?  Are there any other architectures you can support?