wolfSSL/wolfssl #4241: undefined reference to `LowResTimer'

baldhead69, 2021-07-25:
Hi,

I am trying to migrate a project from mbedTLS to wolfssl in esp-idf v4.2.1.
I am using esp32.

I follow this link for configuration:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_tls.html

At compile time, the link error message appear:
.../esp-wolfssl/wolfssl/src/internal.c:30779: undefined reference to `LowResTimer'

Some suggestion ?

NOTE: https://github.com/espressif/esp-idf/issues/7312

Thank's.

baldhead69, 2021-07-26:
No one ?

kabuobeid, 2021-07-26:
Hi @baldhead69 ,

Thanks for reaching out to us.  I am able to reproduce the issue, it's coming from this repo's user_settings.h: https://github.com/espressif/esp-wolfssl/blob/master/port/user_settings.h
You will need to either undefine `NO_ASN_TIME`, or undefine `OPENSSL_ALL`.  I would suggest undefing both.
`NO_ASN_TIME` will disable validation of certificate times ie. expiration checks.  `OPENSSL_ALL` enables every single OpenSSL compatibility API we have which is almost certainly overkill.  For most users who want OpenSSL compatibility, `OPENSSL_EXTRA` is sufficient.

I would also recommend trying to use our ESP-IDF example here: https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF

Thanks,
Kareem

baldhead69, 2021-07-26:
hi @kabuobeid 

I made your suggestions and:

components/communication/esp-wolfssl/wolfssl/src/internal.c:8185:14: fatal error: task.h: No such file or directory
     #include "task.h"

kabuobeid, 2021-07-26:
Hi @baldhead69 ,

That #include comes from `FREERTOS` being defined, you will need to make sure your FreeRTOS task.h header is in your include path.

Thanks,
Kareem

baldhead69, 2021-07-26:
@kabuobeid 

I put "#include "freertos/task.h" at the place of #include "task.h" and now compiled ok, but it's a workaround.

A definitive solution is needed.

Thank's.

baldhead69, 2021-07-28:
Hi,

How to only save necessary files on a esp-idf with wolfssl project ?

The full folder are big.

dgarske, 2021-07-28:
Hi @baldhead69 ,

The issue with `freertos/task.h` vs `task.h` is an include path issue. You can typically add `-I` to your CFLAGS to provide additional include paths. Example: `CFLAGS="-I myfreertosdir`. You do make a good point because typically the FreeRTOS includes do not have "freertos/" in front. Let me know if you cannot resolve with an include path change.

For wolfSSL the only required files are:
* src/*.c
* wolfcrypt/src/*.c
* wolfssl/ (recursive all headers)

The wolfssl-root also needs added to your include path.

Thanks,
David Garske, wolfSSL

baldhead69, 2021-07-28:
Hi @dgarske 

is there any command i can run in git that does this ?
Only save the required files with git files too maybe.

Thanks.

dgarske, 2021-07-28:
Hi @baldhead69 ,

Not that I am aware of, but I do have a couple local scripts to clone and cleanup the repo.

Here is one you might find useful. Example usage `./trim-wolf.sh 4.8.1`

```
#!/bin/bash

if [ -z "$1" ]; then
    echo "usage: ./trim-wolf.sh [version]"
    exit 1
fi

if [ ! -d "wolfssl-$1" ]; then
    echo "Cloning wolfSSL version $1"
    git clone --branch v$1-stable --depth=1 https://github.com/wolfssl/wolfssl wolfssl-$1
fi
cd wolfssl-$1

# blank options.h
rm ./wolfssl/options.h
cp ./wolfssl/options.h.in ./wolfssl/options.h
rm ./cyassl/options.h
cp ./cyassl/options.h.in ./cyassl/options.h

# cleanup 
echo "Removing files not applicable embedded use"

for dir in ./IDE/*; do
    [ "$dir" = "./IDE/Espressif" ] && continue
    rm -rf "$dir"
done

rm -rf ./build-aux
rm -rf ./IPP
rm -rf ./lib
rm -rf ./linuxkm
rm -rf ./m4
rm -rf ./mcapi
rm -rf ./mplabx
rm -rf ./mqx
rm -rf ./rpm
rm -rf ./swig
rm -rf ./tests
rm -rf ./testsuite
rm -rf ./tirtos
rm -rf ./wolfcrypt/user-crypto
rm -f  ./wolfcrypt/src/*.S
rm -f  ./wolfcrypt/src/*.asm

cd ..

echo "Done"
```

Thanks,
David Garske, wolfSSL

baldhead69, 2021-07-28:
Thanks @dgarske

closed 2021-10-11 by baldhead69