wolfSSL/wolfssl #4245: gcc :  ‘asm’ operand has impossible constraints, on 32bit intel

vaintroub, 2021-07-26:
In file included from wolfssl/wolfssl/wolfcrypt/src/tfm.c:51:
wolfssl/wolfssl/wolfcrypt/src/tfm.c: In function ‘fp_sqr_comba’:
wolfssl/wolfcrypt/src/asm.c:709:1: error: ‘asm’ operand has impossible constraints
 __asm__(                                                  \
 ^~~~~~~
wolfssl/wolfssl/wolfcrypt/src/tfm.c:3148:11: note: in expansion of macro ‘SQRADD2’
           SQRADD2(*tmpx++, *tmpy--);
           ^~~~~~~
wolfssl/wolfssl/wolfcrypt/src/asm.c:700:1: error: ‘asm’ operand has impossible constraints
 __asm__(                                                  \
 ^~~~~~~
/home/wlad/10.4/extra/wolfssl/wolfssl/wolfcrypt/src/tfm.c:3154:11: note: in expansion of macro ‘SQRADD’
           SQRADD(A->dp[ix>>1], A->dp[ix>>1]);

kabuobeid, 2021-07-26:
Hi @vaintroub ,

Thank you for reaching out to us.  I will work on reproducing your issue here.

Thanks,
Kareem

elms, 2021-07-26:
Hi @vaintroub,

Thanks for the report. I have reproduced and believe I know a solution. I was able to compile by adding `-O2`. The exact command I used to test was `cmake .. -DWOLFSSL_USER_SETTINGS=yes -DCMAKE_C_FLAGS="-m32 -I$PWD -O2" && make`.

I'll work on putting a PR up. If you are able, can you confirm that adding `-O2` solves the issue?

vaintroub, 2021-07-27:
That does not solve it in all circumstances. We use sometimes -fno-omit-frame-pointers, also with optimized build. 
I assume you  need to add -fomit-frame-pointers, to free a register . Not only that, on downlevel compilers, such as GCC 4.8.5, that we are also using, you'd need to add -fno-PIC, to free yet another register - we compile everything with -fPIC. However -fno-PIC fortunately works on Intel 32bit.

Here is an example where -fomit-frame-pointer -fPIC fails : 
https://buildbot.askmonty.org/buildbot/builders/kvm-bintar-centos6-x86/builds/5541/steps/compile/logs/stdio
(look for tfm.c.o on how it is compiled, and for error: on how it fails)

Here is an example where -fomit-frame-pointer -fno-PIC succeeds (same machine)
https://buildbot.askmonty.org/buildbot/builders/kvm-bintar-centos6-x86/builds/5549/steps/compile/logs/stdio

elms, 2021-07-29:
@vaintroub Do I understand correctly that you have a workaround for now?

The only solution that I know of that would avoid this completely is using `WOLFSSL_NO_ASM`. I'll discuss other options with the team.

Out of curiosity have you considered `WOLFSSL_SP_MATH`?

stintel, 2021-08-18:
We're also seeing this in OpenWrt after switching the default GCC version from 8 to 10.

SparkiDev, 2021-08-20:
Hi @vaintroub,

Please try pull request https://github.com/wolfSSL/wolfssl/pull/4317

I've made a small change to make the assembly code snippets work.

Thanks.

stintel, 2021-08-20:
> Please try pull request #4317

This seems to fix the problem for 32bit x86 in OpenWrt, thanks!

vaintroub, 2021-08-20:
Unfortunately, it is a not easy to test for me at the moment, but once it is in, but I'm happy to hear about the progress,
Once it is in main tree, I will try to remove workarounds and see what it says.
But if it works for you with -fno-omit-frame-pointer  -fPIC, and/or with Debug compilation, it should be fine here too.

SparkiDev, 2021-08-22:
Hi @vaintroub

The pull request has been merged to master.

Let me know when you have a chance to test it.

Thanks,
Sean :-)

elms, 2022-05-16:
@vaintroub Haven't heard back for a while and worked for others, so I'm marking it closed. Please open a new bug if you have any issues.