wolfSSL/wolfssl #4248: disable weak ssl ciphers by default?

vuvova, 2021-07-26:
Since 25 Aug 2016 OpenSSL disables RC4 by default, one needs to enable it explicitly using the enable-weak-ssl-ciphers option to Configure.

Should WolfSSL also do it, define NO_RC4 by default? For example,
```c
#ifndef ENABLE_RC4
#define NO_RC4
#endif
```

dgarske, 2021-07-26:
Hi @vuvova ,

Our RC4 is disabled by default in ./configure, but you are right if providing build options manually `NO_RC4` must be defined.

```
./configure --help | grep rc4
  --enable-arc4           Enable ARC4 (default: disabled)
```

You will also notice all our template user_settings.h have `NO_RC4` defined:
https://github.com/wolfSSL/wolfssl/tree/master/examples/configs

I've made note of this and will discuss internally.

Thanks,
David Garske, wolfSSL

dgarske, 2021-07-28:
Hi @vuvova , 

Thank you for this report. I've put up a PR #4260 to resolve this. Let me know if you see any issues.

Thanks,
David Garske, wolfSSL

embhorn, 2022-08-23:
Closing this older issue as resolved.

closed 2022-08-23 by embhorn