wolfSSL/wolfssl #4272: Where to find libwolfssl ?

gojimmypi, 2021-08-02:
I'm trying to get the `WolfSSL` examples in this repo to compile. I have a [blog](https://gojimmypi.github.io/ESP32-SSH-server/) with some of the details of problems I've encountered.

I started with trying to [configure TLS 1.3 DH key size](https://github.com/espressif/esp-wolfssl/issues/11) setting for the ESP32. With no success there, I tried to download this repo like this:

```
cd /mnt/c/workspace/
git clone https://github.com/wolfSSL/wolfssh.git --recursive
cd wolfssh
./autogen.sh
./configure
```

Unfortunately, the above yields this error during the `./configure` step:

```
...
checking for wolfCrypt_Init in -lwolfssl... no
configure: error: libwolfssl is required for wolfssh. It can be obtained from https://www.wolfssl.com/download.html/ .
```

I could not find any reference to `libwolfssl` at the referenced download link. Any tips on how to configure this repo with `libwolfssl`?

There's a `libwolfssl.c` in the wolfssl-4.8.1.zip file I downloaded , located in `\IDE\INTIME-RTOS` - but when copying it to `wolfssh/src/` in this repo, I still see the same error.

Note that my real goal is to get a WolfSSL SSH Server running on the ESP32 with the `esp-idf`.

Thank you, Any suggestions will be appreciated.

dgarske, 2021-08-02:
Hi @gojimmypi ,

wolfSSH requires the wolfSSL library. The steps are:

```
git clone https://github.com/wolfssl/wolfssl.git
cd wolfssl
./autogen.sh
./configure --enable-ssh
make
sudo make install
sudo ldconfig
```

See instructions here:
https://github.com/wolfSSL/wolfssh#dependencies

Then you can ./configure and build wolfSSH.

Let us know if you have any issues. 

Thanks,
David Garske, wolfSSL

dgarske, 2021-08-02:
Hi @gojimmypi ,

Another thing to consider is you will be cross compiling for the ESP32. The steps you mentioned above for wolfSSH and the ones I just posed are for a local build. If your target is an embedded device the steps are quite different.

Good blog about building here:
https://www.wolfssl.com/docs/espressif/

There is a port of ESP32 for wolfSSL here:
https://github.com/espressif/esp-wolfssl

If you are seeing errors like `#warning "For timing resistance / side-channel attack prevention consider using harden options"` then it usually means you are not providing wolfSSL build options. You have a couple options for configuration of the library.

1) Define `WOLFSSL_USER_SETTINGS` and provide your own user_settings.h. Note you must define this when building wolfSSL and your application.
2) If using ./configure the build options are output to `wolfssl/options.h` which you can also include in your application.

See the FAQ section 1 for details:
https://www.wolfssl.com/docs/frequently-asked-questions-faq/#How_do_I_manage_the_build_configuration_for_wolfSSL?

Thanks,
David Garske, wolfSSL

gojimmypi, 2021-08-03:
ah yes, that's quite helpful. Thank you. I had started with the zip download that seems to have different contents. It was a real struggle. I was not aware of the cross-dependency.

I've added your commands noted in [your first comment, above](https://github.com/wolfSSL/wolfssl/issues/4272#issuecomment-891199577) in https://github.com/wolfSSL/wolfssh/pull/356

I'll check the Espressif details soon.

gojimmypi, 2022-06-04:
tools such as `readelf` are helpful here:

```
readelf -d /usr/local/lib/libwolfssl.so | grep SONAME
```

compiling with Ubuntu WSL1 on Windows 10 with additional `-Wl,-rpath=/usr/local/lib` (thanks [SO](https://stackoverflow.com/a/67753144/5432351)) was also helpful:

```
gcc -o server-dtls server-dtls.c -Wall -I/usr/local/include -Os -L/usr/local/lib -lm -lwolfssl  -Wl,-rpath=/usr/local/lib
```

but the best option was to add the `--prefix=/usr/` at wolfssl [install](https://github.com/wolfSSL/wolfssl/blob/master/INSTALL) time:

```
./configure  --enable-tls13 --prefix=/usr/ && make && sudo make install
```

Thanks & credit @dgarske for the tips.

Also: When I first asked the question, I think perhaps I didn't realize at the time that wolfssl needs to be installed into the [components directory](https://github.com/espressif/esp-idf/tree/master/components) of the [ESP-IDF](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html). See also [wolfSSH #364](https://github.com/wolfSSL/wolfssh/issues/364#issuecomment-1136500140)

My goal of getting the [SSH Server on the ESP32](https://github.com/gojimmypi/wolfssh/tree/ESP32_Development/examples/ESP32-SSH-Server) was finally realized! :) Also implemented on [ESP8266](https://github.com/gojimmypi/wolfssh/tree/ESP8266_Development/examples/ESP8266-SSH-Server).

closed 2022-06-04 by gojimmypi