wolfSSL/wolfssl #4281: unable to connect to ssh example echoserver with putty

gojimmypi, 2021-08-04:
I'm working with the example [ssh server](https://github.com/wolfSSL/wolfssl/tree/master/examples/echoserver)  and [ssh client](https://github.com/wolfSSL/wolfssl/tree/master/examples/echoclient) in WSL. I am able to run the server app and successfully connect with the client app. The problem occurs when I try to connect with `putty`.

If I try to connect using my local host IP address:
[image]

When connecting with host IP address, the [ssh server](https://github.com/wolfSSL/wolfssl/tree/master/examples/echoserver) does not show any output, and putty simply says "connection refused".
[image]

When using `localhost` 127.0.0.1, `putty` seems to connect, but the ssh server refuses with "Remote side unexpectedly closed network connection":
[image]

The sample [ssh server](https://github.com/wolfSSL/wolfssl/tree/master/examples/echoserver) reports an error of "unknown type in record hdr" when the above  "Remote side unexpectedly closed network connection" message is encountered:
[image]

There's nothing immediately obvious in Wireshark for the putty failure, although there's no "hello":
[image]

For reference, here's what the successful [ssh client](https://github.com/wolfSSL/wolfssl/tree/master/examples/echoclient) connecting to [ssh server](https://github.com/wolfSSL/wolfssl/tree/master/examples/echoserver) looks like in Wireshark:
[image]

Note that to be extra sure this is not a local Windows Firewall issue, I have 2 explicit inbound rules for TCP and UDP port 11111
[image]

[image]

I'm using the latest version of [putty](https://www.chiark.greenend.org.uk/~sgtatham/putty/latest.html)
[image]

Any tips on what I may be doing wrong? Thanks

JacobBarthelmeh, 2021-08-05:
Hi @gojimmypi , thanks for using wolfSSL + wolfSSH. There's a couple of items would like to check on to make sure the setup is as expected.

1) The default port for wolfSSH is 22222. Please check that if using port 11111 that the echoserver with wolfSSH has been started using ./examples/echoserver/echoserver -p 11111. wolfSSH is going to be using the SSH protocol which would not understand any TLS protocol packets.

2) echoclient is a TLS example bundled with wolfSSL (wolfSSL examples for TLS use the default 11111 port number). If trying to do a SSH connection using the wolfSSH examples please use wolfssh/examples/client/client.

3) Before, with putty, we have run into cases where it defaults to a different cipher than OpenSSH. Something like aes-ctr which can be turned on in wolfSSL by using the macros WOLFSSL_AES_COUNTER and WOLFSSL_AES_DIRECT.

Regards,
Jacob

gojimmypi, 2021-08-05:
Hi @JacobBarthelmeh and thanks for your reply.

I'm fairly certain the default port for my WolfSSH is `11111` and not `22222`, as noted in [wifi_connect.h](https://github.com/wolfSSL/wolfssl/blob/3e894a980448c1e743a19db04363a78d75b6ec7c/IDE/Espressif/ESP-IDF/examples/wolfssl_server/main/include/wifi_connect.h#L33) - not only as defined in the code, but there's a "connection refused" when attempting port `22222` (or any other port). At least I get an error on port `11111`.

I tried enabling `WOLFSSL_AES_COUNTER` and `WOLFSSL_AES_DIRECT` in

`./esp-idf/components/wolfssl/include/user_settings.h`
and
`./esp-idf/examples/protocols/wolfssl_server/user_settings.h`

I finally have the server also working on the ESP32, alas I get the same error. Here's putty:

[image]

And here's the full log on `COM4` with the ESP32 attached:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6896
load:0x40078000,len:14280
load:0x40080400,len:3744
entry 0x40080684
I (27) boot: ESP-IDF v4.4-dev-2359-g58022f859 2nd stage bootloader
I (27) boot: compile time 19:40:14
I (27) boot: chip revision: 1
I (31) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (47) boot.esp32: SPI Flash Size : 4MB
I (52) boot: Enabling RNG early entropy source...
I (57) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (95) boot_comm: chip revision: 1, min. application chip revision: 0
I (102) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1f390h (127888) map
I (157) esp_image: segment 1: paddr=0002f3b8 vaddr=3ffb0000 size=00c60h (  3168) load
I (158) esp_image: segment 2: paddr=00030020 vaddr=400d0020 size=a31c0h (668096) map
I (404) esp_image: segment 3: paddr=000d31e8 vaddr=3ffb0c60 size=02c44h ( 11332) load
I (409) esp_image: segment 4: paddr=000d5e34 vaddr=40080000 size=13f98h ( 81816) load
I (444) esp_image: segment 5: paddr=000e9dd4 vaddr=50000000 size=00010h (    16) load
I (454) boot: Loaded app from partition at offset 0x10000
I (454) boot: Disabling RNG early entropy source...
I (466) cpu_start: Pro cpu up.
I (466) cpu_start: Starting app cpu, entry point is 0x400810f0
I (0) cpu_start: App cpu up.
I (480) cpu_start: Pro cpu start user code
I (481) cpu_start: cpu freq: 160000000
I (481) cpu_start: Application information:
I (485) cpu_start: Project name:     tls_server
I (490) cpu_start: App version:      v4.8.0-stable-161-ga802c270e-di
I (497) cpu_start: Compile time:     Aug  5 2021 19:40:45
I (503) cpu_start: ELF file SHA256:  ed503ba5257ad259...
I (509) cpu_start: ESP-IDF:          v4.4-dev-2359-g58022f859
I (516) heap_init: Initializing. RAM available for dynamic allocation:
I (523) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (529) heap_init: At 3FFB8830 len 000277D0 (157 KiB): DRAM
I (535) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (542) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (548) heap_init: At 40093F98 len 0000C068 (48 KiB): IRAM
I (555) spi_flash: detected chip: gd
I (559) spi_flash: flash io: dio
I (563) sleep: Configure to isolate all GPIO pins in sleep state
[... 51 lines cut ...]
I (14937) wifi:state: run -> init (2a0)
I (14937) wifi:pm stop, total sleep time: 3260965 us / 4073742 us

I (14937) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (14947) example_connect: Wi-Fi disconnected, trying to reconnect...
I (16997) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1
I (17007) wifi:state: init -> auth (b0)
I (17007) wifi:state: auth -> assoc (0)
I (17017) wifi:state: assoc -> run (10)
I (18047) wifi:connected with [omit], aid = 3, channel 1, BW20, bssid = [omit]
I (18047) wifi:security: WPA2-PSK, phy: bgn, rssi: -86
I (18057) wifi:pm start, type: 1

I (18207) wifi:AP's beacon interval = 102400 us, DTIM period = 3
W (19187) wifi:<ba-add>idx:0 (ifx:0, [omit]), tid:0, ssn:0, winSize:64
I (20117) esp_netif_handlers: example_connect: sta ip: 192.168.1.28, mask: 255.255.255.0, gw: 192.168.1.1
I (20117) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.1.28
I (20127) example_connect: Connected to example_connect: sta
I (20127) example_connect: - IPv4 address: 192.168.1.28
I (20137) tls_server: Set dummy time...
I (20137) tls_server: The current date/time is: Wed Apr 28 23:00:00 2021
E (23857) tls_server: wolfSSL_accept error -311
E (23857) tls_server: ERROR: failed to read
E (23857) tls_server: ERROR: failed to write
```

gojimmypi, 2021-08-06:
Update: When debugging the [SSH Server](https://github.com/wolfSSL/wolfssl/tree/master/examples/server) in Visual Studio, and attempting to connect with the OpenSSH `ssh  127.0.0.1 -p 11111` command (which generated the same error described above), the resulting error is triggered here with an unexpected `rh->type` value:

[image]

... as `83` is not one of the enumeration options in the `switch` statement.  (`UNKNOWN_RECORD_TYPE` is `-311`)

gojimmypi, 2021-08-06:
I should add that I am able to get the _WolfSSL client_ to talk to the _WolfSSL server_. 

Here's the same server in Visual Studio:
[image]

A precompiled exe client from a DOS prompt talking to the above server:
[image]

And the respective exchange as seen by Wireshark:
[image]

The question is: how to get a client _other_ than the WolfSSL client, to talk to the WolfSSL server?

My real goal is to talk to the WolfSSL Server on an embedded device, specifically the Espressif ESP32 as noted in https://github.com/wolfSSL/wolfssl/issues/4272 - but as I am getting the same error in the Windows exe versions - I'm hoping the problem is more easily diagnosed with Visual Studio instead of the embedded device.

JacobBarthelmeh, 2021-08-10:
Hi @gojimmypi,

I suspect that a missmatch in protocols is happening. Where the TLS example server is started up and then a SSH client is trying to be used to connect to the TLS example server. The TLS example will not know anything about SSH protocol packets and report a bad record header. What is the print out of the example when passing the -h flag to it? For the example echoserver bundled with wolfSSH it should be:

```
$ cd ~/Documents/wolfssh
$ ./examples/echoserver/echoserver -h
echoserver 1.4.7
 -?            display this help and exit
 -1            exit after single (one) connection
 -e            expect ECC public key from client
 -E            use ECC private key
 -p <num>      port to connect on, default 22222
 -N            use non-blocking sockets
```

If no options print out then it is the TLS echoserver and a TLS client should be used to connect to it.

Regards,
Jacob

gojimmypi, 2021-08-10:
Hi @JacobBarthelmeh and thanks for your reply.

Do you have an example of using an ordinary SSH client to talk to the wolfSSL SSH server, and/or perhaps an example of specifying alternate keys for talking to the embedded wolfSSL SSH server with an arbitrary ssh client?

When I run `echoserver -h`, nothing happens; it immediately starts in server mode, listening for packets. If I try to connect to `echoclient`, nothing happens, no packets are echoed. If instead I simple start the `echoserver` (without the `-h`) in one terminal session, I am able to connect to by running `echoclient` in another window & observing the echoed text when  typing in the `echoclient` terminal. The echo examples work great. I am able to observe the exchange between localhost client & server on port `11111` in Wireshark:

[image]

The problem I am having, is using the [wolfSSL client](https://github.com/wolfSSL/wolfssl/tree/master/examples/client) from the [Visual Studio solution](https://github.com/wolfSSL/wolfssl/blob/master/wolfssl.sln) to talk to an actual SSH server. I have an instance of OpenSSH server running, and I can successfully connect with `ssh`, `rsh`, etc.

I was hoping if I could figure out why the wolfSSH client would not talk to a non-wolfSSH server, that perhaps I could ultimately determine why my `ssh` client will not talk to the [embedded ESP32 server](https://github.com/wolfSSL/wolfssl/tree/master/IDE/Espressif/ESP-IDF/examples/wolfssl_server). Instead I only see in the serial output:

```
I (4079108) esp_netif_handlers: example_connect: sta ip: 192.168.1.31, mask: 255.255.255.0, gw: 192.168.1.1
I (4079108) example_connect: Got IPv4 event: Interface "example_connect: sta" address: 192.168.1.31
E (4282478) tls_server: wolfSSL_accept error -313
E (4282478) tls_server: ERROR: failed to read
E (4282478) tls_server: ERROR: failed to write
```

Just for completeness, I also tried connecting the `echoclient` to the server (_not_ the `echoserver`); unsurprisingly, no luck there.

I have some [additional notes](https://github.com/gojimmypi/gojimmypi.github.io/blob/master/_posts/2021-08-05-wolfSSH-ssh-debugging.md) regarding how I am connecting, etc.

Any tips will be greatly appreciated. Thank you :)

JacobBarthelmeh, 2021-08-10:
Hi @gojimmypi 

For SSH connections please use the wolfSSH example echoserver (https://github.com/wolfSSL/wolfssh/tree/master/examples/echoserver). When compiled and then ran with -h it will have the previously mentioned help display. The echoserver currently being used is the TLS one from the wolfSSL directory which is unable to understand SSH packets, only TLS ones.

Regards,
Jacob

gojimmypi, 2021-08-10:
Hi @JacobBarthelmeh 

I'll give that a try. 

In the meantime, are you saying that the (non-echo example) [server](https://github.com/wolfSSL/wolfssl/tree/master/examples/server) and [client](https://github.com/wolfSSL/wolfssl/tree/master/examples/client) are not SSH and/or not operational at this time?

Are there any options of using a non-wolfSSL client to talk to any of the examples?

Thanks for your help.

JacobBarthelmeh, 2021-08-23:
Hi @gojimmypi 

wolfSSL (TLS protocol implementation) comes with an echoserver and echoclient.

wolfSSH (SSH protocol implementation) comes with an echoserver.

All of the examples are operational. But work only for the protocol that they were implemented for. SSH connections will not be able to connect to example TLS servers.

Regards,
Jacob

gojimmypi, 2021-08-24:
Hi @JacobBarthelmeh

Yes, of course I know the difference between SSH and TLS - however it appears I neglected to link to the proper example directories that I was referring to in my [message, above](https://github.com/wolfSSL/wolfssl/issues/4281#issuecomment-896067192). I had intended to refer to the [wolfSSL/wolfssh](https://github.com/wolfSSL/wolfssh/tree/master/examples) example [server](https://github.com/wolfSSL/wolfssh/tree/master/examples/server), and this [client](https://github.com/wolfSSL/wolfssh/tree/master/examples/client), both in the in the SSH repo. My bad for the same name, wrong repo links in my message. Thanks for pointing that out. 

Do you have an example of a non-wolfSSH client, such as `putty` or the Linux `ssh` command....  to connect to the wolfSSH example server? I was able to get the examples to _talk to each other_, but I was not able to get a non-wolfSSL SSH client to talk to the wolfSSL SSH server.

Thanks,

Jim

gojimmypi, 2021-08-29:
@JacobBarthelmeh I'm revisiting this today; I've started over on a fresh computer and erg: I think you are right! It seems I may have been mixing up the [wolfssh examples](https://github.com/wolfSSL/wolfssh/tree/master/examples)  `ssh` on port `22222` with the [wolfssl examples](https://github.com/wolfSSL/wolfssl/tree/master/examples) `tls` on port `11111`. (oops, my bad, sorry about that) 

Further, for some reason I thought SSH packets were TLS encrypted, but clearly that was wrong, too.  Here are encrypted SSH packets being sent over TCP port `22222`:

[image]

Thanks for your patience; I can only imagine the eye roll when you read my response above.  lol  ;)

JacobBarthelmeh, 2021-08-31:
No problem @gojimmypi, let me know if facing any outstanding issues after the fresh attempt.

closed 2021-09-30 by JacobBarthelmeh