wolfSSL/wolfssl #4298: memory leaks detected by using wolfSSL with OpenVPN

ordex, 2021-08-12:
When using OpenVPN master with wolfSSL master, it is possible to observe the following memleak by simply starting the server and then shutting it down:

```
==10554== 
==10554== HEAP SUMMARY:
==10554==     in use at exit: 120 bytes in 1 blocks
==10554==   total heap usage: 551 allocs, 550 frees, 264,460 bytes allocated
==10554== 
==10554== 120 bytes in 1 blocks are still reachable in loss record 1 of 1
==10554==    at 0x483F835: malloc (vg_replace_malloc.c:380)
==10554==    by 0x4918C12: _InitRng.isra.0 (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==10554==    by 0x49734D3: wolfSSL_RAND_Init (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==10554==    by 0x49AAB81: wolfSSL_Init (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==10554==    by 0x49AADB8: wolfSSL_add_all_algorithms (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==10554==    by 0x184818: tls_init_lib (ssl_openssl.c:80)
==10554==    by 0x17E008: init_ssl_lib (ssl.c:339)
==10554==    by 0x12D3EA: init_static (init.c:796)
==10554==    by 0x14BE2A: openvpn_main (openvpn.c:151)
==10554==    by 0x4C197FC: (below main) (in /lib64/libc-2.33.so)
==10554== 
==10554== LEAK SUMMARY:
==10554==    definitely lost: 0 bytes in 0 blocks
==10554==    indirectly lost: 0 bytes in 0 blocks
==10554==      possibly lost: 0 bytes in 0 blocks
==10554==    still reachable: 120 bytes in 1 blocks
==10554==         suppressed: 0 bytes in 0 blocks
==10554== 
==10554== For lists of detected and suppressed errors, rerun with: -s
==10554== ERROR SUMMARY: 0 errors from 0 contexts (suppressed: 0 from 0)
```

If you allow a client to connect and then shut down the server, you get the following:
```
==31038==
==31038== HEAP SUMMARY:
==31038== in use at exit: 168 bytes in 3 blocks
==31038== total heap usage: 1,208 allocs, 1,205 frees, 656,134 bytes allocated
==31038==
==31038== 24 bytes in 1 blocks are definitely lost in loss record 1 of 3
==31038== at 0x483F835: malloc (vg_replace_malloc.c:380)
==31038== by 0x49A8C19: wolfSSL_BIO_new (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==31038== by 0x189CFA: key_state_ssl_init (ssl_openssl.c:1890)
==31038== by 0x17F5A9: key_state_init (ssl.c:940)
==31038== by 0x17FD6B: tls_session_init (ssl.c:1127)
==31038== by 0x1819A6: tls_multi_init_finalize (ssl.c:1278)
==31038== by 0x12FF5C: do_init_finalize_tls_frame (init.c:3083)
==31038== by 0x12FF5C: do_init_frame_tls (init.c:3333)
==31038== by 0x12FF5C: init_instance (init.c:4253)
==31038== by 0x1310C7: inherit_context_child (init.c:4491)
==31038== by 0x146362: multi_create_instance (multi.c:772)
==31038== by 0x13FD8E: multi_get_create_instance_udp (mudp.c:103)
==31038== by 0x146761: multi_process_incoming_link (multi.c:3119)
==31038== by 0x14035B: multi_process_io_udp (mudp.c:211)
==31038== by 0x14035B: tunnel_server_udp (mudp.c:325)
==31038==
==31038== 24 bytes in 1 blocks are definitely lost in loss record 2 of 3
==31038== at 0x483F835: malloc (vg_replace_malloc.c:380)
==31038== by 0x49A8C19: wolfSSL_BIO_new (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==31038== by 0x189CFA: key_state_ssl_init (ssl_openssl.c:1890)
==31038== by 0x17F5A9: key_state_init (ssl.c:940)
==31038== by 0x17FD6B: tls_session_init (ssl.c:1127)
==31038== by 0x12FF5C: do_init_finalize_tls_frame (init.c:3083)
==31038== by 0x12FF5C: do_init_frame_tls (init.c:3333)
==31038== by 0x12FF5C: init_instance (init.c:4253)
==31038== by 0x1310C7: inherit_context_child (init.c:4491)
==31038== by 0x146362: multi_create_instance (multi.c:772)
==31038== by 0x13FD8E: multi_get_create_instance_udp (mudp.c:103)
==31038== by 0x146761: multi_process_incoming_link (multi.c:3119)
==31038== by 0x14035B: multi_process_io_udp (mudp.c:211)
==31038== by 0x14035B: tunnel_server_udp (mudp.c:325)
==31038== by 0x14B2E8: openvpn_main (openvpn.c:287)
==31038==
==31038== 120 bytes in 1 blocks are still reachable in loss record 3 of 3
==31038== at 0x483F835: malloc (vg_replace_malloc.c:380)
==31038== by 0x4918C12: _InitRng.isra.0 (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==31038== by 0x49734D3: wolfSSL_RAND_Init (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==31038== by 0x49AAB81: wolfSSL_Init (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==31038== by 0x49AADB8: wolfSSL_add_all_algorithms (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==31038== by 0x1877A8: tls_init_lib (ssl_openssl.c:80)
==31038== by 0x1811B8: init_ssl_lib (ssl.c:339)
==31038== by 0x12CFDA: init_static (init.c:796)
==31038== by 0x14AF61: openvpn_main (openvpn.c:151)
==31038== by 0x4C197FC: (below main) (in /lib64/libc-2.33.so)
==31038==
==31038== LEAK SUMMARY:
==31038== definitely lost: 48 bytes in 2 blocks
==31038== indirectly lost: 0 bytes in 0 blocks
==31038== possibly lost: 0 bytes in 0 blocks
==31038== still reachable: 120 bytes in 1 blocks
==31038== suppressed: 0 bytes in 0 blocks
==31038==
==31038== For lists of detected and suppressed errors, rerun with: -s
==31038== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)
```

While on the client the result is this (seems identical to the snippet above):
```
==11284== 
==11284== HEAP SUMMARY:
==11284==     in use at exit: 168 bytes in 3 blocks
==11284==   total heap usage: 932 allocs, 929 frees, 577,702 bytes allocated
==11284== 
==11284== 24 bytes in 1 blocks are definitely lost in loss record 1 of 3
==11284==    at 0x483F835: malloc (vg_replace_malloc.c:380)
==11284==    by 0x49A8C19: wolfSSL_BIO_new (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==11284==    by 0x186D6A: key_state_ssl_init (ssl_openssl.c:1890)
==11284==    by 0x17C409: key_state_init (ssl.c:940)
==11284==    by 0x17D017: tls_session_init (ssl.c:1127)
==11284==    by 0x17E815: tls_multi_init_finalize (ssl.c:1278)
==11284==    by 0x130744: do_init_finalize_tls_frame (init.c:3083)
==11284==    by 0x130744: do_init_frame_tls (init.c:3333)
==11284==    by 0x130744: init_instance (init.c:4253)
==11284==    by 0x131760: init_instance_handle_signals (init.c:4046)
==11284==    by 0x14C003: tunnel_point_to_point (openvpn.c:67)
==11284==    by 0x14C003: openvpn_main (openvpn.c:283)
==11284==    by 0x4C197FC: (below main) (in /lib64/libc-2.33.so)
==11284== 
==11284== 24 bytes in 1 blocks are definitely lost in loss record 2 of 3
==11284==    at 0x483F835: malloc (vg_replace_malloc.c:380)
==11284==    by 0x49A8C19: wolfSSL_BIO_new (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==11284==    by 0x186D6A: key_state_ssl_init (ssl_openssl.c:1890)
==11284==    by 0x17C409: key_state_init (ssl.c:940)
==11284==    by 0x17D017: tls_session_init (ssl.c:1127)
==11284==    by 0x130744: do_init_finalize_tls_frame (init.c:3083)
==11284==    by 0x130744: do_init_frame_tls (init.c:3333)
==11284==    by 0x130744: init_instance (init.c:4253)
==11284==    by 0x131760: init_instance_handle_signals (init.c:4046)
==11284==    by 0x14C003: tunnel_point_to_point (openvpn.c:67)
==11284==    by 0x14C003: openvpn_main (openvpn.c:283)
==11284==    by 0x4C197FC: (below main) (in /lib64/libc-2.33.so)
==11284== 
==11284== 120 bytes in 1 blocks are still reachable in loss record 3 of 3
==11284==    at 0x483F835: malloc (vg_replace_malloc.c:380)
==11284==    by 0x4918C12: _InitRng.isra.0 (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==11284==    by 0x49734D3: wolfSSL_RAND_Init (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==11284==    by 0x49AAB81: wolfSSL_Init (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==11284==    by 0x49AADB8: wolfSSL_add_all_algorithms (in /home/ordex/exp/builds/wolfssl-master/lib/libwolfssl.so.24.5.1)
==11284==    by 0x184818: tls_init_lib (ssl_openssl.c:80)
==11284==    by 0x17E008: init_ssl_lib (ssl.c:339)
==11284==    by 0x12D3EA: init_static (init.c:796)
==11284==    by 0x14BE2A: openvpn_main (openvpn.c:151)
==11284==    by 0x4C197FC: (below main) (in /lib64/libc-2.33.so)
==11284== 
==11284== LEAK SUMMARY:
==11284==    definitely lost: 48 bytes in 2 blocks
==11284==    indirectly lost: 0 bytes in 0 blocks
==11284==      possibly lost: 0 bytes in 0 blocks
==11284==    still reachable: 120 bytes in 1 blocks
==11284==         suppressed: 0 bytes in 0 blocks
==11284== 
==11284== For lists of detected and suppressed errors, rerun with: -s
==11284== ERROR SUMMARY: 2 errors from 2 contexts (suppressed: 0 from 0)
```

No memleaks are reported when the same code is compiled/linked against OpenSSL

julek-wolfssl, 2021-08-13:
@ordex I have posted a fix for this report here: https://github.com/wolfSSL/wolfssl/pull/4301. Thanks for the report!
This takes care of the `definitely lost` memory. The `still reachable` memory is stuff allocated with `SSL_library_init`. To free this memory, OpenVPN should call `OPENSSL_cleanup` in `tls_free_lib`.

ordex, 2021-08-13:
@julek-wolfssl from the OpenSSL manpage:

       The OPENSSL_cleanup() function deinitialises OpenSSL (both libcrypto and libssl). All resources
       allocated by OpenSSL are freed. Typically there should be no need to call this function directly
       as it is initiated automatically on application exit. This is done via the standard C library
       atexit() function. In the event that the application will close in a manner that will not call
       the registered atexit() handlers then the application should call OPENSSL_cleanup() directly.
       Developers of libraries using OpenSSL are discouraged from calling this function and should
       instead, typically, rely on auto-deinitialisation. This is to avoid error conditions where both
       an application and a library it depends on both use OpenSSL, and the library deinitialises it
       before the application has finished using it.

Assuming that the user should call OPENSSL_cleanup() is a non-negligible deviation from the OpenSSL API, no?
Basically it would be required for wolfSSL only.

Not a big deal, because we are talking about a constant memleak (not increasing over time) that manifests itself at program exit, but still worth pointing out.

julek-wolfssl, 2021-08-13:
That's actually quite interesting. I was assuming that OpenSSL just doesn't have any global structures to free. I added an atexit call to the PR. Let's see how it does in code review.

julek-wolfssl, 2021-12-03:
https://github.com/wolfSSL/wolfssl/pull/4301 has been merged and this issue is resolved. Thank you for your report @ordex!

closed 2021-12-03 by julek-wolfssl