wolfSSL/wolfssl #4319: Undefined referance to ssl.h ibrary functions

hobik, 2021-08-20:
I am trying to use WolfSSL library on stm32f4xx platform with LWIP/NON-RTOS. When I start the simple server/client example. I generated the source code on stm32cubeMx and its makefile with wolfssl.

Than I try to use the library, but I get below error:
`/home/duygu/Documents/Nucleo_F4_WolfSSL/Src/main.c:96: undefined reference to `wolfSSL_Init'
`
<wolfssl/ssl.h> include in main.c but I have an error. if I use any function this library I couldnt use.
I checked makefile but every header file included. 

main.c:
```
/* USER CODE BEGIN Header */
/**
  ******************************************************************************
  * @file           : main.c
  * @brief          : Main program body
  ******************************************************************************
  * @attention
  *
  * <h2><center>&copy; Copyright (c) 2021 STMicroelectronics.
  * All rights reserved.</center></h2>
  *
  * This software component is licensed by ST under Ultimate Liberty license
  * SLA0044, the "License"; You may not use this file except in compliance with
  * the License. You may obtain a copy of the License at:
  *                             www.st.com/SLA0044
  *
  ******************************************************************************
  */
/* USER CODE END Header */

/* Includes ------------------------------------------------------------------*/
#include "main.h"
#include "lwip.h"
#include <wolfssl/options.h>
#include <wolfssl/ssl.h>
/* Private includes ----------------------------------------------------------*/
/* USER CODE BEGIN Includes */
#define WOLFSSL_STM32F4
#define HAVE_LWIP_NATIVE
#define WOLFSSL_USER_IO
#define HAVE_LWIP_NATIVE

//#define NO_WOLFSSL_CLIENT
/* USER CODE END Includes */

/* Private typedef -----------------------------------------------------------*/
/* USER CODE BEGIN PTD */

/* USER CODE END PTD */

/* Private define ------------------------------------------------------------*/
/* USER CODE BEGIN PD */
/* USER CODE END PD */

/* Private macro -------------------------------------------------------------*/
/* USER CODE BEGIN PM */

/* USER CODE END PM */

/* Private variables ---------------------------------------------------------*/
RNG_HandleTypeDef hrng;

UART_HandleTypeDef huart4;
UART_HandleTypeDef huart3;

PCD_HandleTypeDef hpcd_USB_OTG_FS;

/* USER CODE BEGIN PV */

/* USER CODE END PV */
[... 309 lines cut ...]
  /* User can add his own implementation to report the HAL error return state */

  /* USER CODE END Error_Handler_Debug */
}

#ifdef  USE_FULL_ASSERT
/**
  * @brief  Reports the name of the source file and the source line number
  *         where the assert_param error has occurred.
  * @param  file: pointer to the source file name
  * @param  line: assert_param error line source number
  * @retval None
  */
void assert_failed(uint8_t *file, uint32_t line)
{ 
  /* USER CODE BEGIN 6 */
  /* User can add his own implementation to report the file name and line number,
     tex: printf("Wrong parameters value: file %s on line %d\r\n", file, line) */
  /* USER CODE END 6 */
}
#endif /* USE_FULL_ASSERT */

/************************ (C) COPYRIGHT STMicroelectronics *****END OF FILE****/

```

dgarske, 2021-08-20:
Hi @hobik ,

A couple things:
1) Make sure you are not building for wolfCrypt only `WOLFCRYPT_ONLY`. If you are then you would use `wolfCrypt_Init()` and `wolfCrypt_Cleanup` instead.

2) With the Cube pack you should not include or use `wolfssl/options.h`. The options.h is a generated file from ./configure and not used by the pack. Instead you should include `#include <wolfssl/wolfcrypt/settings.h>`. At the top of that file it will include `wolfSSL.I-CUBE-wolfSSL_conf.h` when `USE_HAL_DRIVER` is set. See https://github.com/wolfSSL/wolfssl/blob/master/wolfssl/wolfcrypt/settings.h#L231

Let me know if that helps.

Thanks,
David Garske, wolfSSL

hobik, 2021-08-23:
Hi dgarske,
1. I checked wolfSSL.I-CUBE-wolfSSL_conf.h file than it has;
```
. /*---------- WOLF_CONF_WOLFCRYPT_ONLY -----------*/
#define WOLF_CONF_WOLFCRYPT_ONLY      1
```

Than I added as your suggestion, I used wolfCrypt_Cleanup() instead of wolfCrypt_Init() . Than I build again than I got these errors:

```
/home/tommy/Documents/Nucleo_F4_WolfSSL/Src/main.c:102: undefined reference to `wolfTLSv1_2_server_method'
/home/tommy/Documents/Nucleo_F4_WolfSSL/Src/main.c:103: undefined reference to `wolfSSL_CTX_new'
/home/tommy/Documents/Nucleo_F4_WolfSSL/Src/main.c:104: undefined reference to `wolfSSL_CTX_set_verify'
```
2. I removed wolfssl/options.h and added #include <wolfssl/wolfcrypt/settings.h> than I defined USE_HAL_DRIVER. But already have same errors.
This is my source code: 
 https://easyupload.io/rp0pku

dgarske, 2021-08-23:
Hi @hobik ,

If you define `WOLF_CONF_WOLFCRYPT_ONLY` then the TLS code is disabled. If you need TLS you must set `WOLF_CONF_WOLFCRYPT_ONLY = 0`.

Also now you are including both `user_settings.h` and `wolfSSL.I-CUBE-wolfSSL_conf.h`. You need to use one or the other. To use `user_settings.h` add `-DWOLFSSL_USER_SETTINGS` to your CFLAGS and remove `#include "user_settings.h"` from your main.h.

To get things building I had to do:
1) Remove "user_settings.h" from main.h.
2) Add "Hardware platform" to `wolfSSL.I-CUBE-wolfSSL_conf.h` at line 130 `#elif defined(STM32F437xx) || defined(STM32F429xx)`.
3) Make sure `NO_STM32_CRYPTO` and `NO_STM32_HASH` are set to disable on chip symmetric support on this device.

Thanks,
David Garske, wolfSSL