wolfSSL/wolfssl #4323: Feature Request: Port wolfSSL to vcpkg

Kiddinglife, 2021-08-21:


dgarske, 2021-08-23:
Hi Jake,

Re #3226. The feature request for "Port wolfSSL to vcpkg" is on our list to consider. If you are interested in providing some context to help us justify accelerating this work this please email facts@wolfssl.com with details.

Thanks,
David Garske, wolfSSL

al-sabr, 2021-12-25:
Plugin wolfSSL inside of any C++ Cmake project would improve the package dependency management. OpenSSL is already ported to vcpkg and everyone is using it. If you want us to switch to wolfSSL this a good motivator.

elms, 2022-04-22:
@Kiddinglife @al-sabr Good news, I have put a draft [PR](https://github.com/microsoft/vcpkg/pull/24348) to add vcpkg support. I'll work through the open issues and target merging after the next wolfSSL release.

elms, 2022-05-18:
vcpkg for wolfSSL, wolfMQTT, and wolfTPM have merged https://github.com/microsoft/vcpkg/pull/24348

closed 2022-05-18 by elms